MichaIng/DietPi #4644: Samba install error (fresh install of bulseye on RPi Zero W)

maartenlangeveld, 2021-08-16:
#### Details:
- Date           | Mon Aug 16 12:38:28 CEST 2021
- Bug report     | 7766cda8-6c4e-4098-a499-f20e1a20372b
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux M-MUSIC-PLAYER 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install samba`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://mirror.truenetwork.ru/raspbian/raspbian/pool/main/m/media-types/media-types_4.0.0_all.deb  Could not connect to mirror.truenetwork.ru:80 (0.0.0.0). - connect (111: Connection refused) Could not connect to mirror.truenetwork.ru:80 (::). - connect (111: Connection refused)
E: Failed to fetch http://mirror.truenetwork.ru/raspbian/raspbian/pool/main/p/python3.9/libpython3.9_3.9.2-1+rpi1_armhf.deb  Unable to connect to mirror.truenetwork.ru:80:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2021-08-16:
That seems more an issue on the Raspberry repository mirror you are connected to

`mirror.truenetwork.ru:80 (0.0.0.0)`

Seems to be blocked by an AdBlocker

maartenlangeveld, 2021-08-16:
Aha, thanks... why connect to russian mirrors anyway? Don't trust those countries...

Joulinar, 2021-08-16:
Well that's a question we are not able to answer. Redirecting to a mirror is done by Raspberry web server.

maartenlangeveld, 2021-08-16:
Ok, thanks!

maartenlangeveld, 2021-08-16:
Finally it installed but config menu (fileserver) still says none

Joulinar, 2021-08-16:
How did you installed SAMBA? Using `dietpi-software` ?

maartenlangeveld, 2021-08-16:
Yes, re-installed and now it's okay! Thanks

Joulinar, 2021-08-16:
Ok going to close the issue for now. Feel free to reopen if needed.

closed 2021-08-16 by Joulinar

MichaIng, 2021-08-16:
> Aha, thanks... why connect to russian mirrors anyway? Don't trust those countries...

And western governments, companies and servers are not known to spy on users and have data leaks regularly? 😄
While the political democracy conditions can be questioned, there are many awesome organisations and individuals contributing to the open source community in "those countries", which is not fair to reduce or set equal to the countries government and political system condition.

Luckily APT is not depending on trust, but the repo itself and each package is checked against the stored trusted key and hashes to be original.