MichaIng/DietPi #4645: Justboom Digi Zero card not recognised on Dietpi Bullseye (RPI Zero W)

maartenlangeveld, 2021-08-16:
```
                    ┌─────────────────────────────────────────────────┤ DietPi-JustBoom ├──────────────────────────────────────────────────┐
                    │ Tweak your High-End Audiophile JustBoom (and others) sound card:                                                     │
                    │                                                                                                                      │
                    │                                           ●─ Global Options ─────────────────────────────●                           │
                    │                          ALSA Mixer       : Launches ALSA Mixer, tweak sound card settings                           │
                    │                          ALSA Output Info : View info on current ALSA playback stream                                │
                    │                          EQ Enable        : [Off] | Select to toggle EQ                                              │
                    │                          Reset            : Reset all options to defaults                                            │
                    │                                           ●─ MPD Specific ───────────────────────────────●                           │
                    │                          Output Frequency :  Hz | Select to change                                                   │
                    │                          Output BitRate   :  Bit | Select to change                                                  │
                    │                          Audio Buffer     : 2048 Bytes | Select to change                                            │
                    │                          SOXR Enable      : [Off] | Select to toggle SOXR resampling                                 │
                    │                                                                                                                      │
                    │                                                                                                                      │
                    │                                  <Ok>                                      <Exit>                                    │
                    │                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

```
                    ┌─────────────────────────────────────────────────┤ DietPi-JustBoom ├──────────────────────────────────────────────────┐
                    │                                                                                                                      │
                    │ ALSA Mixer will now be launched, when finished, press F10 or ESC to exit.                                            │
                    │                                                                                                                      │
                    │                                                        <Ok>                                                          │
                    │                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```
Nothing happens, mene quits and on commandline following is displayed:

```
root@M-MUSIC-PLAYER:~# dietpi-justboom
cannot open mixer: No such file or directory
```

MichaIng, 2021-08-16:
Can you try to run it from console:
```sh
alsamixer
```

maartenlangeveld, 2021-08-16:
`root@M-MUSIC-PLAYER:~# alsamixer
cannot open mixer: No such file or directory`

MichaIng, 2021-08-16:
If you just enabled onboard WiFi or switched the sound card, did you do a reboot already? I can replicate the error when using a non-existing sound card ID in `/etc/asound.conf`, which would fit to that case where on RPi we toggle things via device tree overlays and boot options.

maartenlangeveld, 2021-08-16:
Yes, I already did a reboot...

MichaIng, 2021-08-16:
Can you check the detected sound devices and compare with the asound.conf:
```sh
aplay -l
cat /etc/asound.conf
```

MichaIng, 2021-08-16:
Okay, so the sound driver failed to load that HAT. But the overlay is present, isn't it?
```sh
grep '^dtoverlay=justboom-digi$' /boot/config.txt
```
Does the kernel show any related errors or messages?
```sh
dmesg -l emerg,alert,crit,err
# or all logs
dmesg
```

maartenlangeveld, 2021-08-16:
```
root@M-MUSIC-PLAYER:~# dmesg -l emerg,alert,crit,err
[    1.625349] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[   14.214332] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   14.509277] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   14.509495] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   14.524070] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb 16 2020 22:39:24 version 7.45.98.97 (r724416 CY) FWID 01-bf41ed64
[   23.014603] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
```

```
root@M-MUSIC-PLAYER:~# grep '^dtoverlay=justboom-digi$' /boot/config.txt
root@M-MUSIC-PLAYER:~#
```

MichaIng, 2021-08-16:
It seems the overlay was not applied, somehow:
```sh
G_CONFIG_INJECT 'dtoverlay=justboom-digi' 'dtoverlay=justboom-digi' /boot/config.txt
#reboot
```

maartenlangeveld, 2021-08-16:
Thank again and again. This solved the issue!

MichaIng, 2021-08-16:
Strange that it was not applied in the first place when selecting the sound card. Probably there was one unclean reboot and something lost in the filesystem cache or so, which caused that missing Samba logs dir as well 🤔. However, great that all got sorted now 🙂.

closed 2021-08-20 by MichaIng