MichaIng/DietPi #4646: Using two (both) UARTs on RPi Compute Module

danergo, 2021-08-16:
I know I can use both UARTs on a RPiCM by adding these nice lines into `config.txt`:

```
enable_uart=1
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33
```

This works nice, I have two serials on `ttyAMA0` (uart0), and on `ttyS0` (uart1).

But, now I need uart1 to have not only RXTX but also CTS and RTS.

According to BCM2835's datasheet, it can provide UART1 on these pins (in 5th ALT mode):

```
GPIO30: CTS1
GPIO31: RTS1
GPIO32: TXD1
GPIO33: RXD1
```
So instead of using `uart1` overlay, I wrote my own:

```
/dts-v1/;
/plugin/;

/*
   Enable /dev/ttyS0 with RTS and CTS signals
   TX is available on GPIO 32
   RX is available on GPIO 33
   CTS is available on GPIO 30
   RTS is available on GPIO 31
*/

/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&uart1>;
        __overlay__ {
                pinctrl-names = "default";
                pinctrl-0 = <&uart1_gpio32 &uart1_ctsrts_gpio30>;
        };
    };
};
```

I compile it on the target RPiCM with this makefile rule:

```
%.dtbo: %.dts
	dtc -@ -H epapr -O dtb -o $@ -Wno-unit_address_vs_reg $^
```

This generates `uart1-full.dtbo`, and I placed this into `/boot/overlays`, and modified `config.txt` accordingly:

```
enable_uart=1
dtoverlay=uart1-full
```

However after reboot, I only have `ttyAMA0` (uart1), and nothing related to uart1. Do you have any idea on what am I missing in my overlay definition?

My kernel is this:

`Linux DietPi 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux`

And I'm on DietPi v7.4.2 currently (I believe this is quite up-2-date).

Thank you!

MichaIng, 2021-08-16:
I was checking a bit, and indeed it should be generally possible. Here an example which enables CTS/RTS on GPIOs 16/17: https://github.com/HiassofT/AtariSIO/blob/master/contrib/rpi/miniuart-ctsrts.dts

I'm not fully aware of the Makefile device tree compiler syntax, could you try to compile it from console via:
```sh
dtc -I dts -O dtb -o /boot/overlays/uart1-full.dtbo uart1-full.dts
```

And I read that on CM, the mini UART is disabled unless the overlay is set, so not sure if it includes setting the status. So you could try:
```
/dts-v1/;
/plugin/;

/*
   Enable /dev/ttyS0 with RTS and CTS signals
   TX is available on GPIO 32
   RX is available on GPIO 33
   CTS is available on GPIO 30
   RTS is available on GPIO 31
*/

/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&uart1>;
        __overlay__ {
                pinctrl-names = "default";
                pinctrl-0 = <&uart1_gpio32 &uart1_ctsrts_gpio30>;
                status = "okay";
        };
    };
};
```

danergo, 2021-08-16:
Hi Michael, I was using the same library for skeleton:)

Will try out your suggestion in a couple of hours, and feedback.

Thanks a lot!

MichaIng, 2021-08-16:
> Hi Michael, I was using the same library for skeleton:)

I had a feeling that it doesn't look that similar coincidentally 😄.

danergo, 2021-08-16:
By any chance: do you know how can I check if my dtbo is actually being used? Or this information won't help?

danergo, 2021-08-16:
Adding `status = "okay";` didn't help, there is no ttyS0 still. I'll keep trying and update this thread once I found the solution.

MichaIng, 2021-08-16:
`cat /proc/device-tree/aliases/uart1` shows you the path of the device within the device tree. In my case it's `soc/serial@7e215040`, so `cat /proc/device-tree/soc/serial@7e215040/status` shows me the status of the UART1 device. And I guess you should see the other entries you applied via overlay as well.

Ah I also found the valid GPIO symbols you can use:
```
ls -l /proc/device-tree/__symbols__/uart*
```
On my RPi, the ones you use are valid.

danergo, 2021-08-16:
Thanks, this is useful.

Some additions: if I add `dtoverlay=uart1` to `config.txt`, it creates `/dev/ttyS0` correctly.

My dtbo is not working (yet), so something is probably missing. Although I have rewritten it to not include the CTS/RTS pins:

```
/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&uart1>;
        __overlay__ {
                pinctrl-names = "default";
                pincrtl-0 = <&uart1_pins>;
                status = "okay";
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
                uart1_pins: uart1_pins {
                        brcm,pins = <32 33>;
                        brcm,function = <2>;
                        brcm,pull = <0 2>;
                };
        };
    };
};
```

danergo, 2021-08-16:
Do you know any good docs about the proper syntax and valid fields of a DTS file? For example, brcm,pull, what is the "2" meaning? What happens if I wrote "1" instead of "2"? Same for `brcm,function`, how can I know the possible values?

MichaIng, 2021-08-17:
Basic syntax can be found here: https://www.raspberrypi.org/documentation/computers/configuration.html#device-trees-overlays-and-parameters
But I have no idea about the individual keys either. That is probably best to ask at the RPi forum.

But would it actually work to both overlays, `uart1` and yours as override of the override, so that you only need to define the `pinctrl-0`? Not sure if this works and overlays are applied in alphabetical order, but worth to give it a try?

danergo, 2021-08-17:
Thanks!

I have found some nice docs about this topic, will share them later.

Now both UARTs are up and running, although I couldn't really test if they
are actually using their CTS/RTS features.

I will create some scope measurements to validate this and then I'm gonna
share my results and the solution.

Cheers

danergo, 2021-08-17:
Hi, this is the doc I've referred about: [https://www.kernel.org/doc/Documentation/devicetree/bindings/pinctrl/brcm,bcm2835-gpio.txt](https://www.kernel.org/doc/Documentation/devicetree/bindings/pinctrl/brcm,bcm2835-gpio.txt
)
This lists the valid fields of a BCM2835 processor.

And this is my DTS which works for UART0 (ttyAMA0):

```
/dts-v1/;
/plugin/;

/{
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&uart0>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&uart0_pins &uart0_ctsrts_pins>;
            status = "okay";
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            uart0_pins: uart0_pins {
	        brcm,pins = <14 15>;
	        brcm,function = <4>; /* alt0 */
                brcm,pull = <0 2>;
            };
            uart0_ctsrts_pins: uart0_ctsrts_pins {
	        brcm,pins = <16 17>;
            	brcm,function = <7>; /* alt3 */
	        brcm,pull = <2 0>;
            };
        };
    };

    fragment@2 {
        target-path = "/chosen";
            __overlay__ {
                bootargs = "8250.nr_uarts=1";
        };
    };
```

And this is for UART1 (/dev/ttyS0):

```
/dts-v1/;
/plugin/;

/{
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&uart1>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&uart1_pins>;
            status = "okay";
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            uart1_pins: uart1_pins {
	        brcm,pins = <30 31 32 33>;
	        brcm,function = <2>; /* alt5 */
                brcm,pull = <2 0 0 2>;
            };
        };
    };

    fragment@2 {
        target-path = "/chosen";
            __overlay__ {
                bootargs = "8250.nr_uarts=1";
        };
    };
```

Both UARTs now works with hardware flow control now.

I have only 2 issues:

1. No idea about `target-path="/chosen";`, but without that section UART1 doesn't seem working.
2. Cutting RTS/CTS wires on UART0 leads to not receive/send any more characters, but after joining back the cable(s), both parties receive the remaining data in buffer. This is not the case on UART1: cutting the RTS/CTS wire(s) doesn't affect communication, however you can't start communication with cut RTS/CTS wires. I have no idea on why. (Maybe this is why it's called "miniUART"?)

MichaIng, 2021-08-17:
Check out:
```sh
ls -l /proc/device-tree/chosen
```
It seems to be another way of adding arguments to the boot/kernel command line, applied before `/boot/cmdline.txt`. So in theory, adding `8250.nr_uarts=1` to the latter should work as well, while of course it is elegant to tie it into the overlay. Since its present in both overlays, I guess that option is now two times there, but that should be an issue:
```sh
cat /proc/cmdline
```

For RTS/CTS to work, I think you need to re-add `&uart1_ctsrts_gpio30` to the `pinctrl-0` value of the UART1 overlay. Given the fact that this symbol exists and the name clearly indicates that it's for UART1, this should work.
Alternatively, you could define a symbol explicitly, like it is done in the UART0 fragment2, defining `uart1_ctsrts_pins: uart1_ctsrts_pins {` and adding/moving the GPIOs 30 and 31 there?

Check out:
```sh
ls -l /proc/device-tree/soc/gpio@7e200000/uart1_ctsrts_gpio30/
```
This is the existing symbol, containing the keys that are otherwise defined/overwritten by the overlay. In my case the containing "files" are empty, but that is probably only because I have UART completely disabled. In your case, probably you can compare the values with how you'd define them, or how they are defined in the UART0 overlay, and then you know whether you can use it as is, or define your own symbol for CTS/RTS pins in the overlay.

danergo, 2021-08-21:
> ls -l ***@***.***/uart1_ctsrts_gpio30/

Michael,

thank you, I believe my serial ports are working now, however, I can't use
them for HCI UART:

    hciattach -s 921600 /dev/ttyS0 any 921600 noflow
    Can't set line discipline: Invalid argument
    Can't initialize device: Invalid argument

This led me to a forum where it was stated that in this error message case,
the kernel might not contain the necessary HCI UART configuration.

Do you know if this could be really the case, or maybe do you have any idea
on how can I include these HCI UART configs? Maybe with insmod somehow?

(Note: On Ubuntu 20.04, the above command succeeds - with another serial
device ofc).

danergo, 2021-08-22:
Thank you!

Both ports are now working with those default pin configurations, and I've also corrected the cmdline issue you've referred about.

closed 2021-08-22 by danergo