MichaIng/DietPi #4647: SMB does not seem to work on DietPi Bullseye (RPi Zero W)

maartenlangeveld, 2021-08-16:
When accessing \\192.168.x.x\dietpi in windows I get the error message:

> Windows cannot access \\192.168.x.x\dietpi

Diagnose does not provide more help:

> Windows can't find "\\192.168.x.x\dietpi". The name might be misspelled.

MichaIng, 2021-08-16:
Check the server status:
```sh
journalctl -u smbd -u nmbd
tail -10 /var/log/samba/log.0
```

maartenlangeveld, 2021-08-16:
```
root@M-MUSIC-PLAYER:~# journalctl -u smbd -u nmbd
-- Journal begins at Mon 2021-08-16 17:07:03 CEST, ends at Mon 2021-08-16 18:46:28 CEST. --
Aug 16 18:44:15 M-MUSIC-PLAYER systemd[1]: Starting Samba NMB Daemon...
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: directory_create_or_exist: mkdir failed on directory /var/log/samba/cores: No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: Failed to create /var/log/samba/cores for user 0 with mode 0700
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: Unable to setup corepath for nmbd: No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: [2021/08/16 18:44:20,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.nmbd': No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: [2021/08/16 18:44:20.075797,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.nmbd': No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: [2021/08/16 18:44:20.076063,  0] ../../source3/nmbd/nmbd.c:960(main)
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   nmbd version 4.13.5-Debian started.
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   Copyright Andrew Tridgell and the Samba Team 1992-2020
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: [2021/08/16 18:44:20.122835,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.': No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]: [2021/08/16 18:44:20.135802,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:20 M-MUSIC-PLAYER nmbd[521]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.nmbd': No such file or directory
Aug 16 18:44:20 M-MUSIC-PLAYER systemd[1]: Started Samba NMB Daemon.
Aug 16 18:44:20 M-MUSIC-PLAYER systemd[1]: Starting Samba SMB Daemon...
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26,  0] ../../lib/util/util.c:364(directory_create_or_exist)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   directory_create_or_exist: mkdir failed on directory /var/log/samba/cores: No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26,  0] ../../source3/lib/dumpcore.c:59(get_default_corepath)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   Failed to create /var/log/samba/cores for user 0 with mode 0700
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26,  0] ../../source3/lib/dumpcore.c:256(dump_core_setup)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   Unable to setup corepath for smbd: No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26.892697,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26.892953,  0] ../../source3/smbd/server.c:1784(main)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   smbd version 4.13.5-Debian started.
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   Copyright Andrew Tridgell and the Samba Team 1992-2020
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26.945676,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26.956517,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]: [2021/08/16 18:44:26.976251,  0] ../../lib/util/debug.c:1104(reopen_one_log)
Aug 16 18:44:26 M-MUSIC-PLAYER smbd[566]:   reopen_one_log: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Aug 16 18:44:27 M-MUSIC-PLAYER systemd[1]: Started Samba SMB Daemon.
```

maartenlangeveld, 2021-08-16:
```
root@M-MUSIC-PLAYER:~# tail -10 /var/log/samba/log.0
tail: cannot open '/var/log/samba/log.0' for reading: No such file or directory
```

Joulinar, 2021-08-16:
Can you check if the log directory is existing at all

`ls -la /var/log`

MichaIng, 2021-08-16:
Ah yeah, `/var/log/samba/log.0` is supposed to not exist, I forgot how Samba log files are structured. But the service logs indeed indicate that the directory does not exist at all:
```sh
mkdir /var/log/samba
systemctl restart smbd nmbd
```

maartenlangeveld, 2021-08-16:
Thanks !
(Maybe the install script should make /var/log/samba dir beforehand ?)

But I still cannot access the SMB share from my windows PC. Now it asks for credentials but still cannot access. But refuses to connect.

```
Aug 16 19:34:34 M-MUSIC-PLAYER systemd[1]: Stopping Samba NMB Daemon...
Aug 16 19:34:34 M-MUSIC-PLAYER systemd[1]: nmbd.service: Succeeded.
Aug 16 19:34:34 M-MUSIC-PLAYER systemd[1]: Stopped Samba NMB Daemon.
Aug 16 19:34:34 M-MUSIC-PLAYER systemd[1]: nmbd.service: Consumed 1.028s CPU time.
Aug 16 19:34:34 M-MUSIC-PLAYER systemd[1]: Starting Samba NMB Daemon...
Aug 16 19:34:35 M-MUSIC-PLAYER systemd[1]: Started Samba NMB Daemon.
Aug 16 19:34:38 M-MUSIC-PLAYER systemd[1]: Stopping Samba SMB Daemon...
Aug 16 19:34:38 M-MUSIC-PLAYER systemd[1]: smbd.service: Succeeded.
Aug 16 19:34:38 M-MUSIC-PLAYER systemd[1]: Stopped Samba SMB Daemon.
Aug 16 19:34:38 M-MUSIC-PLAYER systemd[1]: smbd.service: Consumed 11.399s CPU time.
Aug 16 19:34:38 M-MUSIC-PLAYER systemd[1]: Starting Samba SMB Daemon...
Aug 16 19:34:39 M-MUSIC-PLAYER systemd[1]: Started Samba SMB Daemon.
Aug 16 19:35:10 M-MUSIC-PLAYER smbd[1720]: pam_unix(samba:session): session opened for user dietpi(uid=1000) by (uid=0)
Aug 16 19:35:57 M-MUSIC-PLAYER smbd[1723]: pam_unix(samba:session): session opened for user dietpi(uid=1000) by (uid=0)
Aug 16 19:35:58 M-MUSIC-PLAYER smbd[1723]: pam_unix(samba:session): session closed for user dietpi
Aug 16 19:35:58 M-MUSIC-PLAYER smbd[1723]: pam_unix(samba:session): session opened for user dietpi(uid=1000) by (uid=0)
Aug 16 19:35:58 M-MUSIC-PLAYER smbd[1723]: pam_unix(samba:session): session closed for user dietpi
Aug 16 19:36:00 M-MUSIC-PLAYER smbd[1724]: pam_unix(samba:session): session opened for user dietpi(uid=1000) by (uid=0)
Aug 16 19:36:00 M-MUSIC-PLAYER smbd[1724]: pam_unix(samba:session): session closed for user dietpi
Aug 16 19:36:00 M-MUSIC-PLAYER smbd[1724]: pam_unix(samba:session): session opened for user dietpi(uid=1000) by (uid=0)
Aug 16 19:36:00 M-MUSIC-PLAYER smbd[1724]: pam_unix(samba:session): session closed for user dietpi
Aug 16 19:36:10 M-MUSIC-PLAYER smbd[1720]: pam_unix(samba:session): session closed for user dietpi
```

MichaIng, 2021-08-16:
> Maybe the install script should make /var/log/samba dir beforehand ?

It does, at least on my test. The `samba-common` package contains it.

Now the log files can help:
```sh
for i in /var/log/samba/log.*; do tail -10 "$i"; done
```

maartenlangeveld, 2021-08-16:
root@M-MUSIC-PLAYER:~# for i in /var/log/samba/log.*; do tail -10 "$i"; done
tail: cannot open '/var/log/samba/log.*' for reading: No such file or directory

MichaIng, 2021-08-16:
> i guess since /var/log is mapped to ramdisk?

It is, but the contained files and directories (structure, not content) should be backed up on shutdown and reloaded on boot. Can you check the RAMlog logs:
```sh
tail -10 /var/tmp/dietpi/logs/dietpi-ramlog.log
ls -l /var/tmp/dietpi/logs/dietpi-ramlog_store
```

maartenlangeveld, 2021-08-16:
Yes, it was but I will invoek windows smb credentials and try again.

MichaIng, 2021-08-16:
Hmm, and you double checked the credentials, share name etc, as given in `/etc/samba/smb.conf`. I just tried to mount a Samba share from one Linux to another with default credentials and share name (all `dietpi`), which works fine. Haven't tried on Windows yet. Is it latest Windows 10? Not sure if older Windows versions might not support the protocol used on latest Linux Samba server implementation, although if not, it should fail before even being able to send credentials 🤔.

Joulinar, 2021-08-16:
Did you used the same share from same IP in the past but with different credentials for user `dietpi`

maartenlangeveld, 2021-08-16:
```
root@M-MUSIC-PLAYER:/# tail -10 /var/tmp/dietpi/logs/dietpi-ramlog.log
Mon Aug 16 19:54:16 CEST 2021 | DietPi-RAMlog: [ INFO ] Storing logs to disk...
Mon Aug 16 19:54:16 CEST 2021 | DietPi-RAMlog: [  OK  ] Stored logs to disk.
Mon Aug 16 19:54:25 CEST 2021 | DietPi-RAMlog: [ INFO ] Restoring logs from disk...
Mon Aug 16 19:54:25 CEST 2021 | DietPi-RAMlog: [  OK  ] Restored logs from disk.
```
```
root@M-MUSIC-PLAYER:/# ls -l /var/tmp/dietpi/logs/dietpi-ramlog_store
total 20
-rw-r--r-- 1 root     root        0 Aug 16 19:45 alternatives.log
drwxr-xr-x 2 root     root     4096 Aug 16 14:44 apt
-rw-rw---- 1 root     utmp        0 Aug 16 19:17 btmp
-rw-r--r-- 1 root     root        0 Aug 16 15:17 dpkg.log
-rw-r--r-- 1 root     root        0 Aug 16 14:17 fontconfig.log
-rw-rw-r-- 1 root     utmp        0 Aug 16 19:44 lastlog
drwxr-x--- 2 www-data www-data 4096 Aug 16 13:07 lighttpd
-rw-r--r-- 1 root     root        0 Aug 16 19:17 minimserver-init.log
drwxr-xr-x 2 mpd      audio    4096 Aug 16 13:34 mpd
drwxr-s--- 2 mysql    adm      4096 Aug 16 13:13 mysql
-rw------- 1 root     root        0 Aug 16 19:54 php7.4-fpm.log
drwx------ 2 root     root     4096 Aug 16 12:17 private
-rw-rw-r-- 1 root     utmp        0 Aug 16 19:54 wtmp
```
Now after latest reboot, /var/log/samba seems persistent.

However, access to smb share is still denied to my Windows PC

```
root@M-MUSIC-PLAYER:/# cat /var/log/samba/log.m-machine-5 |more
[2021/08/16 20:29:08.322636,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.322937,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
  error probing vfs module 'widelinks': NT_STATUS_UNSUCCESSFUL
[2021/08/16 20:29:08.323286,  0] ../../source3/smbd/vfs.c:379(smbd_vfs_init)
  smbd_vfs_init: widelinks enabled and vfs_init_custom failed for vfs_widelinks module
[2021/08/16 20:29:08.323429,  0] ../../source3/smbd/service.c:636(make_connection_snum)
  make_connection_snum: vfs_init failed for service IPC$
[2021/08/16 20:29:08.343253,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.343490,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
  error probing vfs module 'widelinks': NT_STATUS_UNSUCCESSFUL
[2021/08/16 20:29:08.343644,  0] ../../source3/smbd/vfs.c:379(smbd_vfs_init)
  smbd_vfs_init: widelinks enabled and vfs_init_custom failed for vfs_widelinks module
[2021/08/16 20:29:08.343780,  0] ../../source3/smbd/service.c:636(make_connection_snum)
  make_connection_snum: vfs_init failed for service IPC$
[2021/08/16 20:29:08.389737,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.389971,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
  error probing vfs module 'widelinks': NT_STATUS_UNSUCCESSFUL
[2021/08/16 20:29:08.390133,  0] ../../source3/smbd/vfs.c:379(smbd_vfs_init)
  smbd_vfs_init: widelinks enabled and vfs_init_custom failed for vfs_widelinks module
[2021/08/16 20:29:08.390271,  0] ../../source3/smbd/service.c:636(make_connection_snum)
  make_connection_snum: vfs_init failed for service IPC$
[2021/08/16 20:29:08.395466,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.395699,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
  error probing vfs module 'widelinks': NT_STATUS_UNSUCCESSFUL
[2021/08/16 20:29:08.395857,  0] ../../source3/smbd/vfs.c:379(smbd_vfs_init)
  smbd_vfs_init: widelinks enabled and vfs_init_custom failed for vfs_widelinks module
[2021/08/16 20:29:08.395994,  0] ../../source3/smbd/service.c:636(make_connection_snum)
  make_connection_snum: vfs_init failed for service IPC$
[2021/08/16 20:29:08.406563,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.406799,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
  error probing vfs module 'widelinks': NT_STATUS_UNSUCCESSFUL
[2021/08/16 20:29:08.406957,  0] ../../source3/smbd/vfs.c:379(smbd_vfs_init)
  smbd_vfs_init: widelinks enabled and vfs_init_custom failed for vfs_widelinks module
[2021/08/16 20:29:08.407251,  0] ../../source3/smbd/service.c:636(make_connection_snum)
  make_connection_snum: vfs_init failed for service IPC$
[2021/08/16 20:29:08.412840,  0] ../../lib/util/modules.c:49(load_module)
  Error loading module '/usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so': /usr/lib/arm-linux-gnueabihf/samba/vfs/widelinks.so: cannot open shared object fil
e: No such file or directory
[2021/08/16 20:29:08.413070,  0] ../../source3/smbd/vfs.c:185(vfs_init_custom)
--More--
```

maartenlangeveld, 2021-08-16:
Hi Michael,

Thanks a lot, that solved my issue !

Let me explain a bit. 

I have added following options to the [global] section of /etc./samba/smb.conf because I need it in my setup (I have dedicated mounted partition containing my music files and /mnt/dietpi_userdata/Music is a symlink to that partition).
        
```
allow insecure wide links = yes
wide links = yes
```
Everything worked fine with DietPi Buster build, I guess because the required module was installed standard before???

MichaIng, 2021-08-16:
> I guess because the required module was installed standard before???

Actually not, that package is not a dependency. But, as the package on Buster does not include the `widelinks.so`, probably it was a core feature in Samba previously, and has been moved to the extended VFS features with a recent version 🤔.

MichaIng, 2021-08-16:
This seems to be the related changelog in Samba v4.12.0: https://www.samba.org

> MS-DFS changes in the VFS
> -------------------------
> 
> This release changes set getting and setting of MS-DFS redirects
> on the filesystem to go through two new VFS functions:
> 
> SMB_VFS_CREATE_DFS_PATHAT()
> SMB_VFS_READ_DFS_PATHAT()
> 
> instead of smbd explicitly storing MS-DFS redirects inside
> symbolic links on the filesystem. The underlying default
> implementations of this has not changed, the redirects are
> still stored inside symbolic links on the filesystem, but
> moving the creation and reading of these links into the VFS
> as first-class functions now allows alternate methods of
> storing them (maybe in extended attributes) for OEMs who
> don't want to mis-use filesystem symbolic links in this
> way.

maartenlangeveld, 2021-08-16:
Hi Michael,

Samba server is working now but final note: it is not asking for credentials (as opposed to samba in Dietpi Buster, using the same smb.conf file...). For me not a problem but might be security risk for others?

Joulinar, 2021-08-16:
I guess you had the share mounted before on Windows? Or was it first time with this IP?

maartenlangeveld, 2021-08-16:
I had it mounted before but I removed the credentials in Windows Credential manager

MichaIng, 2021-08-16:
Hmm, that should not be possible, unless Windows caches those somewhere. The `valid users = dietpi` directive forces authentication with the password of the `dietpi` system user, at least it should. I'll do some test tomorrow in Windows.

maartenlangeveld, 2021-08-16:
I will reboot my PC and test again, maybe Windows does not empty credentials from cache when the credentials are revoked. Will let you know!

Joulinar, 2021-08-16:
I did some testing on RPi1 Bullseye and could not replicate one of the issue.

- /var/log/samba was created during installation
- Windows was asking for user/password while connection

I had the experience in the past, sometimes Windows is keeping the share hidden in the background, even if disconnected. Usually this can be checked using `net use` in a terminal window.

maartenlangeveld, 2021-08-17:
Hi,

Just checked, after reboot I had to provide samba credentials on my Windows PC when accessing the samba share. Obviously Windows 10 caches the credentials and does not clear its cache when the credentials are revoked.

Thanks, everything working now!

@MichaIng @Joulinar **your support is very much appreciated !!!**

closed 2021-08-17 by MichaIng