MichaIng/DietPi #4648: DietPi RPi images already install Bullseye but official Raspbian still Buster...

maartenlangeveld, 2021-08-16:
I use my RPi Zero Was a music player. Running upmpdcli. However upmpdcli at the moment does offer a debian bullseye install package but not a RASPBIAN one. 

I contacted the maker of upmpdcli to ask when upmpdcli becomes availbe for Raspbian Bulleye, he replied that todate Raspbian Bullseye is not released officially.

I don't think it's wise to ditribute DietPi images with Bullseye whilst Raspbian Bullseye is still not officialy released.

maartenlangeveld, 2021-08-16:
I understand, makes sense.

However cannot install buster version:

```
root@M-MUSIC-PLAYER:/etc/apt/sources.list.d# apt install upmpdcli
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 upmpdcli : Depends: libjsoncpp1 (>= 1.7.4) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2021-08-16:
Ah indeed that one changed its name from `libjsoncpp1` to `libjsoncpp24`. Try to install it manually:
```sh
cd /tmp
curl -sSfLO http://raspbian.raspberrypi.org/raspbian/pool/main/libj/libjsoncpp/libjsoncpp1_1.7.4-3_armhf.deb
apt install ./libjsoncpp1_1.7.4-3_armhf.deb
```

maartenlangeveld, 2021-08-16:
Hi Michael,

Thanks again for yor truly great support. This helped me a lot.

Joulinar, 2021-08-16:
@MichaIng 
all my test system are running Bullseye now. 

RPi1 - ARMv6 (SD card)
RPi3B+ - ARMv7 (USB boot)
RPi4B - ARMv8 (USB boot)

MichaIng, 2021-08-16:
Mine are running "Bookworm" now 😄. A benefit of switching immediately is, that there is no change. The new testing is exactly the old testing and changes are merged from unstable gradually. So chances are there to detect breaking changes as fast as they are merged to testing. The problem is, until now there has not been a `base-files` upgrade, so `/etc/os-release` and `/etc/debian_version` still report Bullseye/11 and hence I also cannot add detection to DietPi yet 😢.

maartenlangeveld, 2021-08-16:
Hi Michael,

Need your assistance with thise one:

```
2021-08-17 00:37:36 root@M-MUSIC-PLAYER:/# apt install upmpdcli-uprcl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-recoll : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Can you help me?

MichaIng, 2021-08-17:
Okay, that is gonna be complicated. Bullseye ships with Python 3.9 and this package depends on Python 3.7 or earlier. You can try to override the dependency, but it may not gonna work when the script indeed uses functions or depends on implementations which are only present on older Python versions. So when doing this, carefully check the behaviour/logs or the related commands:
```sh
apt install python3
cd /tmp
curl -sSfLO 'https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian/pool/main/r/recoll/python3-recoll_1.29.2-1~ppa1~buster_armhf.deb'
dpkg -i --force-depends python3-recoll_1.29.2-1~ppa1~buster_armhf.deb
# to check whether this pattern is present like that
grep 'python3 (< 3.8)' /var/lib/dpkg/status
# replace it
sed -i 's/python3 (< 3.8)/python3 (< 3.10)/' /var/lib/dpkg/status
apt install upmpdcli-uprcl
```

maartenlangeveld, 2021-08-17:
Hi Michael,

Thanks!! Haves succeeded to instal with your guidance. Will see if it keeps working correctly and keep an eye on log files.

maartenlangeveld, 2021-08-19:
Hi @MichaIng,

Your prayers have been heard, since yesterday the developer of upmpdcli has made raspbian bullseye packages available! :-)

When will you include upmpdcli in DietPi sofware list? It's a great tool that turns mpd into a DLNA and OpenHome player. It works great with Android BubbleUPnP and BubbleDS control points. The latter one support radiolist and metada script (Python) support (scripts reside on mpd player with updmpcli, that push metadata such as song info and cover art to the control point). 

> > Bullseye is not released officially, so he cannot provide a Raspbian build.
> 
> Lol, why does a stable release need to be done before packages are build for it. Sadly many developers handle it like that, but that causes a lot of trouble since it makes it impossible to test whether 3rd party software works stable BEFORE a distro version is released to be stable. We start to create testing version packages as fast as the new testing version got its `base-files` package, so that it can be identified via `/etc/os-release`. So bugs can be identified and source or build changes applied to have a 100% functional package once the stable release is there.
> 
> Well, back to the actual issue: Simply keep using the Raspbian Buster repository of upmpdcli, which should work pretty fine on Bullseye as well. Generally older packages mostly work fine on newer distros, as shared library dependencies are defined via `>=`, so only have a minimum required version, not an exact or maximum one.

MichaIng, 2021-08-20:
Related PR: #2245
Give it a bump at: https://feathub.com/MichaIng/DietPi/+19

I'll mark the issue as closed then. Feel free to reopen if required.

closed 2021-08-20 by MichaIng