MichaIng/DietPi #4649: Image | Quartz64

ravenclaw900, 2021-08-16:
### Creating an image request

#### Formal device information
- Device name | 
Quartz64
- Official product URL | 
https://www.pine64.org/quartz64a/
- Image download URL | 
https://wiki.pine64.org/wiki/Quartz64#Software_and_OS_Image_Downloads

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- No

#### If not, are there install instructions for Debian available?

- Not yet, probably due to new chipset (RK3588)

Obviously we'll have to wait a bit for a reliable image before building DietPi for it, just wanted to put the request out there.

MichaIng, 2021-08-16:
Also the Debian arm64 kernel does not contain a device tree for that SoC yet. I'm currently downloading the Manjaro Linux image, probably we can take something from there for a testing image 😛.

MichaIng, 2021-08-16:
It has a dedicated bootloader and a dedicated boot/EFI partition. Theoretically we could use them untouched and move/dd them into a debootstraped arm64 image 🙊. The kernel can also be found here: https://mirrors.dotsrc.org/manjaro/arm-unstable/core/aarch64/

MichaIng, 2022-02-12:
Today a nice surprise from @pine64 arrived, many thanks for this 👍😃❤️:

[image]

From left to right, we have:
- Quartz64 Model B, not sure about the RAM size, yet: https://www.pine64.org/quartz64b/
- Quartz64 Model A, with 8 GiB RAM: https://www.pine64.org/quartz64a/
- SoQuartz baseboard, which a little different connectors and headers, compared to the Quartz64 model A: (no product page available yet)
- SoQuartz compute module for mounting to the base board, with 4 GiB RAM: https://wiki.pine64.org/wiki/SOQuartz

More information about all products can be found on the wiki: https://wiki.pine64.org/wiki/Quartz64

As can be seen, any OS development is in very early stage, also Armbian has leaked that there is an unreleased image, but far away from being ready for publishing to end users. There is a repository with a CI pipeline for producing Debian installer images for the different boards, which also does regular releases: https://gitlab.com/pgwipeout/quartz64_ci/-/jobs
I'll see whether this gets us bootstrapped, whether we can adapt it for our own image build script, how well a single image can run on the different board variants (given all device trees are present, of course) etc. Will be much fun to start tinkering with these boards, and probably we can be one of the distros which provide stable (though likely not with all hardware features fully supported at start) image for the Quartz64 family.

MichaIng, 2022-04-09:
Tested with the latest [pgwipeout](https://gitlab.com/pgwipeout/quartz64_ci/-/jobs) images:
- On Quartz64 A, dtb and mainline U-Boot images boot, but no HDMI output is produced. First I thought as of the pending upstream kernel patch (https://patchwork.kernel.org/project/linux-rockchip/list/?series=630407), but pgwipeout uses a patched kernel and the screen actually gets an image, it shows a static underscore, but it looks like that console is not used for kernel/boot outputs. LEDs indicate boot and operation state. Sadly I toasted my USB-UART adapter. New one ordered, arriving next week. I hope it's not the 2A PSU being insufficient. Generally booting from SD card and eMMC works, but not from USB. Not sure whether this is a hardware/ROM limitation, or probably insufficient PSU. The USB flash isn't powered (LEDs) at all when the board is powered.
- For Quartz64 B, I needed to order a 3.5mm/1.35mm power cable first. USB power inputs would have made things easier, but there will be reasons.

So more on that end next Tuesday. I hope get UART output and probably video output is a question if the kernel command-line only. Else even without HDMI (for now), with DietPi SSH server starting OOTB that wouldn't be such an issue.

I'll also try to debootstrap DietPi and install/flash pgwipeout kernel/U-Boot, and [Manjaro](https://github.com/manjaro-arm/quartz64-bsp-images) Linux to verify that my PSU is not the issue and that/how the boards fully boot.

MichaIng, 2022-04-09:
Nice, it was really just the missing local console output, HDMI works pretty well, no issues with 2A PSU. Added `console=tty1` to kernel command-line and was able to boot into buildroot recovery and Debian installer. The installer is for whatever reason Debian Buster, and installs the generic Debian arm64 4.19 kernel, and no bootloader. However, I installed it on the free space of the same SD card and configured extlinux to use the new partition as root. Worked well, now DietPi-PREP is running, doing the Bullseye upgrade as well.

While it would be possible to ship a DietPi image like this, with GPT partitioning, dedicated bootloader, boot env and EFI partitions, I'd actually prefer to ship with a simple MBR partition table with a single ext4 partition. I'll try it via flashing U-Boot to the raw block device manually.

However, working so far 🙂:
```sh
root@Quartz64A:~# cat /var/lib/dietpi/dietpi-benchmark/results
BENCH_VERSION=2
BENCH_HW_MODEL=22
BENCH_CPU='10.70'
BENCH_ROOTFS_WRITE='34'
BENCH_ROOTFS_READ='59'
BENCH_RAM_WRITE='401'
BENCH_RAM_READ='1351'
BENCH_CUSTOMFS_WRITE='Not tested'
BENCH_CUSTOMFS_READ='Not tested'
BENCH_CPU_TEMP_START='44'
BENCH_CPU_TEMP_END='59'
BENCH_NET_LAN_SPEED='Not tested'
root@Quartz64A:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     46 °C / 114 °F : Optimal temperature
 Governor     |     schedutil

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1416 MHz      408 MHz    1800 MHz
 CPU1         |      1416 MHz      408 MHz    1800 MHz
 CPU2         |      1416 MHz      408 MHz    1800 MHz
 CPU3         |      1416 MHz      408 MHz    1800 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.

root@Quartz64A:~# free -m
               total        used        free      shared  buff/cache   available
Mem:            7687          74        7516          48          96        7458
Swap:              0           0           0
root@Quartz64A:~# cat /proc/device-tree/model
Pine64 RK3566 Quartz64-A
root@Quartz64A:~# uname -a
Linux Quartz64A 5.17.0 #1 SMP PREEMPT Thu Apr 7 13:38:56 UTC 2022 aarch64 GNU/Linux
```
This SoC is not as powerful as expected, like Odroid XU4/Exynos5, Odroid C4/Amlogic S905X3, Allwinner H6. For the last two, it is not so unexpected as the CPU cores are similar:
- Amlogic S905X3: Quad-Core Cortex-A55 @ 2.0 GHz
- Allwinner H6: Quad-Core ARM Cortex-A53 @ 1.8GHz (probably 2 GHz with Armbian kernel)
- Rockchip RK3566: Quad-Core ARM Cortex-A55 @ 1.8GHz

The RK3566 is meant as a successor for the RK3288 (ASUS Tinker Board), while the RK3399 is still the Hexa-Core flagship which as two Cortex-A72 on top of the four Cortex-A53 cores.

RAM speed is also in a similar area. Further kernel tuning may of course increase performance.

Sadly I don't have the optional WiFi module. But it's built into model B (PSU arriving Tuesday) and SOQuartz.

Next step is trying to forge an MBR single-partition image.

MichaIng, 2022-06-06:
DietPi images for the three variants are available for testing: https://dietpi.com/downloads/images/testing/

MichaIng, 2022-06-26:
Wasted several hours debugging on all 3 models why they deny to boot (from Samsung EVO Plus `MB-MC128HA/EU` SD card), before realising that when testing it successfully some days ago, I used the other SD card Samsung PRO Endurance `MB-MJ128KA`, which still works fine. So somehow it's a bid picky about the SD card 🤔.

Also, while model A and SOQuartz boot fine, model B does not, waiting for root filesystem in systemd boot stage. I think it is because of my bad power adapter cable. 3W PSU, but bought a fitting cable with probably bad quality. Also HDMI doesn't work on this model (on both others it does), which is another indicator for a power/voltage issue.

If someone has a model B and time to test, that would be great. And I'll evaluate in the meantime whether I buy a better cable, or the original PSU from PINE64 in the meantime.

MichaIng, 2022-07-02:
WireGuard module is missing: https://gitlab.com/pgwipeout/quartz64_ci/-/blob/main/quartz64_defconfig
I'll open a MR and rebuild with some other possible enhancements (e.g. bumping Linux to v5.19, a related branch exists already).

However, otherwise it works fine.

closed 2022-07-02 by MichaIng