MichaIng/DietPi #4650: kodi won't install on Bullseye using dietpi-software

sibofax, 2021-08-16:
#### Details:
- Date           | Mon Aug 16 22:45:57 BST 2021
- Bug report     | cdda07d5-4f8f-4968-9e66-3304fa3d9d64
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux box 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install libcec6 libnfs13 kodi fbset`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Package 'kodi' has no installation candidate
```

sibofax, 2021-08-16:
apt search kodi only returns add-on or dev packages

Joulinar, 2021-08-16:
Hi, 

which image you have download? ARMv6 or ARMv7?

sibofax, 2021-08-16:
Hi - it was ARMv6  
https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Bullseye.7z

Joulinar, 2021-08-17:
At the moment Raspbian Bullseye did not contain the Kodi package.

You could try the ARMv7 image

MichaIng, 2021-08-17:
Okay that is nasty. The RPi Kodi package is unmaintained for over a year now, which implies no 64-bit build, no Bullseye build, and I have no idea if/when this will happen 🤔. And Raspbian does not ship the main package built on any distro version, though the sources are there. Debian ships all those, but not GPU accelerated on RPi, so no a pleasant experience.

There are some build instructions and promising comments first that it works fine with GPU acceleration: https://www.raspberrypi.org/forums/viewtopic.php?p=1824600#p1824600
But latest comments state that it does not work, probably related to the 5.4 => 5.10 kernel upgrade, not sure. But this could be generally tried.

Not sure how to handle this and I likely won't find the time to build and ship own RPi Kodi packages. Kodi is quite a major software option and having it available for the major SBC is somehow important. But I'm neither keen to keep shipping Buster images only for this, nor doing some complicated repo mix do get the non-accelerated Debian build on Raspbian Bullseye available. What we could do, is either for this case or in general keep the RPi Buster repo added on Bullseye, until the most important packages have been rebuild, but I fear this opens possible package and library conflicts. Also, the RPi Kodi version starts to getting old, and people are asking for v19 already, with Bullseye where this is natively shipped (on Debian), it seems even stranger to keep shipping Kodi 18.

But we should do something about it, not yet sure what. I'll keep an eye on the RPi forum, to get any news, and track how other users deal with it, suffering from the same issue. And probably I can do at least a test compiling of latest Kodi, to see if it goes through at least and if the result starts on RPi.

Joulinar, 2021-08-17:
Quite a large amount of depending packages. Question is if all of them are available on Bullseye... 🤔 

Stupid question: Would it be possible to add Raspbian repository with a lower prio, just to get stuff that might not be available on Debian repository?

MichaIng, 2021-08-17:
Ah that is true, at least it is build against the wrong CEC and NFS library versions (those package names have changed, but they are no direct dependencies, only installed by dietpi-software). Not that does not make sense.

> Would it be possible to add Raspbian repository with a lower prio, just to get stuff that might not be available on Debian repository?

That is practically the case automatically, as APT prefers newer package versions over older ones, and does not downgrades without interactive confirmation (or a forcing command line option). The priority can be changed via APT pinning, but that is only required when you add a newer repository, like we did for WireGuard and PHP7.3 on Stretch.

> probably this doesn't even make sense ...

Your point with the large amount of dependencies is pretty valid: Even if we manage to install the older package, it is nearly guaranteed that something does not work due a mismatch between installed shared library versions and the one the Kodi binaries were compiled against. This needs to be redone on Bullseye.

Joulinar, 2021-08-17:
should we make a note on our download page about Kodi and the challenges around? Atm it seems Bullseye is not best choise if you like to run Kodi.

MichaIng, 2021-08-17:
I'll at least add a check to dietpi-software to show it as disabled on Raspbian Bullseye, unless there is a package found. Not sure if we should add another note to the already large amount of download options etc for RPi, but indeed it makes sense to not waste users time which download DietPi to run Kodi to then recognise that it doesn't work... I mean this was the case with the 64-bit image already, but not great.

I'll try to build Kodi ASAP, the more I think about it, the more I think own packages are the only solution, as there may be none in the RPi repo for a long time.

Joulinar, 2021-08-17:
Removing it for Bullseye might be the only way for now as we have 7.5 close to the release.

sibofax, 2021-08-17:
Just to add that the ARMv7 image doesn't appear to work either - comes up with 
"failed to open zone.tab
ERROR: unable to create GUI. Exiting"
Then drops to an ssh prompt
I've not done any more troubleshooting apart from trying root rather than a user as the auto login - but same symptoms.
So this was from a clean install of DietPi - however - I know that Kodi works fine in Buster.  As you have mentioned, it may be that using Bullseye for Kodi is a bit of overkill taking into account repos and dependencies.  The raspberry foundation still have LibreElec on their downloads page (which I've just tried) and that is Kodi 18.7.

MichaIng, 2021-08-19:
Btw, Kodi is compiling, it just takes a very long time. While watching the output and assuring that compiling flags are set correctly to have an ARMv6 compatible binary, I recognised a false flag chosen by the build script based on given ARMv6 CPU/ARCH. When offering a fix upstream, the developer made me aware of a special input CPU `arm1176jzf-s` for Raspberry Pi 1/Zero, which sets some additional flags and was implemented by RPi developers some years ago. I never saw this being used anywhere in the forum, so this might be one of the reasons the users were not too successful in getting Kodi running on their device.

I hope I can do a test tonight, so hoping for a solution shipped with DietPi v7.5 already.

MichaIng, 2021-08-21:
Package is finally done and works, I mean Kodi starts and I'm able even on the slow RPi Zero to navigate through the menu. Time over for more testing, but independently of a DietPi release, we can fix/tweak/enhance the package, if required: https://github.com/MichaIng/DietPi/commit/6e47e822bdedaa24fbc6356ee50b7554097f3198
Changelog: https://github.com/MichaIng/DietPi/commit/54de6b159c8a95057df0ac5ea19999d39830b617

Joulinar, 2021-08-22:
time to create our own apt package server 😉

MichaIng, 2021-08-22:
Yes, on the ToDo: https://medium.com/sqooba/create-your-own-custom-and-authenticated-apt-repository-1e4a4cf0b864
Actually, instead of using tools which automate all of this, I'm fans of writing own scripts, so every step is well understood, like the hashing and signing steps and creating the related meta files. And then I have in mind an atomic update process: Two instances of the APT repos are hosted, one with the live version and once for the staging version. When updates are done to the staging version, `https://dietpi.com/apt/` is simply a symlink which is changed from the staging to the live version. So there is no time for inconsistencies during the update. Only the Cloudflare cache needs to be cleared as well then, at least for the meta and list files which do not change their names 🤔.

closed 2021-08-22 by MichaIng