MichaIng/DietPi #4651: AUTO_SETUP_AUTOMATED=1 Keeps failing

skavinvarnan, 2021-08-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
bullseye
- Kernel version | `uname -a`
Linux mytesthost 5.10.52+ 1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Raspberry Pi Zero W
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4 A
- SDcard used | (EG: SanDisk ultra)
Samsung Evo PRO 32GB

#### Steps to reproduce
1. Install a brand new image. Change config ```AUTO_SETUP_AUTOMATED=1```
2. Boot it on raspberry pi zero

#### Expected behaviour
1. Should connect to the internet. 
2. Should do all the setup process without issue

#### Actual behaviour
1. Raspberry pi is connecting to the internet
2. But fails on the ```/boot/dietpi/dietpi-update``` step
[image]

#### Extra details
1. When i force the auto setup on the second launch but manually changing to ```AUTO_SETUP_AUTOMATED=1``` the process runs smoothly.
2. There is no problem with my ISP or internet speed. Its a decent broadband at 50Mbps
3. I'm able to reproduce this on every launch

Digging the code found that ```/boot/dietpi/func/run_ntpd``` runs smoothly but ```/boot/dietpi/dietpi-update``` on the first launch

**Possibly fixes:**
1. Please have a retry mechanism for 2-3 time before failing the process here https://github.com/MichaIng/DietPi/blob/112e56c836a34ea70ddf11f89ac5a03253258271/dietpi/dietpi-update#L103
2. Or I believe a hack solution like adding a ```sleep 10``` should work here https://github.com/MichaIng/DietPi/blob/112e56c836a34ea70ddf11f89ac5a03253258271/dietpi/dietpi-login#L196

Joulinar, 2021-08-18:
Hi,

it looks like you have a DNS resolution issue as you are failing to resolve github host. Do you use any AdBlocker on your network? Did you tried different DNS server?

skavinvarnan, 2021-08-18:
No I don't have AdBlocker. But let me try using a different DNS server.  

Even if its a problem with DNS Server. How does it work on the second launch 🤔

MichaIng, 2021-08-18:
Strange indeed, also since G_CHECK_CON and G_CHECK_DNS (both succeeding with a single attempt, as I see in the output) and a time sync run first, assuring that network is up with a bunch of successful connections. No idea how then resolving that one hostname can fail reproducible a single time and succeed the next attempts. The DNS server has been called already several times at this point (time sync to `[0-3].debian.pool.ntp.org`).

We could allow up to `CONFIG_G_CHECK_URL_ATTEMPTS` retries for this download, which defaults to `2`.

skavinvarnan, 2021-08-19:
The solution to add a delay actually worked

Tests done. 
Without delay: 5 tried | 5 failed
With 30 seconds delay: 1 try | 0 failed
[image]

With 5 seconds delay: 2 try | 0 failed
[image]

Creating a PR soon

Joulinar, 2021-09-24:
@MichaIng 
what to do with this issue? Related PR has been closed already. Did you finished your testing?

MichaIng, 2021-11-11:
I'm unable to replicate such issue. Also with the v7.7 rework of how the system waits for network and time sync at boot thing may have changed. And with v7.8 these checks are not done in `dietpi-login` anymore but from within `dietpi-update`. I'll mark this hence as closed.

closed 2021-11-11 by MichaIng