MichaIng/DietPi #4652: DietPi-Software | Resilio

w1v1, 2021-08-18:
### Creating a software request

#### Formal software information
- Software title | Resilio
- Short description | Resilio Sync is a fast, reliable, and simple file sync and share solution, powered by P2P technology.
- Official URL (if available) | https://www.resilio.com/individuals/
- GitHub URL (if available) | 
- (Official) install documentation | https://help.resilio.com/hc/en-us/articles/206178924-Installing-Sync-package-on-Linux

#### Are there similar/alternative software titles available with DietPi-Software?
-  OwnCloud, Syncthing

#### What makes your requested software better than the above solutions, if available?

- Far more reliable than the alternatives, faster sync speeds, much easier to configure & use, apps for all devices, and a lot less maintenance to run. Although 'free' (individual) edition has some limitations regarding number of sync devices (3), optional payment to use gives unlimited. Also good for low-cost commercial use when using RPI.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Better conformity to the DietPi landscape (user/owners, permissions, etc) using the *.deb install package.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

- No need, Resilio does the updates, compiling and patches. Can assist with testing if required.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2021-08-18:
Installation seems quite simple.

- add key for repository
- add repository
- `apt install resilio-sync`
- add user to dietpi group
- enable/start service
- web interface is listen on port `https://<ip>:8888`

working fine

```
root@DietPi3:~# systemctl status resilio-sync
● resilio-sync.service - Resilio Sync service
     Loaded: loaded (/lib/systemd/system/resilio-sync.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-08-18 16:25:09 CEST; 19min ago
       Docs: https://help.resilio.com
   Main PID: 8070 (rslsync)
      Tasks: 20 (limit: 2178)
        CPU: 17.451s
     CGroup: /system.slice/resilio-sync.service
             └─8070 /usr/bin/rslsync --config /etc/resilio-sync/config.json

Aug 18 16:25:06 DietPi3 systemd[1]: Starting Resilio Sync service...
Aug 18 16:25:09 DietPi3 systemd[1]: Started Resilio Sync service.
root@DietPi3:~#
```

there are not that much config options

```
root@DietPi3:~# cat /etc/resilio-sync/config.json
{
    "storage_path" : "/var/lib/resilio-sync/",
    "pid_file" : "/var/run/resilio-sync/sync.pid",

    "webui" :
    {
        "force_https": true,
        "listen" : "0.0.0.0:8888"
    }
}
root@DietPi3:~#
```

Main settings seems to be stored in a data file `settings.dat`

```
root@DietPi3:/var/lib/resilio-sync# cat settings.dat
d10:.fileguard40:9495B6EFBBAED12CEA161BD1A78DD4650229444616:actual_bind_porti39950e24:agreed_userterms_versioni1e7:born_oni0e14:born_on_remotei0e17:check_update_betai0e11:computer_id16:x2OD42yID8PQMrya26:connect_cpu_max_cache_sizei52428800e28:direct_torrent_max_file_sizei10485760e18:diskio_cache_limiti0e3:fgti0e16:history_log_sizei100e18:history_time_limiti30e12:install_timei1629296708e24:is_firstrun_window_showni1e24:is_webui_credentials_seti1e3:liti0e10:lit_remotei0e8:log_sizei100e15:mainWndLocation16:21:net.utp2_initial_ratei10000000e14:net.utp2_limiti0e17:net.utp2_max_lossi50e17:net.utp2_max_ratei125829120e17:net.utp2_min_ratei262144e23:net.utp2_rtt_correctioni10000e16:offer_last_showni0e7:peer_id20: 25:peer_max_active_downloadsi500e9:proxy.p2pi1e10:proxy.porti0e2:rti0e8:selfcert1803:
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
-----BEGIN PRIVATE KEY-----
...
-----END PRIVATE KEY-----
22:settings_saved_systimei1629296826e2:tdi0e2:tui0e10:ui_versioni0e14:webui.hashword12:webui.listen12:0.0.0.0:888810:webui.salt44:xx=eroot@
DietPi3:/var/lib/resilio-sync#
```

w1v1, 2021-08-18:
The only issue may be that users forget that it runs as rslsync user, not dietpi - so the folders it has access to would also need to have the same or at least in the same group or something. It may require the user group change so users can use their storage devices (usually as group dietpi)? 

The transfer speed is far superior to other alternatives and the user-interface is very intuitively simple when compared. Its set and forget in function.

MichaIng, 2021-08-18:
For such cases, usually we change the primary group to `dietpi` and umask to `002`, so the service has write access to media files in `/mnt/dietpi_userdata`, downloaded via BitTorrent, added by file servers or other media-related software. And the other way round, files created by rslsync with `dietpi` group and 775/664 modes are automatically accessible by file serves and media-related software.

Joulinar, 2021-08-18:
> The only issue may be that users forget that it runs as rslsync user and group, not dietpi - so the folders it has access to would also need to have the same or at least in the same group or something. It may require the user group change so users can use their storage devices (usually as group dietpi)?

I don't think we will change away from running it as user `rslsync` as we moving away from using user `dietpi` to more application dedicated users. But user group `dietpi` will stay. This we usually use to share data between different apps. Therefore my statement above: add user to dietpi group

BTW: do you use it as free personal edition? Because it's as well available as PRO version

w1v1, 2021-08-19:
> BTW: do you use it as free personal edition? Because it's as well available as PRO version

The personal edition has all the same basic functions as Syncthing, except simpler, faster and far better GUI with the optional 'Pro' upgrade for extra features like selective file sync (to save space), allowing limited and selective folder/file sharing with anyone, and probably a few other features (which Syncthing does not have)