MichaIng/DietPi #4655: Disk problem afther using and stopping usb disk 8TB!

cocoflan, 2021-08-18:
Very Strange problem, My 8TB disk is lost partition and so all my data, after i stopped a disk check with "badblocks -b 4096 -c 1 /dev/hdb1"

I am now checking if i can get the lost partition back, hope i will.

Also in my SMARTinfo i see Time in overtemperature 515 minutes, so also strange. No errors for the rest in SMART.

Greetings

MichaIng, 2021-08-19:
You are aware that a bad blocks check basically writes every single byte of the disk and reads back the value, optionally multiple times? So it is a format inherently. If it would spare the partition table and metadata or any other actual files, it wouldn't be able to check those parts for bad blocks.

Now this tool can be used on a partition as well, due to the nature that in Linux drives and partitions are accessible as block devices the same way, but this means you only check part of the drive, and as well the containing data is erased.

As this is a constant read/write cycle, I think first it writes only, and then it reads everything back, it also makes sense that this causes a much higher load then usual and hence possibly the over temperature, when not made for 24/7 I/O: Usual drives for personal use are not even made for being 24/7 powered, but special drives like WD RED are made for this, which I would always recommend for any server use at least. But also those are not made for 24/7 I/O, hence e.g. shouldn't be used with e.g. a camera which writes the constant video stream to disk. There are again higher quality drives specifically made for this kind of usage. Now when you do a bad blocks test of the whole drive, with lets say 150 MiB/s, then in theory it takes 15.5 hours to write the whole disk, which is far exceeding what a common drive is produced for.

The most important question is why you do this test in the first place? I did such once on a 64 GiB SD card after I got heavy filesystem corruption two times, but knowing well that the test itself can give it the last death strike. So without some evidence and e.g. the idea to return it as a warranty case to the vendor/manufacturer, I cannot think of many cases where it would make sense to run such a test, especially not on relatively large but slow HDDs 🙂.

cocoflan, 2021-08-19:
yes my mistake, I know, thanks for the clear explanation. 
Won't happen to me a 2nd time anymore, luckily I have backups of everything that was on the disk.

MichaIng, 2021-08-19:
> luckily I have backups of everything that was on the disk

Okay, I'm glad for that 🙂. I mean common recovery tools could have been tried as well, but after 515 minutes, I guess only a minor part of files would have been recoverable.

MichaIng, 2021-11-07:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-07 by MichaIng