MichaIng/DietPi #4656: Can't install LXQt on bullseye

LieDanG, 2021-08-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`Bullseye`
- Kernel version | `uname -a`
`Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi4`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`Lxqt`
- Was the software title installed freshly or updated/migrated?
`Fresh install`
- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

#### Steps to reproduce

1. Install lxqt from software list on bullseye image
2. Fail

#### Expected behaviour

- ...

#### Actual behaviour

- Cant find https://github.com/MichaIng/DietPi/blob/master/.conf/desktop/lxqt/lxqt-bullseye.7z since it doesn't exist

#### Extra details

- ...

Joulinar, 2021-08-18:
Hi,

Pls can you share the whole output from SSH terminal. Thx

LieDanG, 2021-08-18:
```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring LXQt: lightweight desktop

[FAILED] DietPi-Software | Checking URL: https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullseye.7z
 - Command: curl -ILfvm 10 https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullseye.7z
RPi 4 Model B (aarch64) | IP: 192.168.1.200 | Use up/down buttons to scroll text
│ Checking URL:
│ https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullse
│ ye.7z
│  - Command: curl -ILfvm 10
│ https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullse
│ ye.7z
│  - Exit code: 22
│  - DietPi version: v7.4.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 |
│ DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: From scratch
│  - Error log:
│   % Total    % Received % Xferd  Average Speed   Time    Time     Time
│ Current
                                  Dload  Upload   Total   Spent    Left
│ Speed
│ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--
│ 0*   Trying 140.82.112.4:443...
│ * Connected to github.com (140.82.112.4) port 443 (#0)
│ * ALPN, offering h2
│ * ALPN, offering http/1.1
│ * successfully set certificate verify locations:
│ *  CAfile: /etc/ssl/certs/ca-certificates.crt
│ *  CApath: /etc/ssl/certs
│ } [5 bytes data]
│ * TLSv1.3 (OUT), TLS handshake, Client hello (1):
│ } [512 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Server hello (2):
│ { [122 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
│ { [19 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Certificate (11):
│ { [2363 bytes data]
│ * TLSv1.3 (IN), TLS handshake, CERT verify (15):
│ { [80 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Finished (20):
│ { [36 bytes data]
│ * TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
│ } [1 bytes data]
│ * TLSv1.3 (OUT), TLS handshake, Finished (20):
│ } [36 bytes data]
│ * SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
│ * ALPN, server accepted to use h2
 * Server certificate:```

│ *  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.;
│ CN=github.com
│ *  start date: Mar 25 00:00:00 2021 GMT
│ *  expire date: Mar 30 23:59:59 2022 GMT
│ *  subjectAltName: host "github.com" matched cert's "github.com"
│ *  issuer: C=US; O=DigiCert, Inc.; CN=DigiCert High Assurance TLS Hybrid ECC
│ SHA256 2020 CA1
│ *  SSL certificate verify ok.
│ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--
│ 0* Using HTTP2, server supports multi-use
│ * Connection state changed (HTTP/2 confirmed)
│ * Copying HTTP/2 data in stream buffer to connection buffer after upgrade:
│ len=0
│ } [5 bytes data]
│ * Using Stream ID: 1 (easy handle 0x55a4af2f90)
│ } [5 bytes data]
│ > HEAD /MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullseye.7z
│ HTTP/2^M
│ > Host: github.com^M
│ > user-agent: curl/7.74.0^M
│ > accept: */*^M
│ > ^M
│ { [5 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
│ { [57 bytes data]
│ * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
│ { [57 bytes data]
│ * old SSL session ID is stale, removing
│ { [5 bytes data]
 * Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
│ } [5 bytes data]
│ * The requested URL returned error: 404
│ * stopped the pause stream!
│ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--
│ 0
│ * Connection #0 to host github.com left intact
│ curl: (22) The requested URL returned error: 404
```

Joulinar, 2021-08-18:
thx for sharing. Maybe it can be included in next release that should become available on upcoming weekend.

Joulinar, 2021-08-18:
a possible workaround until we have a final fixe.

Once you are hit by the issue, you should have a very small menu at the bottom. There you should be able to select `change command` and you should be able to replace `bullseye` with `buster`. 

Don't get confused, it would need to be done twice

**1st**
from `curl -ILfvm 10 https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullseye.7z`
into `curl -ILfvm 10 https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-buster.7z`

**2nd**
from `curl -sSfL https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-bullseye.7z -o lxqt-bullseye.7z`
into `curl -sSfL https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-buster.7z -o lxqt-bullseye.7z`

MichaIng, 2021-08-19:
Solved. One thing I recognised, when starting one of our terminal tools from a root user desktop, the whiptail initially has a minimal height, so that most text and menu is not shown. When navigating to a submenu or "Exit" > "Cancel", it is aligned correctly to the terminal size. It seems like the qterminal is started initially with minimal dimensions, which are then "quickly" changed to the last stored (or hardcoded, when settings changed) dimensions. While this is done quick enough for my eyes, it is too slow for our function, which estimates the terminal size to apply whiptail dimensions. E.g. in case of dietpi-software it works, as it does quite some processing before the menu is shown. Not sure how to solve, essentially it looks like a bug in qterminal to me, which should not start to execute the passed command before all its settings have been applied.

closed 2021-08-20 by MichaIng