MichaIng/DietPi #4658: Dietpi-Wifi configuration is not working as intended

taichikuji, 2021-08-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`bullseye`
- Kernel version | `uname -a`
`Linux DietPi 5.10.52-v8+ #1441 SPM PREEMPT Tue Aug 3 10:14:03 BST 2021 aarch64 GNU/LINUX`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V 3A`
- SDcard used | (EG: SanDisk ultra)
`Samsung EVO Select micro SDXC UHS-I U1`
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`Home Assistant`
- Was the software title installed freshly or updated/migrated?
`The entire OS was installed freshly`
- Can this issue be replicated on a fresh installation of DietPi?
`You can`
- Bug report ID | `echo $G_HW_UUID`
`I have none`

#### Steps to reproduce

1. Flash DietPi v8 as usual
2. Configure the Dietpi.txt and Dietpi-Wifi.txt configuration files to match what I require. This includes Wifi configurations with my router that has a WPA2-PSK password, including characters that could be problematic. 
Example of the password used in the Dietpi-Wifi.txt:
```
Di\X)63JWpp,HPp7F{`X
```
3. The WiFi SSID and password do appear correctly in the configuration inside Dietpi-config, going through the network settings, but refuses to connect.

#### Expected behaviour

- I expect the WiFi to connect correctly, regardless of the characters used as a password, since every other server / device I own and use do work.

#### Actual behaviour

- WiFi credentials are saved correctly, but isn't able to connect. Neither with DHCP or STATIC IP configurations.

#### Extra details

- I believe the issue has to do with the special characters like ``` \ , { and ` ``` but I do not think these characters should / have to be avoided to succesfully connect.

MichaIng, 2021-08-19:
Many thanks for your report.

Could you try to remove those characters, probably narrow it down to a specific one? I suspect the ` tick to be the culprit, but not 100% sure currently. For the shell script part, the string is wrapped into regular single quotes `'`, which should make all characters within literal, except of course other single quotes, which is the reason why we escape them automatically: https://github.com/MichaIng/DietPi/blob/8b15bdf/dietpi/func/dietpi-wifidb#L172

But it is applied to the wpasupplicant configuration, and I am not sure whether this has limitations, and whether escaping is possible there. We wrap it into double quotes there, and I remember that a contained double quote character does not cause issues, as the parser uses the first and the last double quote character and takes intermediate ones literal: https://github.com/MichaIng/DietPi/blob/8b15bdf/dietpi/func/dietpi-wifidb#L114

MichaIng, 2021-08-19:
Ah, and to check whether it is in our script or a limitation of wpasupplicant, could you check whether the password is correct in `/etc/wpa_supplicant/wpa_supplicant.conf`, between double quotes?

taichikuji, 2021-08-19:
Sure! I will test changing the password with an example with each character and let you know the results ASAP. Thank you very much for your quick response!

taichikuji, 2021-08-19:
> Ah, and to check whether it is in our script or a limitation of wpasupplicant, could you check whether the password is correct in `/etc/wpa_supplicant/wpa_supplicant.conf`, between double quotes?

Checked and it appears the same way I type it in the configuration file. None of these symbols are ignored or the password is shown differently.

I tried removing all of them (leaving only letters and numbers), I'm still having no luck, it seems like it struggles connecting through DHCP, doesn't give me any DHCP leases. I thought maybe that was the issue but I still don't think so. It still gives me issues using an static IP.

Honestly kinda confused, nothing seems to work right now.

One thing I should point out is that I'm currently using the WiFi board chip included in the RPi 4, but I have multiple USB WiFi dongles that I tried with no luck either.

MichaIng, 2021-08-20:
So it is not the password/key. What issues does it give you when using a static IP (which I would anyway always use for a server)?

Does the router/DHCP server show any logs or list the Pi as LAN member at all?

And can you show the full output of the interface configuration service:
```sh
journalctl -u ifup@wlan0
```

taichikuji, 2021-08-28:
I've tried disconnecting the USB3.0 devices ( it was a mass storage SSD device ). It didn't fix it.

I've done multiple tests today through Ethernet and Wi-Fi and I found out about something really weird that happened as well. Essentially it didn't connect to the Wi-Fi until I connected the Ethernet cable. And the Ethernet setting was disabled through the settings.

I've gotten it to work now somehow, I think there's nothing to worry about, and it must've been a problem on my end. Thank you very much however for your help! I will close the issue now.

closed 2021-08-28 by taichikuji