MichaIng/DietPi #4661: mkinitramfs does not work - missing firmware/drivers

timoteicampian, 2021-08-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
 ```
 root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='applied'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`11.0`
- Kernel version | `uname -a`
`Linux DietPi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra): `SanDisk ultra)`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated? `new installation`
- Can this issue be replicated on a fresh installation of DietPi? `yes`
 `8da0099c-0ca7-454c-9013-87b6f03c31d7`
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. fresh install
2. ` mkinitramfs -o /boot/initrd`
```
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156b-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156a-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153c-1.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
dropbear: WARNING: Invalid authorized_keys file, SSH login to initramfs won't work!
```

#### Expected behaviour

- the same task worked well in previous version (DietPi_RPi-ARMv6-Buster)

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-08-20:
Generating an initramfs takes a while. What you see are not errors but warnings, which are expected, so give it some more time.

Btw, why do you install an initramfs in the first place? Raspberry Pi does not require it to boot. And you seem to have `dropbear-initramfs` installed, but a warning that it won't work due to invalid `authorized_keys`, i.e. no valid public authentication key is present.

timoteicampian, 2021-08-20:
many thanks for your reply!

I have fixed the issue with `dropbear-initramfs`, by sharing the `authorized_keys` of `OpenSSH`; both have to operate in parallel on different ports for different purposes.

`initramfs` is used for some pre-boot actions (decrypting some custom images, mounting them in an `overlayfs`, `pivot_root` + `chroot` into a custom `init`, etc.)
but after installing `initramfs` and reboot, I have no wifi/network access (`ping, ssh`) anymore (not tested yet with ethernet cable).

any suggestions?

MichaIng, 2021-08-20:
Okay, makes sense.

The initramfs shouldn't have any effect on the later system, network or so, as long as within that phase there are no steps done which do make a difference.

Please check the logs of the interface config service. When it's onboard WiFi, the following should be it:
```sh
journalctl -u ifup@wlan0
```