MichaIng/DietPi #4663: error on NanoPi

giusCB, 2021-08-19:
#### Details:
- Date           | Thu Aug 19 18:45:22 CEST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | NanoPi NEO (armv7l) (ID=60)
- Kernel version | `Linux DietPi 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `apt-get --allow-releaseinfo-change-suite -q update`
- Exit code      | 100
- Software title | DietPi-Pre-patches
#### Steps to reproduce:
got error on a first boot

#### Additional logs:
```
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Ign:5 https://apt.armbian.com buster InRelease
Err:6 https://apt.armbian.com buster Release
  Redirection from https to 'http://armbian.hosthatch.com/apt/dists/buster/Release' is forbidden [IP: 173.45.128.12 443]
Reading package lists...
E: The repository 'https://apt.armbian.com buster Release' does not have a Release file.
```

Joulinar, 2021-08-19:
Hi,

I guess you opened same issue on our Forum? https://dietpi.com/phpbb/viewtopic.php?p=36960#p36960

If yes, pls don't do that. It's not going to speed up thinks, just creating additional overhead to manage both. Thx for understanding.

Let's continue on this for now within GitHub.

giusCB, 2021-08-19:
Ok, very sorry  giovedì,

Joulinar, 2021-08-19:
probably it's similar to this issue. https://github.com/MichaIng/DietPi/issues/4600

It was an issue outside DietPi and according @MichaIng it should have been fixed by Armbian 🤔

giusCB, 2021-08-19:
solved in this way: select subshell from menu, edit /etc/apt/sources.list.d/armbian.list , change from https to http
I hope it will be useful to someone in the future.

MichaIng, 2021-08-19:
Very strange. This should actually not happen anymore after [this](https://github.com/armbian/dl-router/pull/14) change in the mirror router, after which the request scheme HTTPS is reserved.

MichaIng, 2021-08-19:
Okay, I see this is a new mirror, and I guess it has been added to the mirror list with explicit plain HTTP scheme. Since it supports HTTPS as well, removing the scheme will solve the issue. I just asked the Armbian dev about it, should be fixed quickly.

MichaIng, 2021-08-19:
Okay it should work again with HTTPS now 🙂.

COUPON4Vitamins, 2021-08-22:
> solved in this way: select subshell from menu, edit /etc/apt/sources.list.d/armbian.list , change from https to http
> I hope it will be useful to someone in the future.

Thanks, got the exact problem and that solved it for me, too.

MichaIng, 2021-08-22:
> got the exact problem

After I posted that is has been fixed server-side two days ago??

closed 2021-09-01 by MichaIng