MichaIng/DietPi #4664: Bug report ID | `20ad25a9-ffd6-4c00-8262-6d02f8402963'

INxYOxBASE, 2021-08-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | Downloaded on 18th August 2021
- Distro version | I don't know what this means
- Kernel version | Unknown
- SBC model | RPi 4 4gb
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | Kingston SDCS2/32GB

#### Additional Information (if applicable)
- Software title | Plex, Blynk, Unbound and PiHole
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `20ad25a9-ffd6-4c00-8262-6d02f8402963'

#### Steps to reproduce

1. Tried to install PiHole, Unbound, Blynk and Plex concurrently on first time setup. Got stuck on the install step for installing Java

#### Expected behaviour

- Setup completes

#### Actual behaviour

- Setup does not complete

#### Extra details

- Here is the printout from terminal that was in blue I was instructed to leave here:

#### Details:
- Date           | Fri Aug 20 10:52:21 NZST 2021
- Bug report     | 20ad25a9-ffd6-4c00-8262-6d02f8402963
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux Pi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install ca-certificates-java openjdk-17-jre-headless openjdk-17-jdk-headless`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to locate package openjdk-17-jre-headless
E: Unable to locate package openjdk-17-jdk-headless
```

INxYOxBASE, 2021-08-19:
I feel it best that if anyone sees this, I am completely green to any sort of coding. I picked up this RPi for cheap at a sale and thought it would be neat to block some ads. I am quickly realising I may be out of my depth. Many thanks for any help, I will continue trying things and update here if anything changes. My next move will be trying to install each of the softwares listed individually and seeing if that changes things.

Thanks

INxYOxBASE, 2021-08-20:
Hi, so using the method of installing each piece of software separately I have now successfully installed Plex, PiHole and Unbound. Attempting to install Blynk gives me the same error as above, most notably this line:

[FAILED] DietPi-Software | APT install for: ca-certificates-java openjdk-17-jre-headless openjdk-17-jdk-headless

I'm unsure if this is because I am running my Pi as a 'headless' setup, that seems to be in that line of code but I may be way off.

With PiHole, Unbound and Plex all installed I'm sure I'll have my work set out for me setting those up to work but will most likely attempt to get Blynk up and running at some point in the future to help run some of the IOT ESP32s I have laying around.

Again, many thanks for any help. Cheers

ravenclaw900, 2021-08-20:
Hello,
What does your `apt.sources.list` file look like?
```
cat /etc/apt/apt.sources.list
```

INxYOxBASE, 2021-08-20:
Hi ravenclaw900,

I ran that command and this is what it returned. I hope this is what you are asking for!

``
root@Pi:~# cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free
``

Joulinar, 2021-08-20:
Hi, pls can you share following output

`apt update && apt install ca-certificates-java openjdk-17-jre-headless openjdk-17-jdk-headless`

INxYOxBASE, 2021-08-20:
Hi Joulinar,

Have just arrived at work for the day so won't be able to share for the next 8hrs sorry. Will get comment here when I'm able to answer. Cheers

INxYOxBASE, 2021-08-20:
Hi Joulinar,

Here is what happened after running that command:

```
root@Pi:~# apt update && apt install ca-certificates-java openjdk-17-jre-headless openjdk-17-jdk-headless
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease                                                   
Hit:3 https://archive.raspberrypi.org/debian bullseye InRelease                 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package openjdk-17-jre-headless
E: Unable to locate package openjdk-17-jdk-headless
root@Pi:~# 
```

Not much of this makes sense to me so hopefully you're able to decode it! Cheers

Joulinar, 2021-08-20:
There is nothing to decode. Simply Raspbian Bullseye did not ship Java17 until now. For now you could wait until RPi devs are going to deliver the missing package or you use the ARMv7 image (instead of ARMv6)

@MichaIng 
Maybe you have another idea

INxYOxBASE, 2021-08-20:
Great, thank you both very much for your help. I think I will wait and continue setting up the other pieces of software in the meantime.

Much appreciated.

closed 2021-08-20 by INxYOxBASE

MichaIng, 2021-08-20:
Solved for v7.5 already with #4593.

Workaround:
```sh
sed -i 's/version=17/version=11/' /boot/dietpi/dietpi-software
```