MichaIng/DietPi #4666: proftpd fails to start with default settings

twikedk, 2021-08-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.4.2
- Distro version | 11.0 (bullseye)
- Kernel version | Linux ServerII 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)
- Power supply used | 5V 2.5A
- SDcard used | Samsung

The installation of proftpd was done on a complete fresh installed bullseye too.

When starting, proftpd fails and in systemd status log I see the following:
```
Aug 20 19:15:47 ServerII systemd[1]: Starting ProFTPD FTP Server...
│ Aug 20 19:15:47 ServerII proftpd[1538]: Checking syntax of configuration file
│ Aug 20 19:15:47 ServerII proftpd[1538]: 2021-08-20 19:15:47,399 ServerII proftpd[1538]: fatal: unknown configuration directive 'IdentLookups' on line 12 of '/etc/proftpd/proftpd.conf'                                                   │ 
│ Aug 20 19:15:47 ServerII systemd[1]: proftpd.service: Control process exited, code=exited, status=1/FAILURE
│ Aug 20 19:15:47 ServerII systemd[1]: proftpd.service: Failed with result 'exit-code'.
│ Aug 20 19:15:47 ServerII systemd[1]: Failed to start ProFTPD FTP Server.
```
I found that I can change the following in **modules.conf**
`#LoadModule mod_ident.c`
to
`LoadModule mod_ident.c`
and proftpd will start

or I can, in **proftpd.conf**, add the **IfModule** part as it also is in the default conf file
```
<IfModule mod_ident.c>
  IdentLookups off
</IfModule>
```
and this also makes me able to start proftpd again.

Maybe the solution is a combination of changes in **modules.conf** and **proftpd.conf**?

MichaIng, 2021-08-20:
Many thanks for your report.

If this is basically a reverse proxy lookup, I'd leave it disabled by default, wrap it into the IfModule tag as you suggested and add a little hint that it may require the adjustment in modules.conf.

twikedk, 2021-08-21:
Oh maybe I should learn to read the comments in the conf file, it does say load module when IdentLookups is set to on. :-)
I have changed my setup to follow your suggestion.

twikedk, 2021-08-21:
Sorry I shouldn't have closed it, saw that it had the **Investigation required** label.

Joulinar, 2021-08-21:
I guess the issue is clear. 

@MichaIng are you going to create an adjusted config file for Bullseye?

MichaIng, 2021-08-21:
The module does not exist on Buster's ProFTPD version, so the module check cannot be used. I checked a bit what this feature does, and indeed it seems similar to a reverse DNS lookup to show a nice username in the logs. We'll have it disabled by default and leave a hint in the config file.

MichaIng, 2021-08-21:
... ah lol, these lookups are disabled already, so it doesn't even change something. And not turning it on fails, but the config itself is not understood. Okay as we want to have it disabled then, we need to keep the config for Stretch and Buster as it is and only adjust  it on Bullseye. Migration to a smaller `/etc/proftpd/conf.d/dietpi.conf` override will be done in a later release.

MichaIng, 2021-08-21:
Quick and dirty fix: https://github.com/MichaIng/DietPi/commit/a8c17c115ab2f78b4bde320b1d60dd83f1e32a79
Changelog: https://github.com/MichaIng/DietPi/commit/91e71a5aacca1c2b0e778187892a7b41dd648b90

closed 2021-08-22 by MichaIng