MichaIng/DietPi #4670: Mount HDDs on VMs

Phil1988, 2021-08-21:
Hey,

<details>
  <summary>Reson for my situation</summary>
  
I have the problem, that the HDD that contains my VM is going to die and looses/breaks data already.

The VM can't be copied (or with tricks it can, but it cant be mounted there)
After 3 days restoring I am now able to boot and use the VM and everything works (now sure how long).

So as the VM is not backup-able I would like to perform a dietpi backup.
As far as I understand this backups everything, like files, databases etc.
So in theory I should be able to start an new VM (on a different HDD of course) and then restore everything.
  
</details>

I am using a VMware VM and want to perform a full backup to a host system attached NTFS HDD.
 
I did try to share
```
apt-get install ntfs-3g -y
mkdir ~/Dietpi_Share
sudo mount -t fuse.vmhgfs-fuse -o allow_other .host:/Dietpi_Share ~/Dietpi_Share
```

But the Drove Manager cant fint the drive:
[image]

What would be the best way to do this in my case?

MichaIng, 2021-08-21:
DietPi-Drive_Manager cannot handle all types of drives, especially not LVM volumes and similar. I guess this is a VMware implementation to mount a physical drive from the host inside of the guest? It has no ability to make this visible like a regular physical drive inside the guest? I know such from VirtualBox where passed through USB drives are least are seen like physical ones, but the shared folder feature provides drives of type "vboxsf", which is explicitly excluded by drive manager, until someone finds the time to implement proper detection and fstab entries for those.

Can you show the following output (after manually mounting that drive):
```
df -T
lsblk # I guess it is not visible here
```

So but as the filesystem is NTFS, you cannot do a system backup to it. Linux system strictly rely on UNIX permissions/modes and symlinks, which both is not supported on NTFS. The ntfs-3g driver emulates UNIX permissions, but I'm not sure if those can reliably be restored on a different system, and symlinks cannot be created at all, AFAIK. Easiest IMO is to attach and pass through a USB drive, so it can be seen like a regular physical drive in the VM, then format is with ext4 (or another native UNIX filesystem type) and do the backup there.

Phil1988, 2021-08-21:
Im not too sure about your questions.
In the past I used raspbian and was able to mount it to /dev/sdb etc... but on Debian or at least at DietPi I'm not sucessfull on installing VMware Tools on the guest system.
It says its installed but it seems that its doesnt have the full functionallty.

The way I did it (above) creates a mounting point
/root/Dietpi_Share/

I can write files to it via Windows (host) and see it on DietPi (guest) and vice versa.

Here is your requested output:
```
root@DietPi:~# df -T
Dateisystem    Typ               1K-Blöcke   Benutzt  Verfügbar Verw% Eingehängt auf
udev           devtmpfs            1990732         0    1990732    0% /dev
tmpfs          tmpfs                399380     10640     388740    3% /run
/dev/sda1      ext4             1479579000 748681380  670678112   53% /
tmpfs          tmpfs               1996888       744    1996144    1% /dev/shm
tmpfs          tmpfs                  5120         0       5120    0% /run/lock
tmpfs          tmpfs               4929536      2652    4926884    1% /tmp
tmpfs          tmpfs                 51200        52      51148    1% /var/log
vmhgfs-fuse    fuse.vmhgfs-fuse 7813991416    754912 7813236504    1% /root/Dietpi_Share
```

```
root@DietPi:~# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0  1,6T  0 disk
└─sda1   8:1    0  1,4T  0 part /
sr0     11:0    1 1024M  0 rom
```

Currently I have a 8TB WD Elements connected via USB 3.0 to the PC.
There is no data on it and I am only trying to find a way to do a full backup that is accessable via Windows.
My goal is to do the backup and store it somewhere on my (NTFS) Raid to have a less likely failing backup somewhere.

For now, it would also be enough to save it somewhere and somehow to have a backup if the HDD finally dies...
which actually looks like it will not take too long :(

Phil1988, 2021-08-21:
So I did add a hard drive by attaching it via VMware (by creating that *.vmdk file).
I did only make it 2TB.

After rebooting I found it
```
root@DietPi:~# sudo fdisk -l
Disk /dev/sda: 1,56 TiB, 1717986918400 bytes, 3355443200 sectors
Disk model: VMware Virtual S
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3a242985

Device     Boot Start        End    Sectors  Size Id Type
/dev/sda1        2048 3006479159 3006477112  1,4T 83 Linux

Disk /dev/sdb: 1,95 TiB, 2147483648000 bytes, 4194304000 sectors
Disk model: VMware Virtual S
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```

It can also be found by the Drive Manager:
[image]

So I formatted it:
[image]

[image]
[image]

I then started the DietPi-Backup and selected this drive:
[image]
[image]

And started the backup:
[image]

This way I do at least have something just in case...
I know its normally not good to start stressing, but the drive really looks bad on sector checks, it sound louder and it gets warmer then normally, so I want to have at least one backup asap :D

Phil1988, 2021-08-21:
session aborted due to router restart....

[image]

I think there is no way to resume this session... 

So I will wait until no more HDD action is going on and hope everything worked :(

MichaIng, 2021-08-22:
Btw, as of your intermediate issue with the router restart: Since `rsync` does an incremental sync, hence also actually transfers files which have changed, it is not a big issue to have it interrupted. It practically continues at the point of interruption, i.e. the already existing files are skipped quickly.

Yes it is indeed a benefit of FAT/exFAT/NTFS that you can plug and repair them on Windows, I agree. I'm basically pissed that Windows still has no native drivers for UNIX filesystems, and every 3rd party solution is very buggy, at least my experiences when testing them where a disappointment. I have an old notebook and some VMs running DietPi for testing and repairing Linux filesystems, but most users have their RPi as only Linux system. So there is currently no perfect solution possible.

Joulinar, 2021-08-22:
well it's on the way, somehow. Just get a Windows build higher than 20211 😉 

https://devblogs.microsoft.com/commandline/access-linux-filesystems-in-windows-and-wsl-2/

MichaIng, 2021-08-22:
But running a WSL2 instance only to access UNIX filesystems is also not easier than running a VM. Only a native Windows integration of those filesystems would really make it easier for novice users: Plug a USB stick with an `ext4` to your Windows machine and have the mounted `ext4` pop up in your explorer, just like it would do when attaching an NTFS/FAT/exFAT filesystem. As fast as opening a command line terminal is required, 90% of users will feeling uncomfortable.

Joulinar, 2021-08-22:
I agree it's not perfect. I guess it's still WIP but it indicates that maybe not sooner but later we will have something more casual than using WSL2

Phil1988, 2021-08-22:
I am still waiting for the backup to finish.

I have lots of these errors:
[image]
[image]

They are all only applying to the "DietPi_VMware-x86_64-Stretch-s002.vmdk" -file (which was by the way also the file that could not normally be copied to another HDD.

As I always have to hit "continue" it takes more time.... and also I am not sure if the backup will be usable anyways when completed.

MichaIng, 2021-08-22:
The constant "critical medium error" don't look good. Do those show up all the time or only before prompt to continue? If this virtual disk caused issues already, I'd not copy anything to it anymore but create a new one and copy things over. Not sure how such a vmdk can get broken that way, I guess a `fsck` doesn't help, does it?

Phil1988, 2021-08-22:
Just a short feedback after almost 2 days backuping:
[image]

The critical medium error always showed up after that prompt.

Is there anywhere a guide how to copy things over?

My idea would be to:
1. create another DietPi VM
2. transfer files from the current (broken) VM to the new one via FTP (this will be tricky because I'm not sure where the files are sitting except of /mnt/dietpi_userdata/). Especially all the config files are problematic to find eg. to forward everything to https etc.
3. backup the mysql database via phpMyAdmin and insert it somehow to the new one...but I guess that the new VM will setup its own databases with its own names, which makes it harder to insert the old database.

I did not use fsck yet.
In fact I cut every sync and data copy to it as much as I can to prevent the VM to be unusable again (I needed a couple of days already to be able to start it)...

MichaIng, 2021-08-23:
Is the faulty vmdk the origin drive you're syncing from or the target drive you're syncing to? While the backup as a whole failed, at least rsync should have copied all it was able to copy. If the faulty drive was the origin (which I guess), then it means a bunch of files were rescued already, which is a good thing.

1. 👍 
2. Do not use network transfer for this amount of data, when you can copy it locally. Attach the old/broken vmdk as local drive to the new VM and use `dietpi-sync` or `rsync` to copy files over. `/mnt/dietpi_userdata` is a good start, also check `/etc`, `/var/lib` and `/root` or `/home/<user>` for specific config and data, but do not copy something over where you're not sure that you applied custom configs or created data with that software title/package. Keep the old drive for a while, so you can copy further things, if you recognise that something is missing.
3. `/mnt/dietpi_userdata/mysql` is migrated by `dietpi-software` when installing MariaDB. But it's still a good idea to create an SQL dump/backup, in case due to incompatible versions or corruption the migration fails.

> Do you think it can not break the VM?

`fsck` does not break anything which was not already broken before, but it may reveal errors which were not recognised before. As long as `rsync` even fails to read/sync from a drive, I think `fsck` is the change to repair/recover broken files to allow syncing them. If metadata was corrupted, e.g. the filesystem link lost, files and in the drives `lost+found` directory (on ext4), so its also worth to have a look there, when something is missing. In case of text files, the content may reveal which one it is/where it belongs to.

Phil1988, 2021-08-23:
The faulty vmdk is the source/origin file.
As far as I know there have been a lot of files copied, but have they been deleted, after the backup failed?
Are these files anyhow usable for a backup if the HDD completely dies now?

Do understand 2. & 3. correctly, that if I use dietpi-sync to transfer the data from `/mnt/dietpi_userdata/mysql` and install MariaDB afterwards, the mySQL database will be the "restored" (best case) ?

I will try that and report back.
Thanks for your help Micha!

MichaIng, 2021-08-23:
> but have they been deleted, after the backup failed?

They shouldn't be deleted, if they were synced already.

> Are these files anyhow usable for a backup if the HDD completely dies now?

Everything that was successfully synced should be fine. In case of the not synced files, hard to say.

> Do understand 2. & 3. correctly, that if I use dietpi-sync to transfer the data from `/mnt/dietpi_userdata/mysql` and install MariaDB afterwards, the mySQL database will be the "restored" (best case) ?

Yes exactly. Generally MariaDB checks its data directory on install, and does a migration, if required, to take over the data as is. As said, a `mysqldump` is still good to have additionally, as it is the more generic SQL format which can be imported regardless of the MariaDB version without any migration.

MichaIng, 2021-11-07:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-07 by MichaIng