MichaIng/DietPi #4672: VNC won't show desktop when HDMI port is unplugged

K92Pi, 2021-08-21:
Hi there! 
I have a constant issue with Dietpi. Currently I'm  using Dietpi 7.4.2 and I'm  running it on a Raspberry Pi 4. I've installed RealVNC server on my Pi and client on my Mac. 
I didn't change anything on the server, it is as is. I connect using an IP address, I didn't specify the port number, however I've tried to connect with the port 5900. It works the same way. I've set the resolution to 1280x720. Every time I disconnect the HDMI cable, the VNC won't show the desktop. I've even reinstalled the whole system, but it doesn't fix this problem. Any suggestions for what should I try to resolve this issue?

Joulinar, 2021-08-21:
Did you try using port 5901?

As well you could check listen ports as follow

`ss -tulpn | grep LISTEN`

MichaIng, 2021-08-21:
The port should be the default VNC port, however, when connecting to a virtual desktop, assure to use screen `:1` instead of the shared desktop at screen `:0`.

But actually, as long as there is an actual X11 desktop session running, the VNC servers should be able to connect to it as well, despite a HDMI cable being attached or now, but I'm not sure whether the RPi collapses some video features which breaks the X11 server, when unplugging it. Would be interesting what happens if you have a screen attached, open a regular desktop session, connect via VNC and then unplug the screen: Does the VNC session break or stay functional? If it breaks, what does the following show?
```sh
journalctl -u vncserver-x11-serviced
```

However, if you do not aim to have a co-session on real screen + VNC, but only want a remote desktop, I'd simply use the virtual X11 session at `:1` anyway. This also allows you to disable desktop auto-login (`dietpi-autostart`) and prevents two concurrent X11 sessions running all the time (the real one and the virtual one).

K92Pi, 2021-08-21:
> Did you try using port 5901?
> 
> As well you could check listen ports as follow
> 
> `ss -tulpn | grep LISTEN`

Yes, I've tried port 5901, it's not working. Connection was refused. 

Here's my ports:
`root@MediaServer:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1024            0.0.0.0:32400      0.0.0.0:*    users:(("Plex Media Serv",pid=1303,fd=56))
tcp   LISTEN 0      1024            0.0.0.0:80         0.0.0.0:*    users:(("lighttpd",pid=1276,fd=4))        
tcp   LISTEN 0      1024          127.0.0.1:32401      0.0.0.0:*    users:(("Plex Media Serv",pid=1303,fd=57))
tcp   LISTEN 0      32              0.0.0.0:53         0.0.0.0:*    users:(("pihole-FTL",pid=719,fd=5))       
tcp   LISTEN 0      1000            0.0.0.0:22         0.0.0.0:*    users:(("dropbear",pid=834,fd=4))         
tcp   LISTEN 0      256           127.0.0.1:5335       0.0.0.0:*    users:(("unbound",pid=584,fd=4))          
tcp   LISTEN 0      128           127.0.0.1:32600      0.0.0.0:*    users:(("Plex Tuner Serv",pid=1735,fd=10))
tcp   LISTEN 0      256           127.0.0.1:8953       0.0.0.0:*    users:(("unbound",pid=584,fd=5))          
tcp   LISTEN 0      50              0.0.0.0:445        0.0.0.0:*    users:(("smbd",pid=3743,fd=44))           
tcp   LISTEN 0      30     10.xx.xx.xxx%wg0:55487      0.0.0.0:*    users:(("qbittorrent",pid=2298,fd=21))    
tcp   LISTEN 0      30     192.168.1.5%eth0:55487      0.0.0.0:*    users:(("qbittorrent",pid=2298,fd=20))    
tcp   LISTEN 0      30         127.0.0.1%lo:55487      0.0.0.0:*    users:(("qbittorrent",pid=2298,fd=19))    
tcp   LISTEN 0      128           127.0.0.1:45605      0.0.0.0:*    users:(("Plex Script Hos",pid=1675,fd=8)) 
tcp   LISTEN 0      5             127.0.0.1:4711       0.0.0.0:*    users:(("pihole-FTL",pid=719,fd=10))      
tcp   LISTEN 0      50              0.0.0.0:139        0.0.0.0:*    users:(("smbd",pid=3743,fd=45))           
tcp   LISTEN 0      5               0.0.0.0:5900       0.0.0.0:*    users:(("vncserver-x11-c",pid=883,fd=11)) 
tcp   LISTEN 0      1024               [::]:80            [::]:*    users:(("lighttpd",pid=1276,fd=5))        
tcp   LISTEN 0      50                    *:85               *:*    users:(("qbittorrent",pid=2298,fd=39))    
tcp   LISTEN 0      32                 [::]:53            [::]:*    users:(("pihole-FTL",pid=719,fd=7))       
tcp   LISTEN 0      1000               [::]:22            [::]:*    users:(("dropbear",pid=834,fd=5))         
tcp   LISTEN 0      50                 [::]:445           [::]:*    users:(("smbd",pid=3743,fd=42))           
tcp   LISTEN 0      50                 [::]:139           [::]:*    users:(("smbd",pid=3743,fd=43))           
tcp   LISTEN 0      5                  [::]:5900          [::]:*    users:(("vncserver-x11-c",pid=883,fd=10)) 
`

Joulinar, 2021-08-21:
Do you use the HDMI output for any case or should it be a headless system?

K92Pi, 2021-08-21:
> The port should be the default VNC port, however, when connecting to a virtual desktop, assure to use screen `:1` instead of the shared desktop at screen `:0`.
> 
> But actually, as long as there is an actual X11 desktop session running, the VNC servers should be able to connect to it as well, despite a HDMI cable being attached or not, but I'm not sure whether the RPi collapses some video features which breaks the X11 server, when unplugging it. Would be interesting what happens if you have a screen attached, open a regular desktop session, connect via VNC and then unplug the screen: Does the VNC session break or stay functional? If it breaks, what does the following show?
> 
> ```shell
> journalctl -u vncserver-x11-serviced
> ```
> 
> However, if you do not aim to have a co-session on real screen + VNC, but only want a remote desktop, I'd simply use the virtual X11 session at `:1` anyway. This also allows you to disable desktop auto-login (`dietpi-autostart`) and prevents two concurrent X11 sessions running all the time (the real one and the virtual one).

Hey! Than you for your reply! I tried to start the Pi with HDMI cable, disconnect the cable and then connect to VNC. It works! 
Then I restarted the Pi, connected to VNC with IP address only, it says it can't show desktop. Then I tried with IP address + port 1, it says the connection was refused by the computer. Then I tried with IP + port 5900, it can't show the desktop with port 5901, the connection was refused by the computer.  
Then I SSH in to the PI to see the active ports:
root@MediaServer:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1024         0.0.0.0:32400      0.0.0.0:*    users:(("Plex Media Serv",pid=1271,fd=56))
tcp   LISTEN 0      1024         0.0.0.0:80         0.0.0.0:*    users:(("lighttpd",pid=1241,fd=4))        
tcp   LISTEN 0      1024       127.0.0.1:32401      0.0.0.0:*    users:(("Plex Media Serv",pid=1271,fd=57))
tcp   LISTEN 0      32           0.0.0.0:53         0.0.0.0:*    users:(("pihole-FTL",pid=709,fd=5))       
tcp   LISTEN 0      1000         0.0.0.0:22         0.0.0.0:*    users:(("dropbear",pid=859,fd=4))         
tcp   LISTEN 0      256        127.0.0.1:5335       0.0.0.0:*    users:(("unbound",pid=573,fd=4))          
tcp   LISTEN 0      128        127.0.0.1:32600      0.0.0.0:*    users:(("Plex Tuner Serv",pid=1618,fd=10))
tcp   LISTEN 0      256        127.0.0.1:8953       0.0.0.0:*    users:(("unbound",pid=573,fd=5))          
tcp   LISTEN 0      128        127.0.0.1:37533      0.0.0.0:*    users:(("Plex Script Hos",pid=1666,fd=4)) 
tcp   LISTEN 0      50           0.0.0.0:445        0.0.0.0:*    users:(("smbd",pid=945,fd=44))            
tcp   LISTEN 0      5          127.0.0.1:4711       0.0.0.0:*    users:(("pihole-FTL",pid=709,fd=11))      
tcp   LISTEN 0      128        127.0.0.1:39849      0.0.0.0:*    users:(("Plex Script Hos",pid=1574,fd=7)) 
tcp   LISTEN 0      50           0.0.0.0:139        0.0.0.0:*    users:(("smbd",pid=945,fd=45))            
tcp   LISTEN 0      5            0.0.0.0:5900       0.0.0.0:*    users:(("vncserver-x11-c",pid=907,fd=11)) 
tcp   LISTEN 0      1024            [::]:80            [::]:*    users:(("lighttpd",pid=1241,fd=5))        
tcp   LISTEN 0      32              [::]:53            [::]:*    users:(("pihole-FTL",pid=709,fd=7))       
tcp   LISTEN 0      1000            [::]:22            [::]:*    users:(("dropbear",pid=859,fd=5))         
tcp   LISTEN 0      50              [::]:445           [::]:*    users:(("smbd",pid=945,fd=42))            
tcp   LISTEN 0      50              [::]:139           [::]:*    users:(("smbd",pid=945,fd=43))            
tcp   LISTEN 0      5               [::]:5900          [::]:*    users:(("vncserver-x11-c",pid=907,fd=10)) 

It's strange, because the 5900 port has 0.0.0.0:5900 IP address and I can't see the VPN connection (Wireguard) as well.

K92Pi, 2021-08-21:
> Do you use the HDMI output for any case or should it be a headless system?

I designed at a headless system. During the test session I have the PI next to my TV, so I can connect the HDMI cable whenever I want, the only thing bothering is when I restart the PI, the TV changes the screen to the newly connected device...

Joulinar, 2021-08-21:
Wireguard VPN is using UDP and not shown this way. 

IP 0.0.0.0 means listen on all interfaces and it should be fine.

K92Pi, 2021-08-21:
Okay. Got it. I thought the VPN firewall blocks the connection. 
Here's the status of the Wireguard client:
interface: wg0
  public key: xxx
  private key: (hidden)
  listening port: 51820
  fwmark: 0xca6c

peer: xxx
  endpoint: xxx.xx.xxx.xxx:1443
  allowed ips: 0.0.0.0/0
  latest handshake: 1 minute, 42 seconds ago
  transfer: 432.64 KiB received, 126.70 KiB sent
  persistent keepalive: every 25 seconds

Joulinar, 2021-08-21:
Does it mean you like to connect to VNC via VPN? Are you able to reach PiHole web interface via VPN?

K92Pi, 2021-08-21:
> Does it mean you like to connect to VNC via VPN? Are you able to reach PiHole web interface via VPN?

The PI is connected to a VPN service at all the times. Pi-hole and other application's web interface is working properly. 

This is my wg0.conf file, the firewall rules are defined in iptables:

GNU nano 5.4                                                /etc/wireguard/wg0.conf                                                         
[Interface]
Address = xx.xx.xx.xxx/24
SaveConfig = true
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -A FORWARD -o %i -j ACCEPT; iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -D FORWARD -o %i -j ACCEPT; iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE
ListenPort = 51820
FwMark = 0xca6c
PrivateKey = kHU6gzqnmEl3hDERjiaXqm+uAlLHVnqcPQcaOJeTzFk=

[Peer]
PublicKey = 5u7NcwzKaX5yFHNoMUzZziOt6vz/i2MlKrM5AQeMQWA=
AllowedIPs = 0.0.0.0/0
Endpoint = xxx.xx.xx.xxx:1443
PersistentKeepalive = 25

K92Pi, 2021-08-21:
By the way, when I'm connected to the VNC server and the TV is on, with the Pi connected, I see the same screen on my laptop and my TV. So what I'm doing on my laptop, shows on the TV as well. This is something I also want to avoid. When I'm doing something on my laptop, I don't want the TV to show the same thing. I'm just guessing, but probably I need to use a different screen. In normal circumstances, it would look like this: IP address:0 and IP address:1. Right?

Joulinar, 2021-08-22:
having the shared screen is somehow expected according our online docs. https://dietpi.com/docs/software/remote_desktop/#realvnc-server

@MichaIng 
do you know if there is a possibility to have 2 X11 sessions running (the real one and the virtual one)? Or would that require to disable autologin? If I understood right, RealVNC will grep the real session if autologin is enabled.

MichaIng, 2021-08-22:
Okay, so to clear things up:
- There are two modes of running a VNC server:
    1. You run it in shared desktop mode, so that it connects to an already running desktop session.
    2. As virtual desktop, so that it connects to its individual desktop session which does not depend on any actual screen or even graphics hardware, possible on a completely headless system.
- Both server types listen different VNC "screen", `:0` for shared desktop and `:1` for virtual desktop sessions by default, which defines the port as `5900 + <screen>`, so `5900` for the shared desktop and `5901` for the virtual desktop.
- To connect to the virtual desktop, you usually need to append the screen index like `192.168.1.100:1` to the IP/domain in the VNC client, as you would do with a port usually. To be true I'm not sure if using the related port only works the same way on all VNC clients. When not adding any screen index or port, `:0` is used automatically, which is the shared desktop session, failing if there is none actually running.

As you do not want to have a shared desktop, I'd disable the related RealVNC service which checks for and automatically connects to real desktop sessions:
```sh
systemctl disable --now vncserver-x11-serviced
```

The virtual desktop session is controlled by our wrapper service:
```sh
systemctl enable --now vncserver
journalctl -u vncserver
```

Btw, do you have a RealVNC enterprise subscription? I'm not advertising it but it would allow a bit more efficient virtual desktop session handling which makes sense to use when having a subscription anyway.
______________
@Joulinar 
Yes both sessions can run concurrently. The RealVNC shared desktop service constantly checks for "real" desktop sessions and connects to the first one found. So with desktop autologin, both sessions will be available by default (given both services enabled), without desktop autologin, only the virtual one, until an actual X session is started.

Joulinar, 2021-08-22:
@MichaIng 
I'm not 100% sure that this is working. Disabling `vncserver-x11-serviced` doesn't seems to have any impact. The service is still active after reboot.

**disabling the service**

```
root@DietPi3:~# systemctl disable --now vncserver-x11-serviced
root@DietPi3:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

Aug 22 14:18:08 DietPi3 vncserver-x11[549]: SPrintConnMgr: Failed to add printer: server-error-service-unavailable
Aug 22 14:20:02 DietPi3 vncserver-x11[549]: Connections: disconnected: 192.168.0.53::57652 (TCP) ([ServerShutdown] VNC Server exited)
Aug 22 14:20:02 DietPi3 systemd[1]: Stopping VNC Server in Service Mode daemon...
Aug 22 14:20:02 DietPi3 vncserver-x11[549]: session ended: user root
Aug 22 14:20:02 DietPi3 vncserver-x11[549]: ServerManager: Server stopped
Aug 22 14:20:02 DietPi3 systemd[1]: vncserver-x11-serviced.service: Succeeded.
Aug 22 14:20:02 DietPi3 systemd[1]: vncserver-x11-serviced.service: Unit process 745 (vncagent) remains running after unit stopped.
Aug 22 14:20:02 DietPi3 systemd[1]: vncserver-x11-serviced.service: Unit process 747 (vncserverui) remains running after unit stopped.
Aug 22 14:20:02 DietPi3 systemd[1]: Stopped VNC Server in Service Mode daemon.
Aug 22 14:20:02 DietPi3 systemd[1]: vncserver-x11-serviced.service: Consumed 4.333s CPU time.
root@DietPi3:~#
```

**starting our service**

```
root@DietPi3:~# systemctl enable --now vncserver
root@DietPi3:~# systemctl status vncserver
● vncserver.service - Manage VNC Server (DietPi)
     Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sun 2021-08-22 14:17:07 CEST; 3min 55s ago
   Main PID: 492 (code=exited, status=0/SUCCESS)
      Tasks: 0 (limit: 2088)
        CPU: 0
     CGroup: /system.slice/vncserver.service

Aug 22 14:17:07 DietPi3 systemd[1]: Started Manage VNC Server (DietPi).
Aug 22 14:17:07 DietPi3 systemd[492]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)
root@DietPi3:~#
```

**no VNC service LISTEN**
```
root@DietPi3:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=495,fd=4))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=495,fd=5))
root@DietPi3:~#
```

**After reboot**

```
root@DietPi3:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=493,fd=4))
tcp   LISTEN 0      5            0.0.0.0:5900      0.0.0.0:*    users:(("vncserver-x11-c",pid=545,fd=11))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=493,fd=5))
tcp   LISTEN 0      5               [::]:5900         [::]:*    users:(("vncserver-x11-c",pid=545,fd=10))
root@DietPi3:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-08-22 14:22:09 CEST; 2min 17s ago
   Main PID: 544 (vncserver-x11-s)
      Tasks: 5 (limit: 2088)
        CPU: 790ms
     CGroup: /system.slice/vncserver-x11-serviced.service
             ├─544 /usr/bin/vncserver-x11-serviced -fg
             ├─545 /usr/bin/vncserver-x11-core -service
             ├─724 /usr/bin/vncagent service 14
             ├─726 /usr/bin/vncserverui service 15
             └─734 /usr/bin/vncserverui -statusicon 5

Aug 22 14:22:09 DietPi3 systemd[1]: Started VNC Server in Service Mode daemon.
Aug 22 14:22:09 DietPi3 vncserver-x11[545]: ServerManager: Server started
Aug 22 14:22:09 DietPi3 vncserver-x11[545]: ConsoleDisplay: Cannot find a running X server on vt1
Aug 22 14:22:14 DietPi3 vncserver-x11[545]: ConsoleDisplay: Found running X server (pid=597, binary=/usr/lib/xorg/Xorg)
root@DietPi3:~#
```

MichaIng, 2021-08-22:
@Joulinar 
Did you set `SOFTWARE_VNCSERVER_SHARE_DESKTOP=1` in `config.txt`? In this case our wrapper service starts `vncserver-x11-serviced`. I think we should change it so that it starts `/usr/bin/vncserver-x11-serviced -fg` directly and having the RealVNC systemd unit disabled by default. So there is only a single service and the dietpi.txt option to handle all kind VNC sessions and also all logs are available via `journalctl -u vncserver` then. For running both types, the RealVNC service can still be enabled.

Ah dammit, RealVNC can only be connected to with RealVNCs own clients, right? 😢

K92Pi, 2021-08-22:
Hi Guys! @MichaIng @Joulinar 

@MichaIng I've followed your description, 
`systemctl disable --now vncserver-x11-serviced`

then
`systemctl enable --now vncserver` - after this, reboot

>root@MediaServer:~# journalctl -u vncserver
-- Journal begins at Sun 2021-08-22 10:44:04 EDT, ends at Sun 2021-08-22 10:48:13 EDT. --
Aug 22 10:44:59 MediaServer systemd[1]: Started Manage VNC Server (DietPi).
Aug 22 10:44:59 MediaServer systemd[823]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)

> root@MediaServer:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-08-22 10:44:59 EDT; 4min 51s ago
   Main PID: 867 (vncserver-x11-s)
      Tasks: 2 (limit: 4915)
        CPU: 726ms
     CGroup: /system.slice/vncserver-x11-serviced.service
             ├─867 /usr/bin/vncserver-x11-serviced -fg
             └─871 /usr/bin/vncserver-x11-core -service

Aug 22 10:44:59 MediaServer systemd[1]: Started VNC Server in Service Mode daemon.
Aug 22 10:44:59 MediaServer vncserver-x11[871]: ServerManager: Server started
Aug 22 10:45:00 MediaServer vncserver-x11[871]: ConsoleDisplay: Cannot find a running X server on vt1
Aug 22 10:48:02 MediaServer vncserver-x11[871]: Connections: connected: 192.168.1.144::54798 (TCP)
Aug 22 10:48:10 MediaServer vncserver-x11[871]: session started: user root permissions f
Aug 22 10:48:10 MediaServer vncserver-x11[871]: Connections: authenticated: 192.168.1.144::54798 (TCP), as root (f permissions)
Aug 22 10:48:10 MediaServer vncserver-x11[871]: SPrintConnMgr: Failed to add printer: server-error-service-unavailable
Aug 22 10:48:13 MediaServer vncserver-x11[871]: Connections: disconnected: 192.168.1.144::54798 (TCP) ([ViewerClosed] VNC Viewer closed)
Aug 22 10:48:13 MediaServer vncserver-x11[871]: session ended: user root

VNC port is present:
>root@MediaServer:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1024         0.0.0.0:32400      0.0.0.0:*    users:(("Plex Media Serv",pid=1251,fd=56))
tcp   LISTEN 0      1024         0.0.0.0:80         0.0.0.0:*    users:(("lighttpd",pid=1234,fd=4))        
tcp   LISTEN 0      1024       127.0.0.1:32401      0.0.0.0:*    users:(("Plex Media Serv",pid=1251,fd=57))
tcp   LISTEN 0      1          127.0.0.1:35571      0.0.0.0:*    users:(("Plex Tuner Serv",pid=1591,fd=79))
tcp   LISTEN 0      128        127.0.0.1:43509      0.0.0.0:*    users:(("Plex Script Hos",pid=1635,fd=4)) 
tcp   LISTEN 0      32           0.0.0.0:53         0.0.0.0:*    users:(("pihole-FTL",pid=711,fd=5))       
tcp   LISTEN 0      1000         0.0.0.0:22         0.0.0.0:*    users:(("dropbear",pid=827,fd=4))         
tcp   LISTEN 0      256        127.0.0.1:5335       0.0.0.0:*    users:(("unbound",pid=574,fd=4))          
tcp   LISTEN 0      128        127.0.0.1:32600      0.0.0.0:*    users:(("Plex Tuner Serv",pid=1591,fd=10))
tcp   LISTEN 0      256        127.0.0.1:8953       0.0.0.0:*    users:(("unbound",pid=574,fd=5))          
tcp   LISTEN 0      128        127.0.0.1:40157      0.0.0.0:*    users:(("Plex Script Hos",pid=1545,fd=9)) 
tcp   LISTEN 0      50           0.0.0.0:445        0.0.0.0:*    users:(("smbd",pid=914,fd=44))            
tcp   LISTEN 0      5          127.0.0.1:4711       0.0.0.0:*    users:(("pihole-FTL",pid=711,fd=11))      
tcp   LISTEN 0      50           0.0.0.0:139        0.0.0.0:*    users:(("smbd",pid=914,fd=45))            
**tcp   LISTEN 0      5            0.0.0.0:5900       0.0.0.0:*    users:(("vncserver-x11-c",pid=871,fd=11))** 
tcp   LISTEN 0      1024            [::]:80            [::]:*    users:(("lighttpd",pid=1234,fd=5))        
tcp   LISTEN 0      32              [::]:53            [::]:*    users:(("pihole-FTL",pid=711,fd=7))       
tcp   LISTEN 0      1000            [::]:22            [::]:*    users:(("dropbear",pid=827,fd=5))         
tcp   LISTEN 0      50              [::]:445           [::]:*    users:(("smbd",pid=914,fd=42))            
tcp   LISTEN 0      50              [::]:139           [::]:*    users:(("smbd",pid=914,fd=43))            
**tcp   LISTEN 0      5               [::]:5900          [::]:*    users:(("vncserver-x11-c",pid=871,fd=10))** 

The problem is still present, can't show the desktop. I've tried to play with ports, adding :1 or :0 to the IP address, but didn't work.

MichaIng, 2021-08-22:
Can you both please paste:
```sh
grep 'SOFTWARE_VNCSERVER_SHARE_DESKTOP' /boot/dietpi.txt
```

Joulinar, 2021-08-22:
I've set it to 1 but it doesn't change the behaviour

```
root@DietPi3:~# grep 'SOFTWARE_VNCSERVER_SHARE_DESKTOP' /boot/dietpi.txt
SOFTWARE_VNCSERVER_SHARE_DESKTOP=1
root@DietPi3:~#
```

MichaIng, 2021-08-22:
So then it works as expected: As you chose shared desktop mode, the RealVNC shared desktop service is started. Change it to `SOFTWARE_VNCSERVER_SHARE_DESKTOP=0` to switch to the virtual desktop.

Joulinar, 2021-08-22:
Yes I did following our discussion. But it doesn't matter if it is `1` or `0`

MichaIng, 2021-08-22:
Okay, that is strange. What happens when you have `SOFTWARE_VNCSERVER_SHARE_DESKTOP=0` and then do:
```sh
systemctl stop vncserver
systemctl stop vncserver-x11-serviced
vncserver start
sleep 5
systemctl status vncserver
systemctl status vncserver-x11-serviced
```

Joulinar, 2021-08-22:
VNC is connecting to the real X11 session

```
root@DietPi3:~# grep 'SOFTWARE_VNCSERVER_SHARE_DESKTOP' /boot/dietpi.txt
SOFTWARE_VNCSERVER_SHARE_DESKTOP=0
root@DietPi3:~# systemctl stop vncserver
root@DietPi3:~# systemctl stop vncserver-x11-serviced
root@DietPi3:~# vncserver start
root@DietPi3:~# systemctl status vncserver
● vncserver.service - Manage VNC Server (DietPi)
     Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2021-08-22 17:18:34 CEST; 51s ago
    Process: 490 ExecStart=/usr/local/bin/vncserver start (code=exited, status=0/SUCCESS)
    Process: 796 ExecStop=/usr/local/bin/vncserver stop (code=exited, status=0/SUCCESS)
   Main PID: 490 (code=exited, status=0/SUCCESS)
        CPU: 22ms

Aug 22 17:16:22 DietPi3 systemd[1]: Started Manage VNC Server (DietPi).
Aug 22 17:16:22 DietPi3 systemd[490]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)
Aug 22 17:18:34 DietPi3 systemd[1]: Stopping Manage VNC Server (DietPi)...
Aug 22 17:18:34 DietPi3 systemd[796]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)
Aug 22 17:18:34 DietPi3 systemd[1]: vncserver.service: Succeeded.
Aug 22 17:18:34 DietPi3 systemd[1]: Stopped Manage VNC Server (DietPi).
root@DietPi3:~#
root@DietPi3:~#
root@DietPi3:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-08-22 17:19:08 CEST; 40s ago
   Main PID: 811 (vncserver-x11-s)
      Tasks: 5 (limit: 2088)
        CPU: 756ms
     CGroup: /system.slice/vncserver-x11-serviced.service
             ├─811 /usr/bin/vncserver-x11-serviced -fg
             ├─812 /usr/bin/vncserver-x11-core -service
             ├─821 /usr/bin/vncagent service 14
             ├─823 /usr/bin/vncserverui service 15
             └─831 /usr/bin/vncserverui -statusicon 5

Aug 22 17:19:08 DietPi3 systemd[1]: Started VNC Server in Service Mode daemon.
Aug 22 17:19:08 DietPi3 vncserver-x11[812]: ServerManager: Server started
Aug 22 17:19:08 DietPi3 vncserver-x11[812]: ConsoleDisplay: Found running X server (pid=598, binary=/usr/lib/xorg/Xorg)
root@DietPi3:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=494,fd=4))
tcp   LISTEN 0      5            0.0.0.0:5900      0.0.0.0:*    users:(("vncserver-x11-c",pid=812,fd=11))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=494,fd=5))
tcp   LISTEN 0      5               [::]:5900         [::]:*    users:(("vncserver-x11-c",pid=812,fd=10))
root@DietPi3:~#
```

K92Pi, 2021-08-22:
Okay, in my system the `SOFTWARE_VNCSERVER_SHARE_DESKTOP=0` was default. I've found an older flash drive with another DietPi and I just checked what display resolution was set: 
[https://imgur.com/GHAqAod](url)
I set the same thing in my new DietPi system and now it's working!

K92Pi, 2021-08-22:
@MichaIng I did this while the HDMI cable was plugged in, no screen on the TV, only VNC desktop:
```
root@MediaServer:~# systemctl stop vncserver
root@MediaServer:~# systemctl stop vncserver-x11-serviced
root@MediaServer:~# vncserver start
root@MediaServer:~# sleep 5
root@MediaServer:~# systemctl status vncserver
● vncserver.service - Manage VNC Server (DietPi)
     Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2021-08-22 11:32:09 EDT; 30s ago
    Process: 857 ExecStart=/usr/local/bin/vncserver start (code=exited, status=0/SUCCESS)
    Process: 2225 ExecStop=/usr/local/bin/vncserver stop (code=exited, status=0/SUCCESS)
   Main PID: 857 (code=exited, status=0/SUCCESS)
        CPU: 17ms

Aug 22 11:18:59 MediaServer systemd[1]: Started Manage VNC Server (DietPi).
Aug 22 11:18:59 MediaServer systemd[857]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)
Aug 22 11:32:09 MediaServer systemd[1]: Stopping Manage VNC Server (DietPi)...
Aug 22 11:32:09 MediaServer systemd[2225]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)
Aug 22 11:32:09 MediaServer systemd[1]: vncserver.service: Succeeded.
Aug 22 11:32:09 MediaServer systemd[1]: Stopped Manage VNC Server (DietPi).
root@MediaServer:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-08-22 11:32:21 EDT; 33s ago
   Main PID: 2241 (vncserver-x11-s)
      Tasks: 5 (limit: 4915)
        CPU: 546ms
     CGroup: /system.slice/vncserver-x11-serviced.service
             ├─2241 /usr/bin/vncserver-x11-serviced -fg
             ├─2242 /usr/bin/vncserver-x11-core -service
             ├─2251 /usr/bin/vncagent service 14
             ├─2253 /usr/bin/vncserverui service 15
             └─2261 /usr/bin/vncserverui -statusicon 5

Aug 22 11:32:21 MediaServer systemd[1]: Started VNC Server in Service Mode daemon.
Aug 22 11:32:21 MediaServer vncserver-x11[2242]: ServerManager: Server started
Aug 22 11:32:21 MediaServer vncserver-x11[2242]: ConsoleDisplay: Found running X server (pid=976, binary=/usr/lib/xorg/Xorg)
```

MichaIng, 2021-08-22:
The virtual VNC session shouldn't depend on any GPU or display settings, only the real (HDMI) session and hence shared session should.

Based on your output, it's still the same: The shared session service starts, the virtual desktop not. So now: Why does it not work on both your systems while it works perfectly fine on mine.

Do you both have desktop autologin enabled?

K92Pi, 2021-08-22:
After restart, I'm able to connect to VNC and see the desktop and there's also signal to the TV. Right now it's a shared desktop. So whatever I do on the VNC, I see the same thing on TV.
I unplugged the HDMI cable and then restarted the Pi, VNC works, but there's no signal for the TV when I plug the cable back.

K92Pi, 2021-08-22:
> The virtual VNC session shouldn't depend on any GPU or display settings, only the real (HDMI) session and hence shared session should.
>
> Based on your output, it's still the same: The shared session [...]

Yes, autologin is [enabled](https://imgur.com/4kOmehF).

Joulinar, 2021-08-22:
> Do you both have desktop autologin enabled?

Yes I do. I thought this was the question to have 2 parallel session. A real and a virtual one.

MichaIng, 2021-08-22:
Ahhh, now I see:
```
[[ -f '/boot/dietpi/.dietpi-autostart_index' && $(</boot/dietpi/.dietpi-autostart_index) == 2 ]] && SHARED_MODE=1
```
So when desktop autologin is enabled, we override the mode to "prevents another VNC server being launched on :1". While it mostly makes sense to use shared desktop mode, when you have a desktop autostarting anyway, it breaks the possibility to have both, virtual and shared desktop running, or to have a local desktop and an independent VNC desktop. And we only override the mode in case of RealVNC, but not for TigerVNC, which isn't consistent.

Fix it via:
```sh
sed -i '/SHARED_MODE=1/d' /usr/local/bin/vncserver
```
I'll do the same in our code.

But the idea behind it stays valid. Do not enable desktop autologin unless you really have a HDMI and keyboard attached that you want to use, else it consumes unnecessarily a lot of resources.

K92Pi, 2021-08-22:
@MichaIng It worked! Now I have a virtual VNC desktop and a "real" desktop on the HDMI port. So, you're recommending to keep only a virtual desktop if I don't use the HDMI port often, because it consumes a lot of resources. Am I understanding right?

MichaIng, 2021-08-22:
> So, you're recommending to keep only a virtual desktop if I don't use the HDMI port often, because it consumes a lot of resources. Am I understanding right?

Yes exactly. Better to either use the LightDM based login mask or common console login + `startx` to start a desktop on demand. I mean as long as you have plenty of free memory, it doesn't matter, so partly its my perfectionism to only run what I really need 😉.

K92Pi, 2021-08-23:
All right! I appreciate your time for working on this!

MichaIng, 2021-08-23:
Let me keep this open until I finished the rework, I'm currently doing 🙂.

closed 2021-09-12 by MichaIng