MichaIng/DietPi #4680: Transmission client permission issues

K92Pi, 2021-08-24:
I have a problem with my Transmission client. I've installed it trough the DietPi-Software, so I will need some explanation how to configure this client. After installation I changed the settings in `/etc/transmission-daemon/settings.json` as I did in the other Raspbian. I was following [this](https://pimylifeup.com/raspberry-pi-transmission/) tutorial previously, but it wasn't written for DietPi. So I'd like to know how can I set permissions for an external HDD? I've set auto login on for the root user. The downloads folder is at ` /mnt/data/Downloads`. I've set the permission `chown -R root:root /mnt/data/Downloads`. - didn't work. Then I checked if the "debian-transmission" user was added to the system, it was! So after that I tried to give permission to this user: `chown -R debian-transmission:debian-transmission /mnt/data/Downloads` - I got the error message "Operation not permitted". 
```

root@DietPi:~# systemctl status transmission-daemon.service
● transmission-daemon.service - Transmission BitTorrent Daemon
     Loaded: loaded (/lib/systemd/system/transmission-daemon.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-08-23 20:22:33 EDT; 10min ago
   Main PID: 2402 (transmission-da)
     Status: "Idle."
      Tasks: 3 (limit: 4178)
        CPU: 1.980s
     CGroup: /system.slice/transmission-daemon.service
             └─2402 /usr/bin/transmission-daemon -f --log-error

Aug 23 20:22:32 DietPi systemd[1]: Starting Transmission BitTorrent Daemon...
Aug 23 20:22:33 DietPi systemd[1]: Started Transmission BitTorrent Daemon.
Aug 23 20:22:34 DietPi transmission-daemon[2402]: [2021-08-23 20:22:34.068] [torrent file] No data found! Ensure your drives are connected or use "Set Location". To re-download, remove the torrent and re-add it. (torrent.c:574)
```

I'm pretty much done. I'm turning to the developers to help me, I've spent so much time to find a solution... Please get back to me! Thank you!

Joulinar, 2021-08-24:
Hi,

can you share following information as you did not fill the questioner while opening the issue.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

As well pls share

```
ls -la /mnt/data/
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
cat /etc/fstab
```

K92Pi, 2021-08-24:
Hi @Joulinar!
Of course, let me share the required information!
```
root@MediaServer:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

```
root@MediaServer:~# echo $G_DISTRO_NAME
bullseye
```
```

root@MediaServer:~# uname -a
Linux MediaServer 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux
```

```
root@MediaServer:~# echo $G_HW_MODEL_NAME
RPi 4 Model B (aarch64)
```

```
root@MediaServer:~# ls -la /mnt/data/
total 3332
drwxrwxrwx   9 root root 262144 Aug 23 23:15  .
drwxr-xr-x   7 root root   4096 Aug 23 22:08  ..
-rwxrwxrwx   1 root root  14340 Aug  1 20:17  .DS_Store
drwxrwxrwx   3 root root 262144 Mar 16 16:10  .Spotlight-V100
-rwxrwxrwx   1 root root   4096 Mar 17 16:58  ._.DS_Store
drwxrwxrwx   2 root root 262144 Mar 16 16:10  .fseventsd
drwxrwxrwx 424 root root 262144 Aug 24 01:49  Downloads
.....
```

```
root@MediaServer:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
sda               119.5G  0 disk                                                 
├─sda1
│    vfat           128M  0 part /boot      8b0ac6f5-01                          xxxx-xxxx
└─sda2
     ext4         119.4G  0 part /          8b0ac6f5-02                          xxxxx
sdb                 3.6T  0 disk                                                 
├─sdb1
│    vfat   EFI     200M  0 part            xxxxxxx xxxx-xxxx
└─sdb2
     exfat  RASPIDATA
                    3.6T  0 part /mnt/data  xxxxxxxxxx  xxxx-xxxx
```
```

root@MediaServer:~# cat /etc/fstab
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1922M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=xxxxxxx / ext4 noatime,lazytime,rw 0 1
PARTUUID=xxxxxxx /boot vfat noatime,lazytime,rw 0 2
UUID=xxxxxxxx /mnt/data exfat noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```

I was doing more and more research on the internet in regards to permission error with Transmission and I've tried a couple of things. However, I used **dietpi-drive_manager** to mount my external USB drive. I've read somebody was able to solve the problem the permission issue with some modification in  **/etc/fstab**. The drive manager automatically generates the permissions in fstab. I didn't save the original line for the mount, but I've changed to this: `UUID=[UUID] /mnt/data exfat defaults,auto,users,rw,nofail,noatime 0 0
`
then these commands will run through the directory setting permissions for both files and directories:
```
find /mnt/data/ -type d -exec chmod 755 {} \;
find /mnt/data/ -type f -exec chmod 644 {} \;
```

Reboot, then magically the Transmission has started working.

Joulinar, 2021-08-25:
I see your disk is using `exfat` as file system format. exFAT, as an extension of FAT. Best to my knowledge exFAT is not capable of storing discretionary access control metadata. This is why all the files on an exFAT volume appear to be 777 permissions. Usually running `chmod` should not have any effect or is leading to error messages.

@MichaIng
pls correct me if I'm wrong.

MichaIng, 2021-08-25:
Many thanks for your report.

First of all, `chown root:root` will never grant any permissions, but always cut/limit them, as the `root` anyway has always full access, but no other user is (or should be) in the `root` group. As Transmission runs as user `debian-transmission` on all Debian-based distributions, `chown -R debian-transmission:debian-transmission` was the right attempt to grant it write access. On DietPi, you may use `debian-transmission:dietpi` which grants other media-related software write access as well. As `/mnt/dietpi_userdata/downloads` is the default download directory on DietPi, you could basically copy owner and modes from there to have functional permissions. This is comparable to the guide you followed when they override the service user with the `pi` user, which IMO is not an ideal way from security perspective: the `pi` user is a login user with pull root permission (via password-less `sudo`), not intended to be used as service user as it grants the running process much too much permissions. DietPi solves this mostly with the shared `dietpi` group.

But the problem now is that your target drive is `exFAT`, which does not support UNIX permissions at all, so all `chown`/`chmod` cannot work. Is there specific reason to use a Windows filesystem type on a Linux server? Else I suggest to use ext4, F2FS or Btrfs, depending on the type of drive and other filesystem features you may want. `ext4` is still a great an reliable allrounder choice for all drives on Linux.

If you need to use exFAT, you need to mount it with explicit user:group so that Transmission has write access, i.e. `uid=transmission-daemon,gid=dietpi,dmask=0775,fmask=0664`, or is not other software requires access `uid=transmission-daemon,dmask=0755,fmask=0644`.

Now that `chmod` on our last attempt worked (?) and Transmission has access without you changing either the owner of the exFAT mount (?) or the service user of Transmission (?), is not possible, so you changed one of these things, which again isn't ideal from security perspective. So which user runs Transmission now and how to the permissions of the drive/content look like?
```sh
ls -l /mnt/data
```
**EDIT: While I was writing, @Joulinar solved the mystery: Hmm that `777` is the default isn't great and I now remember that I wanted to change this already 🤔. FAT is `root:root 755/644`, expected and reasonable when mounting a drive as root and a default umask of 022. Shall we change this at least to `dietpi:dietpi 775:664` to not fully expose the drive to each and every user? Best would be to have it choosable as well directly in `dietpi-drive_manager`.**

Joulinar, 2021-08-25:
usually there is a download or media accessing the disk. Therefore using dietpi user/group might be a good option

K92Pi, 2021-09-03:
@MichaIng 
Sorry, for bothering you, but could you please tell me how my /etc/fstab should look like? I'm still struggling with the permissions problem. (I have a new system) The setup is the same, external drive is exfat, mounted to /mnt/data

MichaIng, 2021-09-03:
It should actually work OOTB as the exFAT drive is by default mounted with 777 mode. Can you verify:
```sh
ls -al /mnt/data
```

K92Pi, 2021-09-03:
Sure,
root@MediaServer:~# ls -al /mnt/data
total 3332
drwxr-xr-x   9 root root 262144 Sep  3 10:11  .
drwxr-xr-x   7 root root   4096 Sep  3 07:04  ..
-rwxr-xr-x   1 root root  14340 Sep  3 06:16  .xxxxxx
drwxr-xr-x   3 root root 262144 Mar 16 16:10  .xxxxxxx
-rwxr-xr-x   1 root root   4096 Mar 17 16:58  ._.xxxxxx
-rwxr-xr-x   1 root root   4096 Mar 18 11:17  ._xxxxxx
-rwxr-xr-x   1 root root   4096 Mar 17 17:02  ._xxxxxx
drwxr-xr-x   2 root root 262144 Mar 16 16:10  .fseventsd
-rwxr-xr-x   1 root root 109907 Mar 18 11:17  xxxxxxx
drwxr-xr-x   3 root root 262144 Sep  3 08:35  xxxxx
drwxr-xr-x 424 root root 262144 Sep  3 09:21  Downloads
drwxr-xr-x   3 root root 262144 Aug 20 18:15 'xxxxxx'
drwxr-xr-x  11 root root 262144 Apr  9 06:30 'xxxxx'
drwxr-xr-x  38 root root 262144 Sep  3 09:11  xxxxx

Joulinar, 2021-09-03:
Can you show as well `cat /etc/fstab`

K92Pi, 2021-09-03:
@Joulinar 

```
root@MediaServer:~# cat /etc/fstab
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1944M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=0e2887fb-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=0e2887fb-01 /boot vfat noatime,lazytime,rw 0 2
UUID=6051-10CB /mnt/data exfat defaults,auto,users,rw,nofail,noatime 0 0
```

MichaIng, 2021-09-03:
Can you please revert it to what drive manager would add. The additional options shouldn't have any effect. I'm currently testing whether indeed exFAT now is mounted with 755/644 as yours here, which would be good, but then means the mount options need to be changed.
```
UUID=6051-10CB /mnt/data exfat noatime,lazytime,rw,nofail,noauto,x-systemd.automount 0 0
```
Then remount it:
```sh
umount /mnt/data
mount /mnt/data
```

K92Pi, 2021-09-03:
All right, now my /etc/fstab changed to this:
`UUID=6051-10CB /mnt/data exfat noatime,lazytime,rw,nofail,noauto,x-systemd.automount 0 0`

I did a reboot just to make sure...

then:

```
root@MediaServer:~# ls -al /mnt/data
total 3332
drwxr-xr-x   9 root root 262144 Sep  3 11:05  .
drwxr-xr-x   7 root root   4096 Sep  3 07:04  ..
-rwxr-xr-x   1 root root  14340 Sep  3 06:16  .
drwxr-xr-x   3 root root 262144 Mar 16 16:10  .
-rwxr-xr-x   1 root root   4096 Mar 17 16:58  ._.
-rwxr-xr-x   1 root root   4096 Mar 18 11:17  ._
-rwxr-xr-x   1 root root   4096 Mar 17 17:02  ._
drwxr-xr-x   2 root root 262144 Mar 16 16:10  .fseventsd
-rwxr-xr-x   1 root root 109907 Mar 18 11:17  
drwxr-xr-x   3 root root 262144 Sep  3 08:35  
drwxr-xr-x 424 root root 262144 Sep  3 09:21  Downloads
drwxr-xr-x   3 root root 262144 Aug 20 18:15 ''
drwxr-xr-x  11 root root 262144 Apr  9 06:30 ''
drwxr-xr-x  38 root root 262144 Sep  3 09:11  
```

I don't see any impact on it.

MichaIng, 2021-09-03:
Ahh, I think this is the result of native Linux exFAT support, so `fuse` is not in use anymore and the native Linux driver obviously chooses more reasonable/secure permissions 🙂.

So to grant permissions for Transmissions, append owner and modes to the fstab line:
```
UUID=6051-10CB /mnt/data exfat noatime,lazytime,rw,nofail,noauto,x-systemd.automount,uid=transmission-daemon,gid=dietpi,dmask=0002,fmask=0002
```
With this, all members of the "dietpi" group have read-write permissions. The two zeros can be skipped btw.

MichaIng, 2021-09-03:
Verified on Buster VM: Mounting there with Linux v4.19 leads to 777 modes. Also no `lsmod | grep exfat` kernel module is available and the mount is hold active via fuse process, visible in `htop`. Killing it kills the mount. Upgrading via Debian backports to Linux v5.10, reboot and the `exfat` kernel module is available + makes allows to mount the drive without fuse. Time to integrate this natively into `dietpi-drive_manager`.

K92Pi, 2021-09-03:
@MichaIng 
It looks like there's no permission error in Transmission now. Hopefully it stays like this! :)

MichaIng, 2021-09-03:
Great. Just keep in mind that drive manager reverts the change until next release.

MichaIng, 2021-09-10:
Done with: #4725

closed 2021-09-12 by MichaIng