MichaIng/DietPi #4687: Unable to open [Customize Look and Feel] (lxappearance) in Bullseye

pinipon, 2021-08-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
bullseye

- Kernel version | `uname -a`
Linux ifmtvpi2 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 3A
- SDcard used | (EG: SanDisk ultra)
ADATA HC I

#### Additional Information (if applicable)
- Software title:
LXDE

- Was the software title installed freshly or updated/migrated?
freshly
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Bullseye freshly installed, LXDE, autologin user => dietpi
2. Once in LXDE, click on Preferences->Customize Look and Feel
3. Nothing happens
4. Running command "lxappearance" on terminal returns "Segementation fault
5. Same procedure repeated as user root: same result as above

#### Expected behaviour

- openning [Customize Look and Feel / lxappearance] should open a window similar in previous release (Buster)

#### Actual behaviour

- No window appears.
- Segmentation fault

#### Extra details

- please ask me

Joulinar, 2021-08-25:
@MichaIng 
I'm not sure if there is any relation to https://github.com/MichaIng/DietPi/issues/1558
At least I found some statements about hanging `lxappearance`

MichaIng, 2021-08-25:
Probably there are additional packages pulled from the RPi repo related to the RPi desktop which we need to exclude now. I set the pins conservative, only blocking those packages known to cause issues or being installed by default with LXDE, but it seems better to block a larger range of everything that is related to the RPi desktop to be future prove and have a pure Debian LXDE desktop. Quite nasty that this is not an own repository component anymore so that it can be added and removed easily.

I'll run some tests, but @pinipon could you show the output of the following command, please:
```sh
dpkg -l | grep '+rpt'
```

MichaIng, 2021-08-28:
I was thinking to block all `lx*` packages, was just afraid there may be some below which are not related to LXDE. But actually it is easy to verify that there is no need to worry: https://archive.raspberrypi.org/debian/pool/main/l/
Those `lxplug-` panel plugin packages do not exist on Debian, but they also are specific for the Raspberry Pi desktop, network stack etc, e.g. the network one is for configuring dhcpcd (which we do not use), cputemp for sure invokes `vcgencmd`, while the generic /sys tunable (and hence the default CPU temp plugin) work exactly the same way, also with less overhead. I didn't test the others, but I do not think they would be used much anyway (as not pre-installed) and everything can be configured via dietpi-tools or native Debian/LXDE plugins, and there is no guarantee these work (or will work in the future) with the Debian lxpanel. So better to block all together, so be on the same side.

But lxhotkey (I guess you mean `lxhotkey-gtk`)? Is not part of those. Probably it is conflicting with one of the other still installed rpt packages on your system. Can you check again which one is still there?
```sh
dpkg -l | grep 'rpt'
```

pinipon, 2021-08-28:
Yes it is `lxhotkey-gtk` package (sorry, my mistake ). I agree better to stay on native Debian/LXDE plugins.
`dietpi@DietPi:~$ dpkg -l | grep 'rpt'`

pinipon, 2021-08-28:
Yes it's `lxhotkey-gtk` (sorry, my mistake).
```
dietpi@DietPi:~$ dpkg -l | grep 'rpt'
ii  alsa-utils                     1.2.4-1+rpt1                   armhf        Utilities for configuring and using ALSA
ii  gpicview                       0.2.5-3+rpt1                   armhf        lightweight image viewer
ii  gtk-update-icon-cache          3.24.24-4+rpt2                 armhf        icon theme caching utility
ii  libasound2:armhf               1.2.4-1.1+rpt1                 armhf        shared library for ALSA applications
ii  libasound2-data                1.2.4-1.1+rpt1                 all          Configuration files and profiles for ALSA drivers
ii  libatopology2:armhf            1.2.4-1.1+rpt1                 armhf        shared library for handling ALSA topology definitions
ii  libcairo-gobject2:armhf        1.16.0-5+rpt1                  armhf        Cairo 2D vector graphics library (GObject library)
ii  libcairo2:armhf                1.16.0-5+rpt1                  armhf        Cairo 2D vector graphics library
ii  libpolkit-agent-1-0:armhf      0.105-31+rpt1                  armhf        PolicyKit Authentication Agent API
ii  libpolkit-gobject-1-0:armhf    0.105-31+rpt1                  armhf        PolicyKit Authorization API
ii  lxlock                         0.5.5-2+rpt2                   all          simple locking utility for LXDE
ii  lxpolkit                       0.5.5-2+rpt2                   armhf        LXDE PolicyKit authentication agent
ii  lxterminal                     0.4.0-1+rpt2                   armhf        LXDE terminal emulator
ii  policykit-1                    0.105-31+rpt1                  armhf        framework for managing administrative policies and privileges
```
In the meantime, I solved the issue of `lxhotkey-gtk` by simply installing the package `lxhotkey-plugin-openbox`:
`sudo apt install lxhotkey-plugin-openbox`
Now it works.

MichaIng, 2021-08-29:
Okay, I guess this is expected. `lxhotkey-plugin-openbox` is a recommendation but not a dependency of `lxhotkey-gtk` > `lxhotkey-core`, and on DietPi recommendations are not installed by default.

pinipon, 2021-08-29:
Yes `lxhotkey-plugin-openbox` is a recommendation, not a dependency. But, once one install LXDE, as `lxhotkey-gtk` is installed by default and create an entry menu in [Preferences] section, you will get this error window/message by default if you click [Setup Hot Keys] and have to spend some time investigating how to correct this error. Now the question is: do you prefer to leave this situation in place at the price to maintain a strictly minimal image or could you sacrifice the installation of one recommended package to avoid this situation and to be ready OTB?
Look, despite the fact that not all my RPi projects where I use/need LXDE need hotkeys, in my case I could simply edit the file `home/dietpi/.config/openbox/lxde-rc.xml` and add my hotkeys in it, but it is not convenient and it is really working under the hood... and I don't like to leave my users with entries generating errors.

Anyway, I want to thank you for your very valuable support and quick replies on these issues, and this why I made a good choice to work exclusively with DietPi images on all my RPi Projects (with and without LXDE).

From my side you can close the topic.

pinipon, 2021-08-30:
Well... I'm struggling now to change icon theme to `papirus-icon-theme`. I don't know if it is related to the issues above, but when I'm:

pinipon, 2021-08-30:
Well... I'm struggling now to change icon theme to `papirus-icon-theme`. I don't know if it is related to the issues above, but when I'm installing the package:
```
apt install papirus-icon-theme
```
the icon theme is not correctly loaded in [Customize Look and Feel] (`lxappearance`).
Same issue in 32 and 64-bit Bullseye image. This not happens in Buster images, it worked well in previous release.

Do you have an idea to solve this issue?

MichaIng, 2021-08-30:
In your 2nd last post, I'll verify this on VM, as actually I tested every menu entry once on Bullseye LXDE, a while ago. When verified, I agree then the plugin should be installed by default. Not nice to have a menu entry when it cannot be used.

About the icon theme: What do you mean by "not correctly loaded"? Can you check the X server logs for errors, in case? When its a root session, `/var/log/Xorg.0.log`, when its a non-root user session, it should be in `~/.local/share/xorg/Xorg.0.log`.

Another idea to get error outputs: Start `lxappearance` from terminal emulator, so it should contain outputs afterwards.

We could go through the remaining RPi desktop packages, starting with `gtk-update-icon-cache` and then the cairo libs, blacklisting and replacing them with the Debian ones step by step and see whether this helps.

MichaIng, 2021-08-30:
Both verified: I'll add the lxhotkey plugin to be installed by default with LXDE.

Not sure about the icon theme, so its not an issue with the RPi desktop, but a general Bullseye issue (if it worked on Buster). Not sure if this bug report is related: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=982901

Switching between Adwaita and nuoveXT works well. `breeze-icon-theme` doesn't work either, but `faenza-icon-theme` works. Breeze is for KDE plasma desktop, so probably it is expected to miss the icons for LXDE/pcmanfm.

The is a newer Papirus version available via backports: https://packages.debian.org/bullseye-backports/papirus-icon-theme
But this doesn't fix the issue. I tend to think that it is an issue with LXDE. As I have not really an idea what the issue is, we should report it to the Debian bug tracker: https://www.debian.org/Bugs/Reporting

pinipon, 2021-08-31:
```
apt install librsvg2-common
```
solve the issue with any icon-theme. So, it would be nice if you could install this package by default when one choose LXDE desktop on `dietpi-software`

MichaIng, 2021-09-01:
Great find. How did you find out about this?

This package was pulled as dependency with the Adwaita icon theme until Buster: https://packages.debian.org/buster/adwaita-icon-theme
But it has been degraded to a recommendation with Bullseye and is hence not automatically installed anymore: https://packages.debian.org/bullseye/adwaita-icon-theme

Actually best would be if all icon themes, which depend on this additional SVG library, pull it as dependency.

pinipon, 2021-09-02:
I decided to install each icon theme on the repository and issues are solved after installing `gnome-icon-theme`. Looking at the logs, `librsvg2-common` is installed as a dependency of `gnome-icon-theme`, so I tried a fresh install+`librsvg2-common` and it solved `breeze-icon-theme` and `papirus-icon-theme` as well.

MichaIng, 2021-09-02:
Awesome. So `gnome-icon-theme` can be taken as a good example to open requests on the bug tracker for having it added as dependency with the other two icon themes as well, and others with make heavy use of SVGs.

MichaIng, 2021-09-12:
Both packages are now installed by default: https://github.com/MichaIng/DietPi/commit/53651b0e482a5921898474c9bb9b522287e478ed

I rechecked on Buster and hotkey setup wasn't working there either, so this makes sense on all distro versions. Many thanks for reporting those issues and finding solutions as well 👍.

closed 2021-09-12 by MichaIng

melonius, 2022-03-25:
try "sudo apt install pipanel"
it works for me.

MichaIng, 2022-03-25:
This was not about the "RPi desktop" but about a regular LXDE on RPi, i.e. the command is `lxappearance`. The RPi desktop packages from the RPi repo are conflicting with a regular LXDE, but it all got fixed already last year.

MichaIng, 2024-03-04:
Reviewing some dependencies, I rechecked whether it makes sense to ask adding `librsvg2-common` as hard dependency to icon theme packages which make much use of SVGs. Adwaita, Papirus and Breeze are all almost only SVG themes, so at first few it makes sense. At 2nd view: The question is whether all applications which make use of these icon themes make use of `librsvg2-common` as well. On the Gnome icon theme, there was no discussion about this, but on Papirus: https://bugs.debian.org/1029056

Two valid points:
1. An icon theme might be used with software which uses another library or has embedded SVG support.
2. Software which allows to select an icon theme cannot know which image format the icon theme uses.

So unless/until best practice has been declared somewhere, adding it as recommendation only seems sane. It is then on desktop environments or other parent packages, which add both, applications who require `librsvg2-common` to make use of SVGs, as well as SVG-providing icon themes, to declare the dependency for now, or on those who assemble the desktop images or installation scripts, like us.

Adwaita had it as hard dependency in the past, but degraded it to a recommendation, worrying about architectures which do not support `librsvg2-common`:
>   * debian/control.in: Downgrade librsvg2-common to recommends
    librsvg2-common is an arch:any package and is not built on all
    architectures as it's written in rust. This should allow one to install
    the package on the architectures without rust support

In the meantime, it is supported by all architectures supported by Debian, including Sid and all inofficial ports. The other arguments remain, hence I'm not going to open a report about this.