MichaIng/DietPi #4688: Fail to update dietpi to v7.5.2 rpi3B+

ezekini, 2021-08-25:
### Creating a bug report/issue
#### Details:
- Date           | Wed Aug 25 09:50:25 -03 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi-rpi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 255
- Software title | DietPi-Update
- 
#### Steps to reproduce:

1. Current version v.7.4.2.  When trying dietpi-update to update to v7.5.2 the following error occurs.

#### Additional logs:
```
Get:1 https://dtcooper.github.io/raspotify raspotify InRelease [1706 B]
Get:2 https://packages.sury.org/php stretch InRelease [6839 B]
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Hit:4 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Get:5 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Fetched 23.5 kB in 2s (9948 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2021-08-25:
same issue as this one https://github.com/MichaIng/DietPi/issues/4219
pls follow the Admin instruction on the top

MichaIng, 2021-08-25:
@ezekini 
As this should not happen anymore, before applying any fix, can you please paste the output of the following commands:
```sh
ls -l /etc/apt/trusted.gpg.d/dietpi-php.gpg
apt-key list '95BD4743'
apt-key list
ls -l /var/lib/apt/lists
```

ezekini, 2021-08-25:
Sorry, I already applied the suggested solution describen in #4249 . Pasting the results in case it is useful still:

```
root@DietPi-rpi:~# ls -l /etc/apt/trusted.gpg.d/dietpi-php.gpg
-rw-r--r-- 1 root root 1769 ago 25 10:18 /etc/apt/trusted.gpg.d/dietpi-php.gpg
root@DietPi-rpi:~# apt-key list '95BD4743'
pub   rsa3072 2019-03-18 [SC] [expires: 2024-02-16]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ unknown] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
sub   rsa3072 2019-03-18 [E] [expires: 2024-02-16]

root@DietPi-rpi:~# apt-key list
/etc/apt/trusted.gpg
--------------------
pub   rsa2048 2012-04-01 [SC]
      A0DA 38D0 D76E 8B5D 6388  7281 9165 938D 90FD DD2E
uid           [ unknown] Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>
sub   rsa2048 2012-04-01 [E]

pub   rsa2048 2012-06-17 [SC]
      CF8A 1AF5 02A2 AA2D 763B  AE7E 82B1 2992 7FA3 303E
uid           [ unknown] Raspberry Pi Archive Signing Key
sub   rsa2048 2012-06-17 [E]

pub   rsa2048 2014-08-04 [SC]
      3FA7 E032 8081 BFF6 A14D  A29A A6A1 9B38 D3D8 31EF
uid           [ unknown] Xamarin Public Jenkins (auto-signing) <releng@xamarin.com>
sub   rsa2048 2014-08-04 [E]

pub   rsa2048 2013-08-27 [SC]
      A236 C58F 4090 91A1 8ACA  53CB EBFF 6B99 D9B7 8493
uid           [ unknown] NzbDrone <contact@nzbdrone.com>
sub   rsa2048 2013-08-27 [E]

pub   rsa4096 2017-06-15 [SC]
      2CC9 B80F 5AE2 B7AC EFF2  BA32 0914 6F2F 7953 A455
uid           [ unknown] David Cooper <david@dtcooper.com>
sub   rsa4096 2017-06-15 [E]

pub   rsa1024 2007-04-29 [C]
      CF62 BC25 167F CF3E 0045  3A9C 249A 1A0D FDA5 DFFC
uid           [ unknown] Totally Legit Signing Key <mallory@example.org>

/etc/apt/trusted.gpg.d/dietpi-php.gpg
-------------------------------------
pub   rsa3072 2019-03-18 [SC] [expires: 2024-02-16]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ unknown] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
sub   rsa3072 2019-03-18 [E] [expires: 2024-02-16]

root@DietPi-rpi:~# ls -l /var/lib/apt/lists
total 12040
-rw-r--r-- 1 root root    25298 abr 12 12:02 archive.raspberrypi.org_debian_dists_stretch_InRelease
-rw-r--r-- 1 root root   151200 abr 12 12:02 archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages.xz
-rw-r--r-- 1 root root    38204 abr 12 12:02 archive.raspberrypi.org_debian_dists_stretch_ui_binary-armhf_Packages.xz
-rw-r--r-- 1 root root     2242 mar  2 15:26 download.mono-project.com_repo_debian_dists_raspbianstretch_InRelease
-rw-r--r-- 1 root root    41700 mar  2 15:26 download.mono-project.com_repo_debian_dists_raspbianstretch_main_binary-armhf_Packages.xz
-rw-r--r-- 1 root root     1706 ago  1 17:10 dtcooper.github.io_raspotify_dists_raspotify_InRelease
-rw-r--r-- 1 root root      500 ago  1 17:10 dtcooper.github.io_raspotify_dists_raspotify_main_binary-armhf_Packages.xz
-rw-r----- 1 root root        0 ago 25 10:18 lock
-rw-r--r-- 1 root root     6839 ago  3 10:13 packages.sury.org_php_dists_stretch_InRelease
-rw-r--r-- 1 root root   190976 ago  3 10:13 packages.sury.org_php_dists_stretch_main_binary-armhf_Packages.xz
drwx------ 2 _apt root     4096 ago 25 10:19 partial
-rw-r--r-- 1 root root    56892 sep  7  2019 raspbian.raspberrypi.org_raspbian_dists_stretch_contrib_binary-armhf_Packages.xz
-rw-r--r-- 1 root root    14972 ago 25 07:37 raspbian.raspberrypi.org_raspbian_dists_stretch_InRelease
-rw-r--r-- 1 root root 11664520 ago 23 19:26 raspbian.raspberrypi.org_raspbian_dists_stretch_main_binary-armhf_Packages.xz
-rw-r--r-- 1 root root    98928 ago 11  2020 raspbian.raspberrypi.org_raspbian_dists_stretch_non-free_binary-armhf_Packages.xz
-rw-r--r-- 1 root root     1360 feb  9  2019 raspbian.raspberrypi.org_raspbian_dists_stretch_rpi_binary-armhf_Packages.xz
```