MichaIng/DietPi #4695: Dietpi Approved docker images.

cocoflan, 2021-08-28:
Might be an idea to be able to choose docker images from the installer after installing docker or portainer. 

And not having to do a full install script that is directly on to the disk structure.

Greeting.

MichaIng, 2021-08-28:
You mean when e.g. selecting **Pi-hole** for installation, while Docker is already installed, to select between the Docker image and the native Pi-hole install?

cocoflan, 2021-08-28:
Yes, and other apps also, would it not be more save and optimal to make use of docker for some installers?

Greetings cocoflan

MichaIng, 2021-08-28:
> would it not be more save and optimal to make use of docker for some installers

I'm not sure what your idea of a Docker image/container is: Instead of the actual software/server, it ships and runs a whole OS (a small one usually, but still a whole Linux userland core) and sets up a bunch of wrappers and hooks to enable communication between host and container. So it is absolutely not "optimal" for resource usage, performance and the ability to configure or customise it. Especially on smaller SBCs, using Docker for everything will limit the amount of services you're able to run, or usage of them, compared to native installs. So it is the opposite of "minimal". And aside of that, if the Docker container is not maintained by the actual upstream software developers, the quality is not always great, is sometimes (due to Docker needs) highly customised and hence not always well supported by upstream developers.

> Less risk to damage the system no?

Yes, before you follow random guides as inexperienced user, installing a Docker image is less risky, and often easier to setup. But `dietpi-software` exists exactly for the reason to save you from the need to worry about compatibility, basic setup and configuration, accidental system damages and all that. So compared to a `dietpi-software` native installation, the related Docker images have mostly downsides, as long as you don't have a strong need to have it isolated for specific reasons. If that was not true (it may be not in every case), then we'd need to enhance our install code to make it true, as it is definitely possible 😉.

There are of course some good reasons  to prefer a Docker container. E.g. compared to APT packages, the software version may be newer, or you explicitly need the isolation as a security measure or the ability to remote control a server stack via API, Portainer, the ability to backup/restore/update the individual server/software stack (Docker image) or so.

Since the Docker images are sometimes highly customised compared to the original software setup, our documentation would often be invalidated, or we'd need to maintain two docs. And installing a Docker container is very easy in very most cases. So I don't see much point in adding these easy steps, which everyone can run on the console (or use Portainer for this), to `dietpi-software`. It's better to have a proper guide about how to use Docker then.

So nope, this request is not something I find reasonable and would consider. I btw don't want to discredit Docker here, the more I learn about it the more I see its value, and hence find it necessary to have in `dietpi-software`, including Docker Compose and Portainer as great extensions. But when considering a minimal optimised resource friendly software install for an SBC with limited disk, RAM and CPU capabilities, or furthermore a system on an SD card where degradation due to disk writes is still a major issue, people get the wrong impression and overrate it. I had insights in quite a few Docker images/sources in the meantime, so I can say all that with some expertise, while earlier it was more based on logic only 😉.

cocoflan, 2021-08-28:
Ok I understand now that this indeed may not be an option, was just an idea of me, thx for the clear info.