MichaIng/DietPi #4697: Error while performing update DietPi from v7.4.2 to v7.5.2

rsmith98146, 2021-08-28:
### Creating a bug report/issue
[DietPi_Error.txt](https://github.com/MichaIng/DietPi/files/7071647/DietPi_Error.txt)

This happened when trying to update to the newest version of DietPi.  Thank you.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...dietpi-update
2. ...

#### Expected behaviour

- ...should update and restart

#### Actual behaviour

- ...error and stop

#### Extra details

- ...all requested info and error report is in attached file.

ravenclaw900, 2021-08-28:
This is a known issue, but it should have been fixed by updating to v7.4.  The fix is in https://github.com/MichaIng/DietPi/issues/4219, however before running it could you post the output of:
```
ls -l /etc/apt/trusted.gpg.d/dietpi-php.gpg
apt-key list '95BD4743'
apt-key list
ls -l /var/lib/apt/lists
```
Also, your system is running Debian Stretch, which is now `oldoldstable`, so we recommend upgrading to Debian Buster at least.

rsmith98146, 2021-08-29:
Thank you, 

I'm running 
Update available:                                                                                │
│  - Current version : v7.4.2                                                                      │
...and trying to update to
│  - Latest version  : v7.5.2

This happened when I tried to run the command dietpi-update instead of apt update/apt upgrade.

I am 1000% a novice at this.  I run dietpi because it was/is an easy and stable platform for my Plex server.  I'm hoping to start working on some other projects.  This is a learning experience for me.  

At one point in the updates I thought dietpi had updated to buster.  It surprised me that it was stretch.  I wouldn't have a clue on how to update that because of the before mentioned dependency on the dietpi structure.

Again thank you for your time and the answers to your questions are below.

root@DietPi:~# ls -l /etc/apt/trusted.gpg.d/dietpi-php.gpg
ls: cannot access '/etc/apt/trusted.gpg.d/dietpi-php.gpg': No such file or directory

root@DietPi:~# apt-key list '95BD4743'
pub   rsa3072 2019-03-18 [SC] [expired: 2021-03-17]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ expired] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>

root@DietPi:~# apt-key list
/etc/apt/trusted.gpg
--------------------
pub   rsa2048 2013-03-21 [SC]
      6D88 237D ABB2 1B75 59A1  B43E E13D 88F7 E3C1 D56C
uid           [ unknown] Linaro repositories

pub   rsa4096 2018-03-22 [SC] [expires: 2026-03-20]
      0077 2D08 7E7A 854A C067  8857 64AB E672 CC01 888F
uid           [ unknown] TinkerBoard Release Key (9/stretch) <tinkerboard@asus.com>

pub   rsa3072 2019-03-18 [SC] [expired: 2021-03-17]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ expired] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>

pub   rsa4096 2015-03-22 [SC]
      CD66 5CBA 0E2F 88B7 373F  7CB9 9720 3C7B 3ADC A79D
uid           [ unknown] Plex Inc.
sub   rsa4096 2015-03-22 [E]

/etc/apt/trusted.gpg.d/debian-archive-buster-automatic.gpg
----------------------------------------------------------
pub   rsa4096 2019-04-14 [SC] [expires: 2027-04-12]
      80D1 5823 B7FD 1561 F9F7  BCDD DC30 D7C2 3CBB ABEE
uid           [ unknown] Debian Archive Automatic Signing Key (10/buster) <ftpmaster@debian.org>
sub   rsa4096 2019-04-14 [S] [expires: 2027-04-12]

/etc/apt/trusted.gpg.d/debian-archive-buster-security-automatic.gpg
-------------------------------------------------------------------
pub   rsa4096 2019-04-14 [SC] [expires: 2027-04-12]
      5E61 B217 265D A980 7A23  C5FF 4DFA B270 CAA9 6DFA
uid           [ unknown] Debian Security Archive Automatic Signing Key (10/buster) <ftpmaster@debian.org>
sub   rsa4096 2019-04-14 [S] [expires: 2027-04-12]

/etc/apt/trusted.gpg.d/debian-archive-buster-stable.gpg
-------------------------------------------------------
pub   rsa4096 2019-02-05 [SC] [expires: 2027-02-03]
      6D33 866E DD8F FA41 C014  3AED DCC9 EFBF 77E1 1517
uid           [ unknown] Debian Stable Release Key (10/buster) <debian-release@lists.debian.org>

/etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg
----------------------------------------------------------
pub   rsa4096 2014-11-21 [SC] [expires: 2022-11-19]
      126C 0D24 BD8A 2942 CC7D  F8AC 7638 D044 2B90 D010
uid           [ unknown] Debian Archive Automatic Signing Key (8/jessie) <ftpmaster@debian.org>

[... 32 lines cut ...]
root@DietPi:~# ls -l /var/lib/apt/lists
total 42172
-rw-r--r-- 1 root root    34629 Sep 11  2020 deb.debian.org_debian_dists_stretch-backports_contrib_binary-armhf_Packages
-rw-r--r-- 1 root root    91837 Aug 28 13:14 deb.debian.org_debian_dists_stretch-backports_InRelease
-rw-r--r-- 1 root root  3309105 Nov  9  2020 deb.debian.org_debian_dists_stretch-backports_main_binary-armhf_Packages
-rw-r--r-- 1 root root    83423 Jul 13  2020 deb.debian.org_debian_dists_stretch-backports_non-free_binary-armhf_Packages
-rw-r--r-- 1 root root   185482 Jul 18  2020 deb.debian.org_debian_dists_stretch_contrib_binary-armhf_Packages
-rw-r--r-- 1 root root 37779475 Jul 18  2020 deb.debian.org_debian_dists_stretch_main_binary-armhf_Packages
-rw-r--r-- 1 root root   282827 Jul 18  2020 deb.debian.org_debian_dists_stretch_non-free_binary-armhf_Packages
-rw-r--r-- 1 root root   117950 Aug 14 00:43 deb.debian.org_debian_dists_stretch_Release
-rw-r--r-- 1 root root     3177 Aug 14 01:58 deb.debian.org_debian_dists_stretch_Release.gpg
-rw-r--r-- 1 root root    93595 Aug 14 00:47 deb.debian.org_debian_dists_stretch-updates_InRelease
-rw-r--r-- 1 root root     4842 Mar 10  2019 deb.debian.org_debian-security_dists_stretch_updates_contrib_binary-armhf_Packages
-rw-r--r-- 1 root root    53021 Aug 28 16:01 deb.debian.org_debian-security_dists_stretch_updates_InRelease
-rw-r--r-- 1 root root   686488 Aug 28 16:01 deb.debian.org_debian-security_dists_stretch_updates_main_binary-armhf_Packages.xz
-rw-r--r-- 1 root root    22535 Aug 11  2020 deb.debian.org_debian-security_dists_stretch_updates_non-free_binary-armhf_Packages
-rw-r--r-- 1 root root     6685 Aug 23 09:14 downloads.plex.tv_repo_deb_dists_public_InRelease
-rw-r--r-- 1 root root      432 Aug 23 09:14 downloads.plex.tv_repo_deb_dists_public_main_binary-armhf_Packages.xz
-rw-r----- 1 root root        0 Nov 11  2020 lock
-rw-r--r-- 1 root root     6824 Mar 16 03:26 packages.sury.org_php_dists_stretch_InRelease
-rw-r--r-- 1 root root   188732 Mar 16 03:26 packages.sury.org_php_dists_stretch_main_binary-armhf_Packages.xz
drwx------ 2 _apt root    12288 Aug 28 16:07 partial
-rw-r--r-- 1 root root    15468 Nov 13  2017 repo.linaro.org_ubuntu_linaro-overlay_dists_stretch_InRelease
-rw-r--r-- 1 root root   119762 Nov 13  2017 repo.linaro.org_ubuntu_linaro-overlay_dists_stretch_main_binary-armhf_Packages
-rw-r--r-- 1 root root    43539 Nov 13  2017 repo.linaro.org_ubuntu_linaro-overlay_dists_stretch_main_source_Sources

Joulinar, 2021-08-29:
First of all, DietPi will never trigger any Debian version upgrade. This would need to be done by yourself. Currently we are working on a online docs entry on how to upgrade to Buster. Next to it we prepare a blog post for Bullseye as well.

To fix your issue, just do

```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
/boot/dietpi/func/dietpi-set_software apt-cache clean # clean the apt cache
apt update # update again the package cache
```

MichaIng, 2021-08-29:
To help us debugging why it was not solved on an earlier update already, could you please try to sone it via:
```sh
bash -xc "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
```
This runs the pre-patch script of our updater in verbose mode, so we can see if/why the condition for updating this key is not key.

Joulinar, 2021-08-29:
@MichaIng 
There was another case in our forum yesterday. But already fixed.

MichaIng, 2021-08-29:
Hence the above test of the pre-patch script to find it out 🧐.