MichaIng/DietPi #4699: [Question] updates with custom packages + dietpi-update

G2G2G2G, 2021-08-30:
>Although we test the updates thoroughly, if you have made any custom changes to Linux configuration files outside of the DietPi programs, an update may trigger a potential issue. 

What does this mean? I can't find any info on it.
Does this mean that stuff we `apt install ___` will get messed up?
or does it only mean if we `apt install ___` and then install it through the diet-pi installer it'll get messed up there if we double -installed it or whatever?

Thanks

Joulinar, 2021-08-30:
It means, if you have done own configuration, installed 3rd party tools, installed software not using apt aso or wrote own scripts. We are not able to take any guarantee as our update will trigger a apt update as first step and this might cause challenges with your own software or scripts. It simply means we are not able to test all kind software installations and combination that are available worldwide. We focus on the software that is available via our own installer. But we can't give a 100% guarantee.

MichaIng, 2021-08-30:
I think we should change that wording in the updater script: Custom configuration changes are actually not an issue (anymore), as we apply patches very carefully to preserve customisations or otherwise with an interactive information or choice prompt. Most of the configuration files DietPi itself installs are now separate override configs, i.e. shipped as `*.conf.d/dietpi-*.conf`, so that either changed Debian/package defaults or changed admin overrides are not an issue. And in cases where our scripts are very intrusive and fully overwrite configuration files, like `dietpi-drive_manager` or the `dietpi-config` network setup, those are never executed, and if we'd consider, then never without interactive admin permission. Also removed configuration lines are not an issue anymore, which still was some years ago.

`apt install` packages should be even less of an issue, as they are forced by Debian policy to use own dedicated configurations and files, never automatically override edited config files (those in `/etc`), be compatible in any possible way. APT is a core element of Debian and we encourage everyone to use it to install software. We explicitly do not implement software into `dietpi-software` when its sufficiently easy to install it via APT and there is no further setup required in our view, so please never hesitate to use it like you would do on any other Debian based distribution.

APT/DEB packages shipped by Debian in a stable release are also by policy guaranteed to not break anything, ship security patches and important fixes or tiny setup enhancements only. The vast reason for issues related to APT package upgrades is coming from 3rd party repositories, which includes the Raspberry Pi and Armbian kernel/firmware repositories, and those for 3rd party software, as here the Debian policy does not apply and things have been usually less tested. I remember broken Sonarr/Radarr/Lidarr in combination with a Mono major version upgrade, and bugs shipped with kernel upgrades by times, across all SBCs. Only on `x86_64`, which uses a Debian kernel with fixed major.minor version and their multi-stage review and testing stages, I can't remember any issues in combination with a kernel package upgrade. But still, such introduced kernel/firmware bugs usually affect very specific features only. 

The number one reason (or equal to kernel upgrades on SBCs) for issues during DietPi updates (@Joulinar correct me in case) are already existing issues, indirectly triggered by either the disk writes during larger package upgrades (undervoltage, disk wear, filesystem corruption becoming apparent) or by the service restarts or the reboot done during/after the update, where an already present issue with the configuration/setup or the network or so starts to take effect.

So IMO the best way to prevent, respectively narrow down/debug issues is to do `apt upgrade` regularly independently of DietPi updates, a `dietpi-backup` (or other kind of system backup) before doing the update, and in case of any issue/error having a look into the console or APT log to identify potentially relevant packages.

So how to put these things into 1-2 sentences for `dietpi-update` now? 😄

Joulinar, 2021-08-30:
@MichaIng 
that's correct. majority of issues is coming from area you describe (under voltage, disk wear, filesystem corruption). Usually they are triggered by `apt upgrade`

G2G2G2G, 2021-08-31:
Yea my compiled software is in /opt/ which I'm not worried about. 
This message always concerned me that stuff like `syncthing` from your packages, with my custom config in home user folder: `.config/syncthing/config.xml` (default) was going to get overwritten or something (the way it's worded makes it sound like that) 

But from the posts above it sounds like the message doesn't even mean that specifically

This is what I'd normally expect (example is normal apt upgrade output):
```
Configuration file '/etc/matrix-synapse/homeserver.yaml'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** homeserver.yaml (Y/I/N/O/D/Z) [default=N] ? 
```

closed 2021-08-31 by G2G2G2G

MichaIng, 2021-08-31:
> This is what I'd normally expect

Yes, APT packages behave anyway like this. Though on DietPi by default it's non-interactively answered with the "default" answer (hence "No" in all cases I know) and "No" if there is no default.

In all other cases where its not about an APT/DEB package, we basically follow the same behaviour, in case with an interactive whiptail prompt, similar to debconf prompts.