MichaIng/DietPi #4703: {Software install} Node-Red V2 install failure

MDAR, 2021-09-01:
#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | 10.10
- Kernel version | Linux openHAB2-C4 4.9.241-arm64 # 1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux
- SBC model | Odroid C4/HC4 (aarch64)
- Power supply used | 5V 2A 
- SDcard used |  eMMC 16GB

#### Additional Information (if applicable)
- Software title | Node-RED
- The software title was being updated from V1.3.4 to whatever the latest is (I think this is V2.0)
- Can this issue be replicated on a fresh installation of DietPi?
I will try tomorrow

- Bug report ID | e9065ed8-76b5-47f2-9cc1-3b8849e8d6ca

#### Details:
- Date           | Wed Sep  1 17:34:32 BST 2021
- Bug report     | e9065ed8-76b5-47f2-9cc1-3b8849e8d6ca
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid C4/HC4 (aarch64) (ID=16)
- Kernel version | `Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `sudo -u nodered npm i --no-audit node-red`
- Exit code      | 1
- Software title | DietPi-Software
- 
#### Steps to reproduce

1.  Ran command `dietpi-software reinstall 122`
2. Also tried to select Node-RED from the software menu
3. Also tried to manually install npm (lots of things got installed, but Node-RED still failed to install, with the same error

#### Expected behaviour

- As per previous setups on various Odoird C4 machines, it should have installed happliy

#### Actual behaviour

- Install is failing 

#### Extra details

- ...

TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:94:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:94:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)
[... 2034 lines cut ...]
Universal Node.js Linux Installer by github.com/taaem, updated by github.com/MichaIng
root permissions verified
Searching latest stable version for aarch64 ...
Found latest stable version for aarch64: node-v16.8.0-linux-arm64.tar.gz
Downloading https://nodejs.org/dist/latest/node-v16.8.0-linux-arm64.tar.gz ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 31.4M  100 31.4M    0     0  8444k      0  0:00:03  0:00:03 --:--:-- 8441k
Finished downloading!
Installing /tmp/node.tar.gz ...
removed '/tmp/node.tar.gz'
Finished installing!
[  OK  ] DietPi-Software | ./node-install.sh
[  OK  ] DietPi-Software | rm node-install.sh

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Node-RED: tool for wiring devices, APIs and online services

[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/node-red
[  OK  ] DietPi-Software | Verifying user "nodered" to run Node-RED
[  OK  ] DietPi-Software | usermod -d /mnt/dietpi_userdata/node-red -s /usr/sbin/nologin nodered
[  OK  ] DietPi-Software | chown -R nodered:nodered /mnt/dietpi_userdata/node-red
[ INFO ] DietPi-Software | APT install for: python3, please wait...
```

jimmywarting, 2021-09-01:
can you make a smaller reproducible example. This is way too long

fyi, v3 switched to ESM only, you can no longer use require to load node-fetch
if you are stuck with cjs then it would be best to stick with v2

MichaIng, 2021-09-01:
Since it is installed via npm, it should be v2.0.5: https://www.npmjs.com/package/node-red
This is also the latest official release: https://nodered.org/
Never heard of v3 yet?

However, the error happens when `socks-proxy-agent` is installed as dependency of `make-fetch-happen`, if I understand it right, so it has not directly something to do with Node-RED.

I'll test it on VM.

MDAR, 2021-09-01:
Hi

thanks for looking at this go quickly.

All I have done is try to update Node-Red using 

`dietpi-software reinstall 122`

Who knows what I have done with the machine,

I'll put a fresh copy of DietPi on it tomorrow and start again.

They only run
* velbus-tcp (snap) - https://github.com/velbus/velbus-tcp-snap
* node-red
* openhab3 - http://www.openhab.org

So not a big deal to start from scratch

Do you want me to test anything before I wipe this machine?

MDAR, 2021-09-01:
FYI

Just tried that command again, and I just see the following after it looks like it is going to work

```
 Command: sudo -u nodered npm i --no-audit node-red                                                                │
             │  - Exit code: 1                                                                                                      │
             │  - DietPi version: v7.5.2 (MichaIng/master) | HW_MODEL: 16 | HW_ARCH: 3 | DISTRO: 5                                  │
             │  - Image creator: DietPi Core Team                                                                                   │
             │  - Pre-image: Meveric                                                                                                │
             │  - Error log:                                                                                                        │
             │ TypeError: Class extends value undefined is not a constructor or null                                                │
             │     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44)      │
             │     at Module._compile (node:internal/modules/cjs/loader:1101:14)                                                    │
             │     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)                                      │
             │     at Module.load (node:internal/modules/cjs/loader:981:32)                                                         │
             │     at Function.Module._load (node:internal/modules/cjs/loader:822:12)                                               │
             │     at Module.require (node:internal/modules/cjs/loader:1005:19)                                                     │
             │     at require (node:internal/modules/cjs/helpers:94:18)                                                             │
             │     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33)        │
             │     at Module._compile (node:internal/modules/cjs/loader:1101:14)                                                    │
             │     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)                                      │
             │     at Module.load (node:internal/modules/cjs/loader:981:32)                                                         │
             │     at Function.Module._load (node:internal/modules/cjs/loader:822:12)                                               │
             │     at Module.require (node:internal/modules/cjs/loader:1005:19)                                                     │
             │     at require (node:internal/modules/cjs/helpers:94:18)                                                             │
             │     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25)       │
             │     at Module._compile (node:internal/modules/cjs/loader:1101:14)

```

It is always this command that throws a tantrum

`sudo -u nodered npm i --no-audit node-red`

I'll come back to it tomorrow

MDAR, 2021-09-02:
Thank you so much for your time and guidance, in just 5 minutes Node-RED was installed again. (With Version 2.0.6)

All I did was the following

Boot up

```
rm -R /usr/local/lib/node_modules
rm -R /usr/local/include/node
rm -R /mnt/dietpi_userdata/node-red/{node_modules,package-lock.json,package.json,.cache,.npm} [Although this folder was not present]

apt purge --autoremove npm
dietpi-software install 122
```

I did try `dietpi-software reinstall 122` but that reported "no changes" as node-red had been removed.

Thank you again.

a quick edit of the `/mnt/dietpi_userdata/node-red/settings.js` and import of a backup and everything was back to how it should be

MichaIng, 2021-09-02:
`dietpi-software reinstall 122` instead of `dietpi-software install 122` would have worked.

MDAR, 2021-09-02:
> `dietpi-software reinstall 122` instead of `dietpi-software install 122` would have worked.

😄 

Either way, I'm delighted that it all back up and running.

(Now I just have to find out if there are any dramatic changes in Node-RED v2 that will affect any of my production machines)

Thanks again 😄

MDAR, 2021-09-16:
Hi @MichaIng 

I'm really sorry about this.

I'm trying to update another machine and this time the `dietpi-software install 122` process is halting when it comes to downloading this file
```https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh```

It seems to be unavailable

Joulinar, 2021-09-16:
could you try to execute following on your system

```
cd /tmp
curl -ILfvm 5 https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh
```

MDAR, 2021-09-16:
> curl -ILfvm 5 https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh

😢 

This is what I get

```
/tmp# curl -ILfvm 5 https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh
*   Trying 185.199.110.133...
* TCP_NODELAY set
* Connected to raw.githubusercontent.com (185.199.110.133) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* Operation timed out after 5001 milliseconds with 0 out of 0 bytes received
* Curl_http_done: called premature == 1
* stopped the pause stream!
* Closing connection 0
curl: (28) Operation timed out after 5001 milliseconds with 0 out of 0 bytes received

```

MDAR, 2021-09-16:
😢 

Now this is super odd.

If I put my laptop onto the mobile hotspot of my phone, I pull down the node-install.sh file from github in a browser.

But it seems the ISP is blocking traffic from GitHub, or at least certain pages.

As far as a I know, there isn't a firewall in place, I can ask tomorrow when their office opens.

Joulinar, 2021-09-16:
the script is simply executed as far as I can see.

MDAR, 2021-09-16:
> the script is simply executed as far as I can see.
> 
> > Is the issue anything to do with the fact that this box is running DietPi 6.34.3 ?
> > Is there an SSL certificate issue at play?
> 
> If I understood correctly, you have same issue putting the URL into a browser on your laptop. Means it is not dedicated on your DietPi device

I'm very quickly coming to the point where I agree with you 100%

I'm also trying to load the ESP32 board details into Ardunio on my laptop and part of that process is being blocked too.

But again, if I connect my laptop to my phone's hot-spot, it downloads.

I'll call the ISP tomorrow and see what is happening,
(I know that where the victims of a DoS attack recently, so maybe things are locked down a little right now)

Thanks very much for your help, I really do appreciate it.

MDAR, 2021-09-17:
@Joulinar 

Thanks for helping last night and keeping my sanity in check.

It turns out that there was an issue with the ISP and this got me to change the MTU value this morning and everything has started to work normally again.

closed 2021-09-17 by MDAR