MichaIng/DietPi #4704: apt update fails with no space left on /boot

photoangell, 2021-09-02:
#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | stretch
- Kernel version | Linux Alfred 4.14.231+ #1 SMP PREEMPT Fri Apr 16 13:31:42 UTC 2021 armv7l GNU/Linux
- SBC model | ODroid XU4
- Power supply used | Odroid psu
- SDcard used | Odroid eMMC 16gb

#### Steps to reproduce
run apt-get update

#### Actual behaviour
```
Selecting previously unselected package linux-image-4.14.241+.
Preparing to unpack .../37-linux-image-4.14.241+_4.14.241-20210728-xu4_armhf.deb ...
find: ‘/media/boot’: No such file or directory
Unpacking linux-image-4.14.241+ (4.14.241-20210728-xu4) ...
dpkg: error processing archive /tmp/apt-dpkg-install-ikxyj5/37-linux-image-4.14.241+_4.14.241-20210728-xu4_armhf.deb (--unpack):
 cannot copy extracted data for './boot/System.map-4.14.241+' to '/boot/System.map-4.14.241+.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
```

#### Extra details
df -h gives me
```
Filesystem                      Size  Used Avail Use% Mounted on
udev                            931M     0  931M   0% /dev
tmpfs                           200M   28M  172M  14% /run
/dev/mmcblk0p2                   15G  5.2G  8.3G  39% /
tmpfs                           998M     0  998M   0% /dev/shm
tmpfs                           5.0M  4.0K  5.0M   1% /run/lock
tmpfs                           998M     0  998M   0% /sys/fs/cgroup
192.168.1.1:/mnt/backup/media/  126G  3.6G  116G   4% /mnt/nfs_router
/dev/sda1                       5.5T  2.7T  2.9T  49% /mnt/12ce5195-90e4-4ff2-b310-cb4609f05582
tmpfs                            50M   16K   50M   1% /var/log
tmpfs                          1023M   12K 1023M   1% /tmp
/dev/mmcblk0p1                   71M   70M  1.2M  99% /boot
//192.168.1.1/media             126G   10G  116G   8% /mnt/media
overlay                         5.5T  2.7T  2.9T  49% /mnt/12ce5195-90e4-4ff2-b310-cb4609f05582/dietpi_userdata/docker-data/overlay2/6ae69764151b972cde4126ef984a9ed82c473fe9f6aa46ef61fdb6b08bd433b8/merged
overlay                         5.5T  2.7T  2.9T  49% /mnt/12ce5195-90e4-4ff2-b310-cb4609f05582/dietpi_userdata/docker-data/overlay2/3a65e29b9085566053ca3fb943a19922d29b9fdbbb193790b5b064bf2c421821/merged
```

i tried apt purge linux-image-4.14.66+
but I get
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 linux-image-4.14-armhf-odroid-xu4 : Depends: linux-image-4.14.241+ but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Joulinar, 2021-09-02:
what about running `apt autopurge` ??

Another option is to move all files ending `4.14.66+` into a temp folder and run `apt --fix-broken install`

photoangell, 2021-09-02:
autopurge is not available, I have autoclean and autoremove, but not autopurge.

I will try moving the files...

Joulinar, 2021-09-02:
`apt autoremove` should be working same

photoangell, 2021-09-02:
ok, I moved the files ending ```4.14.66+``` to a temp folder, tried ```apt --fix-broken install``` and I ran out of space again.
just here...
```
Processing triggers for initramfs-tools (0.130) ...
ln: failed to create hard link '/boot/initrd.img-4.14.241+.dpkg-bak' => '/boot/initrd.img-4.14.241+': Operation not permitted
cp: error writing '/boot/initrd.img-4.14.241+.dpkg-bak': No space left on device
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
```

ls -al now says
```
total 56533
drwxr-xr-x  3 root root   16384 Sep  3 00:14 .
drwxr-xr-x 22 root root    1024 Jan 28  2021 ..
-rwxr-xr-x  1 root root    5447 May 13  2020 boot.ini
-rwxr-xr-x  1 root root  151306 Apr 16 14:34 config-4.14.231+
-rwxr-xr-x  1 root root  151341 Jul 28 17:57 config-4.14.241+
drwxr-xr-x  4 root root    4096 Jul  8 19:49 dietpi
-rwxr-xr-x  1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x  1 root root   11510 Jul  8 19:49 dietpi.txt
-rwxr-xr-x  1 root root   56387 Jul 28 18:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x  1 root root   63543 Jul 28 18:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x  1 root root   62488 Jul 28 18:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x  1 root root   63518 Jul 28 18:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x  1 root root   63662 Jul 28 18:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x  1 root root 5756412 Jul  8 19:44 initrd.img-4.14.231+
-rwxr-xr-x  1 root root 5755763 Sep  3 00:14 initrd.img-4.14.241+
-rwxr-xr-x  1 root root 2711917 Apr 16 14:34 System.map-4.14.231+
-rwxr-xr-x  1 root root 2728720 Jul 28 17:57 System.map-4.14.241+
-rwxr-xr-x  1 root root 5756476 Sep  3 00:14 uInitrd
-rwxr-xr-x  1 root root 5728441 Jul  8 19:44 uInitrd-4.14.180+
-rwxr-xr-x  1 root root 5756476 Sep  3 00:14 uInitrd-4.14.231+
-rwxr-xr-x  1 root root 5755769 Sep  3 00:11 uInitrd-4.14.241+
-rwxr-xr-x  1 root root 5731992 Apr 16 14:34 vmlinuz-4.14.231+
-rwxr-xr-x  1 root root 5764120 Jul 28 17:57 vmlinuz-4.14.241+
-rwxr-xr-x  1 root root 5764120 Sep  3 00:11 zImage
```

photoangell, 2021-09-02:
I subsequently tried ```apt autoremove```
and I got
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  linux-image-4.14.180+
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 58.6 MB disk space will be freed.
Do you want to continue? [Y/n]
(Reading database ... 110708 files and directories currently installed.)
Removing linux-image-4.14.180+ (4.14.180-20201013-xu4) ...
update-initramfs: Deleting /boot/initrd.img-4.14.180+
Setting up initramfs-tools (0.130) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
ln: failed to create hard link '/boot/initrd.img-4.14.241+.dpkg-bak' => '/boot/initrd.img-4.14.241+': Operation not permitted
update-initramfs: Generating /boot/initrd.img-4.14.241+
Image Name:   uInitrd
Created:      Fri Sep  3 00:14:45 2021
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    5756412 Bytes = 5621.50 kB = 5.49 MB
Load Address: 00000000
Entry Point:  00000000
W: APT had planned for dpkg to do more than it reported back (3 vs 7).
   Affected packages: initramfs-tools:armhf
```
and it seems to be ok now

photoangell, 2021-09-02:
should I move the ``` 4.14.66+ ``` files back into /boot ?

Joulinar, 2021-09-02:
No I don't think it is needed. I guess you could do a `reboot` now. Check again `apt update && apt upgrade` and once more `apt autoremove.` If this is fine check files on /boot and remove all files ending `4.14.180+` if available.

photoangell, 2021-09-02:
rebooted all ok. 
I removed the 180 files and also did
```
apt purge linux-image-4.14.180+
apt purge linux-image-4.14.66+
```

nice and clean now. thanks for all the help

Joulinar, 2021-09-03:
@MichaIng 
I guess kernel `4.14.231+` could be removed as well?

MichaIng, 2021-09-03:
Great it got solved. Not sure why the old kernel versions were not autoremoved before, but the package build stack got switched in the meantime, so on future kernel upgrades, the old one(s) should be autoremoved.

`apt autopurge` btw does only exist on Buster and up, not yet on Stretch. There `apt autoremove --purge` is the equivalent.

Generally if APT runs into these issues of wanting to finish newly installed package configurations before doing what you want to, DPkg itself can be used:
```sh
dpkg -P linux-image-4.14.231+
dpkg -P linux-image-4.14.180+
dpkg -P linux-image-4.14.66+
```
It denies to purge if another package depends on them, which shouldn't be the case here, but for completeness, also this can be overridden by adding the `--force-depends` flag.

Joulinar, 2021-09-03:
ah I totally overlooked it is Debian Stretch

@photoangell 
you are running oldold Debian version. You may consider to update to Debian Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

photoangell, 2021-09-03:
Thanks. It's been running rock solid since 2019 so it's due for a dist upgrade. A project for the weekend!!! I will upgrade to Buster and then Bullseye.

Thanks again for exhaustive help and information. Much appreciated.

Joulinar, 2021-09-04:
I'm going to close this one as original issue has been fixed. Feel free to reopen or create a new issue if you face challenges on the disp upgrade.

closed 2021-09-04 by Joulinar