MichaIng/DietPi #4705: Portainer install fails on Bullseye

simonfisher, 2021-09-02:
### Creating a bug report/issue

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | bullseye
- Kernel version | Linux DietPi 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | Odroid one
- SDcard used | eMMC

#### Additional Information (if applicable)
- Software title | Portainer
- Was the software title installed freshly or updated/migrated?
Installed on a prior DietPi install, then upgraded to bullseye, portainer wasn't working so attempted reinstall
- Can this issue be replicated on a fresh installation of DietPi? Unsure

- Bug report ID | 9cf73ff2-fbf5-4048-95e4-7f112c4f12c4

#### Steps to reproduce
1. in Buster, install Portainer
2. Upgrade from Buster to Bullseye using steps here: https://dietpi.com/blog/?p=811
3. Realise can't access Portainer via http address
4. Attempt reinstall of Portainer, using:  dietpi-software reinstall 185

#### Expected behaviour
Working Portainer...

#### Actual behaviour
This crash log:
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Portainer: Simplifies container management in Docker (standalone host)

[ INFO ] DietPi-Software | Docker will be restarted to be able to deploy the container.
[  OK  ] DietPi-Software | systemctl daemon-reload
[  OK  ] DietPi-Software | systemctl restart docker
[  OK  ] DietPi-Software | docker rm -f 8c756efbe038
[  OK  ] DietPi-Software | docker rmi bd3c978cdaec
[ INFO ] DietPi-Software | Portainer will be deployed now. This could take a while...
[ INFO ] DietPi-Software | docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce, please wait...
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
651a8e6e1630: Pulling fs layer
56e38df73332: Pulling fs layer
ffed15de3e09: Pulling fs layer
56e38df73332: Verifying Checksum
56e38df73332: Download complete
651a8e6e1630: Download complete
651a8e6e1630: Pull complete
56e38df73332: Pull complete
ffed15de3e09: Verifying Checksum
ffed15de3e09: Download complete
ffed15de3e09: Pull complete
Digest: sha256:8f077d2d1ba2e771ea8cc63af0a37d211f61354a0d094234f832e588ab571888
Status: Downloaded newer image for portainer/portainer-ce:latest
4d94617915bb9d2365a2b79383b69059a86d58321f34c8efe810d1bab93d7c5d
docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: invalid argument: unknown.
[FAILED] DietPi-Software | docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce

#### Extra details

- ...

simonfisher, 2021-09-02:
To add: this issue maybe the same or similar to that described here: https://archlinuxarm.org/forum/viewtopic.php?f=15&t=15354

An incompatibility somewhere with 'cgroup'...

Joulinar, 2021-09-03:
I guess kernel version is to low according this GitHub post https://github.com/opencontainers/runc/issues/2959#issuecomment-847521031

It seem 4.15 is required at least to be able to use for cgroup v2, But I guess there is no newer kernel than 4.14 for Odroid 

@MichaIng 
I guess we need your inside 😃

MichaIng, 2021-09-03:
Can you try to reinstall Docker to assure the packages from the Bullseye suite are installed:
```sh
apt install --reinstall docker-ce-cli containerd.io docker-ce
```
I do not believe yet that Docker now strictly requires Linux v4.15+ without any major version increment.

simonfisher, 2021-09-03:
That command doesn't fully work as it seems...

```
root@DietPi:~# apt install --reinstall docker-ce-cli containerd.io docker-ce
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reinstallation of docker-ce is not possible, it cannot be downloaded.
Reinstallation of docker-ce-cli is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 16.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 https://download.docker.com/linux/debian bullseye/stable armhf containerd.io armhf 1.4.9-1 [16.6 MB]
Fetched 16.6 MB in 1s (12.4 MB/s)         
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 43614 files and directories currently installed.)
Preparing to unpack .../containerd.io_1.4.9-1_armhf.deb ...
Unpacking containerd.io (1.4.9-1) over (1.4.9-1) ...
Setting up containerd.io (1.4.9-1) ...
```

This output above is from the second time I ran the command - the first time the output got refreshed by my silly iPad SSH client. So I'm not sure if containerd.io actually got updated. Either way, a reinstall of portainer gives the same error after this.

Are the reports that docker-ce and docker-ce-cli can't be downloaded indicative of a wider problem? 

I dont think I'm in a position/have time at the moment to do detailed testing with multiple SDs, kernel changes, and I also can't easily hook up a monitor etc. sorry.

simonfisher, 2021-09-03:
I can add though, I tried the fix in the link I posted above (appending systemd.unified_cgroup_hierarchy=0 to the line in boot.ini), and that allowed the reinstall of portainer to complete without error, and docker/portainer is running fine now!

Are there any negative implications likely to this change?

MichaIng, 2021-09-04:
Strange error with the package download, I don't remember if I have seen this already, and no reason given either. The packages are definitely there: https://download.docker.com/linux/debian/dists/bullseye/pool/stable/armhf/

Great find with the cgroup downgrade, I missed that one. So looks like Bullseyes systemd by default uses a new cgroup implementation/version which is not supported by Linux v4.14 or older yet. So we need to add this at best directly to all our images (via next update) where we know that the kernel version is that low currently.

The setting is described here: https://manpages.debian.org/bullseye/systemd-sysv/init.1.en.html#KERNEL_COMMAND_LINE
There is even a precise check whether the kernel supports this feature or not, so no need to know or guess which kernel version is installed and/or whether the feature has been backported etc:
```sh
grep cgroup2 /proc/filesystems
```
We should then also set `systemd.legacy_systemd_cgroup_controller=0` to force full legacy mode, else for systemd itself, cgroup2 would still be tried to use.
@simonfisher 
Probably you can currently even see a related error message at boot, when systemd tries to mount the cgroup2 hierarchy, which should then fail:
```sh
dmesg -l emerg,alert,crit,err
```

And while it seems to be not used by default on Buster, the kernel command line parameter is understood already, so we can add it to our default boot config files without need to worry that users face boot warnings when creating Buster images via PREP.

G2G2G2G, 2021-09-30:
>It seem 4.15 is required at least to be able to use cgroup v2, But I guess there is no newer kernel than 4.14 for Odroid

for what it's worth I've ran the 5.3.11 kernel for years on 3 HC2 devices just fine, 
`Linux d2 5.3.11-odroidxu4 #5.99.191113 SMP PREEMPT Wed Nov 13 08:51:20 CET 2019 armv7l GNU/Linux`

and this has 5.4 kernel https://www.armbian.com/odroid-hc1/  (as Michalng mentioned)

>If Docker still won't work, and you are in mood for some backups and testing, we may push the Linux v5.4 bump for Odroid XU4 (and alike): #3861

I can test with 1 of my 3 devices in many user production environment (they only use samba & nfs) is there an image available?

MichaIng, 2021-10-16:
The kernel command line arguments are applied on next update on future images, at least for devices where we know how to do it: https://github.com/MichaIng/DietPi/commit/372429025aadc6e7c303d4e1df9a2e7c9bf5f2e1

E.g. on NanoPi M2 there is no boot configuration file, so I'm no sure how to achieve it there. In those cases users are informed about the issue. It may leave the question open about how to fix it, but it is probably better to get issues about that question that such with "Docker fails to install".

MichaIng, 2021-10-16:
Wrong value to disable hybrid mode: https://github.com/MichaIng/DietPi/commit/73ecb61611ea17ce4650da14ab22f8e0f7514df0

Dammit, either the cgroupsv2 capability test is not precise or it alone is not the only issue. I installed Linux 4.9 on a Bullseye system, and even that it is <4.15, is shows:
```
# grep cgroup2 /proc/filesystems
nodev   cgroup2
```
Installing Docker works, but installing a container (Portainer tested) fails:
```
docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: invalid argument: unknown.
...
level=error msg="Handler for POST /v1.41/containers/32737e8533cff0b30d948a1e7fadfd06cb0615e8b6e256b091496ef4dc1d115f/start returned error: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: invalid argument: unknown"
```
I applied the kernel command line arguments and retried:
```sh
# grep 'cgroup2' /proc/filesystems
nodev   cgroup2
```
But Portainer installs and runs fine. Removing the arguments breaks Portainer startup again. Even on a Debian Stretch system (where that kernel comes from) the above shows this output, even that the kernel natively does not support it, but Portainer still installs and runs fine.

So generally the cgroup2 mount type is available on all (x86_64) kernel versions down til Stretch, and it can be mounted successfully. Whether or not it is mounted however depends on the Debian/systemd version:
#### Bullseye:
```
# df -aT | grep cgroup
cgroup2        cgroup2             0       0         0    - /sys/fs/cgroup
```
- Only cgroup2 is mounted by default.
#### Buster
```
# df -aT | grep cgroup
df: /proc/sys/fs/binfmt_misc: No such device
tmpfs          tmpfs        1021348      0   1021348   0% /sys/fs/cgroup
cgroup2        cgroup2            0      0         0    - /sys/fs/cgroup/unified
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/systemd
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/pids
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/cpu,cpuacct
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/freezer
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/devices
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/net_cls,net_prio
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/memory
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/rdma
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/blkio
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/perf_event
cgroup         cgroup             0      0         0    - /sys/fs/cgroup/cpuset
```
- cgroupsv1 is mounted to the default mount point, but v2 to the dedicated `/sys/fs/cgroup/unified`, which looks like this hybrid thing.
#### Stretch
```
# df -aT | grep cgroup
tmpfs          tmpfs         1026040       0   1026040   0% /sys/fs/cgroup
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/systemd
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/blkio
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/devices
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/memory
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/cpuset
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/net_cls,net_prio
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/pids
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/cpu,cpuacct
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/perf_event
cgroup         cgroup              0       0         0    - /sys/fs/cgroup/freezer
```
- Only v1 is mounted.

Now I guess those kernel command line arguments control whether systemd mounts v1, v2 or hybrid. And indeed:
- `systemd.unified_cgroup_hierarchy=0` disables the `cgroup2` mount at `/sys/fs/cgroup`, which is the default until Bullseye.
- `systemd.legacy_systemd_cgroup_controller(=1)` disables the hybrid mode `cgroup2` mount at `/sys/fs/cgroup/unified`, which is default only until Buster. This has of course no effect if the `cgroup2` mount at `/sys/fs/cgroup` is enabled.
- Otherwise on all Debian versions this can be changed freely and the mounts do not fail. But obviously cgroupsv2 cannot be successfully used with the v4.9 kernel, if fails on all Debian versions when manually installing that one.
- So the question is, how can be know whether the kernel supports is or not 🤔.

closed 2021-10-16 by MichaIng

G2G2G2G, 2021-10-17:
@MichaIng ohh I figured it out, I was asking for a new image a few times.. I always thought you guys used armbian kernels on some of your distros but it's only debian.. (I'm slow)

so I just have to take the current image and upgrade to bullseye. 
I'll do that tonight

MichaIng, 2021-10-18:
Armbian is also only Debian 😉. For Odroids we use Meveric's Odroid repository for kernel and some GPU accelerated libraries/packages: https://dietpi.com/meveric/
For most other SBCs (not RPi) we use Armbian's repository for kernel and firmware. But both has the Debian repository for 99.9% for available packages.