MichaIng/DietPi #4706: DietPi image based on jessie (still) available for sparkySBC?

seniorgod, 2021-09-03:
Dear all,

because i have to compile old software with gcc 4 i'd like to know if there is a old image available of DietPi / Jessie for sparky SBC.

Joulinar, 2021-09-03:
Hi,

these old images are not available anymore.

MichaIng, 2021-09-03:
We only host the Bullseye and Buster images. Probably you're able to install or at least manually extract and use the gcc-4 binaries from the Debian Jessie packages: http://ftp.debian.org/debian/pool/main/g/gcc-4.9/

seniorgod, 2021-09-04:
ok,

i asked because i thought there is an easy way to build kernel image for sparky. 
I thought about to include the the RealTime patch (no idea if it increase audio quality, on Raspi it do) 
That's why i asked for jessie and gcc4 because the downloaded Sparky/LINUX repo from git require it.

Meanwhile i found that there is a sparkysbc/sparky_sdk repo which seems to include a build setup to build the sparky Linux image via cross compiling on a ubuntu i386 computer.

My question is: 
How do you compile the sparky kernel?
With that ubuntu cross compile toolchain from sparky_sdk ? (the toolchain obviously include the correct gcc) 
Did you compile the kernel in a different way?
How do you configure the sparky kernel? Are Control groups and namespaces part of the kernel.
Is there any chance to include the RT patch?

MichaIng, 2021-09-04:
I'm afraid we cannot answer these questions and you should ask them on the Sparky SBC repositories or by contacting Allo directly. If you find a good way to compile the Sparky SBC kernel on an x86 machine, I would be highly interested in this as well, as there are some issues with our old kernel build which I would try to solve with a fresh build. Our kernel seems to include patches from Volumio and it is finally not transparent anymore which patches/commits went into it and which not.

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng