MichaIng/DietPi #4707: mpd crashes on DietPi (due to album art ?)

seniorgod, 2021-09-04:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

debian version 11.0

Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux

sparkySMB 

#### Additional Information (if applicable)
mpd installed from dietpi software list 

#### Steps to reproduce
since update to DietPi 7.5.x i faces problems with mpd
i stripped down everything in mpd to find out what it is.
mpd runs stable as long if i only load a playlists including radio streams, meaning playing mp3 from public url from local radio stations. 
The problems occur when i start to read in my local music. 
I read in music directly via smb 
     music_directory "smb://192.168.178.2/medien_read_only/squeeze 
when i update mpd via the update command to read in music directory, it starts to update the music directory add the music to the library. During the update process my iOS client for mpd starts to update/communicate with mpd. When that's start mpd crashes:

Sep 04 12:18 : client: [18] command returned 0
Sep 04 12:18 : client: [19] process command "status"
Sep 04 12:18 : client: [19] command returned 0
Sep 04 12:18 : client: [18] process command "albumart "FLAC_20101126/Tim Bendzko/[2019] Filter/" "0""
Freed frame ../../source3/param/loadparm.c:2713, expected ../../source3/param/loadparm.c:3939.
Freed frame ../../source3/libsmb/libsmb_context.c:139, expected ../../source3/param/loadparm.c:3939.
Freed frame ../../source3/libsmb/libsmb_context.c:47, expected ../../source3/param/loadparm.c:3939.
no talloc stackframe at ../../source3/param/loadparm.c:847, leaking memory
Freed frame ../../source3/libsmb/libsmb_context.c:744, expected ../../source3/param/loadparm.c:2645.
Bad talloc magic value - unknown value

perhaps it relates to a "album art" error which is discussed at GitHub at the mpd project 

when i start mpd again, i have to delete the DB and and comment the music directory in order to get back a stable mpd

#### Expected behaviour
mpd should read in my music directory from smb like before

seniorgod, 2021-09-04:
since mpd 0.22.11. there seem to be a fix 

....
ver 0.22.11 (2021/08/24)
* protocol
  - fix "albumart" crash
 ......

is it possible to update in Dietpi 7.5.2 to mpd. 0.22.11 ?

MichaIng, 2021-09-04:
We install the package from Debian: https://packages.debian.org/bullseye/mpd
So we'd need to report this bug in ask for cherry picking this patch into stable Debian. I guess it is a specific album art type or affects only FLAC or so, that it hasn't been recognised yet: https://bugs.debian.org/cgi-bin/pkgreport.cgi?archive=0;dist=unstable;ordering=normal;repeatmerged=0;src=mpd

MichaIng, 2021-11-09:
Hmm, v0.22.11 fixed a bug that was introduced with a feature v0.22.10:
- https://github.com/MusicPlayerDaemon/MPD/issues/1224
- https://github.com/MusicPlayerDaemon/MPD/issues/1225
- https://github.com/MusicPlayerDaemon/MPD/commit/cfe024e

So this is not the right fix. Were these the issues you meant or which one do you mean?

Actually the final issue is with Samba/`libsmbclient`. Check for `Bad talloc magic value - unknown value` in search machines, and you'll get a lot of matching results around Samba. It uses `libtalloc2` for memory allocation and something seems to be going wrong, leading to the logged memory leaks. But of course it could also be related to the way MPD invokes `libsmbclient`. Also the proceeding `Freed frame` logs come from MPD, while most deal with code in `libsmb_context.c`.

Another factor in your case is that Sparky SBC uses a very old Linux v3.10.38. Not sure, but for people with deeper knowledge for debugging it, this may be an important info as well.

I think to first get some clarification where exactly it is coming from, which MPD code is called and how it invokes `libsmbclient` with this FLAC file albumart, it makes sense to report this at the MPD repository: https://github.com/MusicPlayerDaemon/MPD/issues

seniorgod, 2021-11-12:
Thanks for the hint to smbclient
I connected the external drive via smb with Dietpi-drive-manager to the filesystem.
I changed the config of mpd to use the local directory (connected via smb)
That solved the problem. 

thanxs

closed 2021-11-12 by seniorgod

MichaIng, 2021-11-12:
Ah, that is a good workaround indeed 👍.