MichaIng/DietPi #4708: Dietpi-Update: DietPi has encountered an error

azenkwed, 2021-09-05:
#### Details:
- Date           | Sun Sep  5 11:47:37 CEST 2021
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux Diepi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
WARNING: The following packages cannot be authenticated!
  libpcre3 libpcre2-8-0 libzip4 php-common
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```

MichaIng, 2021-09-05:
Many thanks for your report. Can you please run the following commands manually in your terminal and copy&paste the output here:
```sh
apt update
apt upgrade
```

MichaIng, 2021-09-18:
As of missing reply, I'll mark this as closed. If the issue persists, feel free to reopen.

closed 2021-09-18 by MichaIng