MichaIng/DietPi #4709: Home Assistant  - Supervised

HyperCriSiS, 2021-09-05:
Hi,

is there any chance that you can add a supervised Home Assistant installation?
[https://github.com/home-assistant/supervised-installer
](url)

You have much more possibilities with a supervised version :)

Regards,
Hyper

Joulinar, 2021-09-05:
Hi,

honestly I don't think so. Best to my knowledge it require `Network Manager`, which is not used on DietPi. I did some test installation in past. Just have a look into our forum https://dietpi.com/phpbb/viewtopic.php?p=29512#p29512

@MichaIng 
your view?

HyperCriSiS, 2021-09-05:
Maybe with a warning that some things are broken then?
HA is so restricted if not supervised :(

Joulinar, 2021-09-05:
@MichaIng 
could explain it better thy I. But if I'm not mistaken `Network Manager` is conflicting with `ifupdown` used by DietPi and it will break DietPi network configuration options.

HyperCriSiS, 2021-09-05:
The only solution would be a docker installation then?

Joulinar, 2021-09-05:
the installation test I did ended up in a mixture of docker container and components (Network Manager) outside the container. Just check the forum post I linked above.

HyperCriSiS, 2021-09-05:
OK, thank you! I will try the installation.

MichaIng, 2021-09-05:
This basically turns the OS into Home Assistant OS. I'm not sure about the benefit compared to flashing Home Assistant OS in the first place, it does not even support any other SBC then the ones where Home Assistant OS images are available for.

We ship Home Assistant (core) for doing what its supposed to do: Managing smart home devices. This is the only way we can guarantee compatibility with other software installed for their purpose, which is one of the main goals of DietPi. And I'm also not keen to have two different HA install options, as those would be the case: The way Home Assistant supervised is installed has nothing to do with the way Home Assistant core is installed, it is a complete different "product".

I can understand that it is convenient to have one web interface to manage everything, but often, the more it can do, the more it demands a specific setup and limits compatibility with other software. And finally there is nothing left for having DietPi as the whole system is turned into HA OS, and we had to deal with a lot of reports why this and that is not working anymore 😉.

MichaIng, 2021-09-05:
Btw, I just updated the FeatHub request, which still had the old name HASSIO: https://feathub.com/MichaIng/DietPi/+105

As it is a very frequently requested implementation, I'll keep this open and maybe I find time to do a test install and see whether we can make it work with ifupdown and limit or change its other demands. But as said, from all I read, even the statements of the HA team themselves, you should simply flash Home Assistant OS if you want this whole package.

HyperCriSiS, 2021-09-05:
Thank you! It would be so nice to have all HA features available on Dietpi.

HyperCriSiS, 2021-09-06:
> 
> 
> Hi,
> 
> honestly I don't think so. Best to my knowledge it require `Network Manager`, which is not used on DietPi. I did some test installation in past. Just have a look into our forum https://dietpi.com/phpbb/viewtopic.php?p=29512#p29512
> 
> @Michag
> your view?

The Installation worked very well. Unfortunately Pihole is broken then. No IP v4 Adpters bound and no valid ipv6 entries in setupVars.
Do you know if there is any possibility to manually fix this?

MichaIng, 2021-09-06:
> and we had to deal with a lot of reports why this and that is not working anymore 😉.

😉

Not having Network Manager installed in the first place would be required for us to even think about implementing it. I don't know about how Network Manager sets up interfaces by default and how HA Supervised may further configure and change this away from what our tools and Pi-hole expect or are able to deal with. To get info about the apparent network setup, use the following three commands to list adapter info, assigned IP addresses and routes:
```sh
ip l
ip a
ip r
```

Would be interesting what happened if you simply purge Network Manager again (breaks network connection at first) and let the hopefully still installed ifupdown again take over (after a reboot). Major question is whether HA Supervised actively makes use of NetMan, or whether only the installer wants it to have a consistent network stack across instances, while every other network stack works just fine. Docker at least sets up the host <> guest network independently.

HyperCriSiS, 2021-09-08:
> ```shell
> ip l
> ip a
> ip r
> ```

This is the output:
```

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:2f:5c:be brd ff:ff:ff:ff:ff:ff
3: hassio: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 02:42:90:2c:1e:8b brd ff:ff:ff:ff:ff:ff
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 02:42:db:88:8d:78 brd ff:ff:ff:ff:ff:ff
6: veth6064ee3@if5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP mode DEFAULT group default 
    link/ether 96:71:9d:86:d6:f7 brd ff:ff:ff:ff:ff:ff link-netnsid 0
8: vethde90985@if7: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP mode DEFAULT group default 
    link/ether 42:b0:03:a3:f8:b3 brd ff:ff:ff:ff:ff:ff link-netnsid 1
10: veth8bdbf21@if9: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP mode DEFAULT group default 
    link/ether 7e:dd:32:55:eb:28 brd ff:ff:ff:ff:ff:ff link-netnsid 1
12: veth7e8d1d0@if11: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP mode DEFAULT group default 
    link/ether ee:26:0f:80:b5:11 brd ff:ff:ff:ff:ff:ff link-netnsid 2
14: vethaf80a34@if13: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP mode DEFAULT group default 
    link/ether ee:21:4d:d7:37:2f brd ff:ff:ff:ff:ff:ff link-netnsid 3
16: veth5f5770a@if15: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP mode DEFAULT group default 
    link/ether da:66:73:93:41:58 brd ff:ff:ff:ff:ff:ff link-netnsid 4
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:2f:5c:be brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.103/24 brd 192.168.0.255 scope global dynamic noprefixroute eth0
       valid_lft 1778892sec preferred_lft 1778892sec
    inet6 2003:de:d71f:8f49:ac1b:ad60:b859:9cc7/64 scope global dynamic noprefixroute 
       valid_lft 7201sec preferred_lft 1sec
    inet6 2003:de:d719:2a5d:1286:a8d7:8b04:9030/64 scope global dynamic noprefixroute 
       valid_lft 172802sec preferred_lft 86402sec
    inet6 fe80::ab28:b984:9c54:e235/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: hassio: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    link/ether 02:42:90:2c:1e:8b brd ff:ff:ff:ff:ff:ff
    inet 172.30.32.1/23 brd 172.30.33.255 scope global hassio
       valid_lft forever preferred_lft forever
    inet6 fe80::42:90ff:fe2c:1e8b/64 scope link 
       valid_lft forever preferred_lft forever
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    link/ether 02:42:db:88:8d:78 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 fe80::42:dbff:fe88:8d78/64 scope link 
       valid_lft forever preferred_lft forever
6: veth6064ee3@if5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP group default 
    link/ether 96:71:9d:86:d6:f7 brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet6 fe80::9471:9dff:fe86:d6f7/64 scope link 
       valid_lft forever preferred_lft forever
8: vethde90985@if7: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP group default 
    link/ether 42:b0:03:a3:f8:b3 brd ff:ff:ff:ff:ff:ff link-netnsid 1
    inet6 fe80::40b0:3ff:fea3:f8b3/64 scope link 
       valid_lft forever preferred_lft forever
10: veth8bdbf21@if9: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP group default 
    link/ether 7e:dd:32:55:eb:28 brd ff:ff:ff:ff:ff:ff link-netnsid 1
    inet6 fe80::7cdd:32ff:fe55:eb28/64 scope link 
       valid_lft forever preferred_lft forever
12: veth7e8d1d0@if11: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP group default 
    link/ether ee:26:0f:80:b5:11 brd ff:ff:ff:ff:ff:ff link-netnsid 2
    inet6 fe80::ec26:fff:fe80:b511/64 scope link 
       valid_lft forever preferred_lft forever
14: vethaf80a34@if13: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP group default 
    link/ether ee:21:4d:d7:37:2f brd ff:ff:ff:ff:ff:ff link-netnsid 3
    inet6 fe80::ec21:4dff:fed7:372f/64 scope link 
       valid_lft forever preferred_lft forever
16: veth5f5770a@if15: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master hassio state UP group default 
    link/ether da:66:73:93:41:58 brd ff:ff:ff:ff:ff:ff link-netnsid 4
    inet6 fe80::d866:73ff:fe93:4158/64 scope link 
       valid_lft forever preferred_lft forever
default via 192.168.0.1 dev eth0 proto dhcp metric 100 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 
172.30.32.0/23 dev hassio proto kernel scope link src 172.30.32.1 
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.103 metric 100 
```

HyperCriSiS, 2021-09-08:
> ive used the install script on dietpi many times without issue. i always use dietpi as the base. with that said i've noticed on bullseye you have to be careful when making network changes after installing home assistant. make sure you dont say yes when asked if you want to uninstall un-needed things as it will uninstall network manager and it's dependencies and it isnt fun to fix

This sounds good! Can you maybe explain the exact steps how you successfully did it?

MichaIng, 2021-09-08:
> default via 192.168.0.1 dev eth0 proto dhcp metric 100

One thing is to assign your IP as static IP. Not sure whether NetMan took over and changed it to DHCP, and no idea how to set it up as static IP in NetMan, if so, though. That may already be the issue with Pi-hole, when the IP changed due to DHCP.

HyperCriSiS, 2021-09-08:
> One thing is to assign your IP as static IP. Not sure whether NetMan took over and changed it to DHCP, and no idea how to set it up as static IP in NetMan, if so, though. That may already be the [...]

ahhh, yes... This was changed, it was static before.

pbanj, 2021-09-08:
@HyperCriSiS I had to manually grab netman, and pretty much all it's dependencies and their dependencies. I then moved them into the cache location for apt and then used apt to install everything as it was easier than trying to do all the deb packages. 

I had reinstalled dietpi a few times because of this issue until I noticed exactly what was causing it, then I went about fixing it. After I was done I took a full image of my sdd so I can just restore to an already all set up state if I ever have to reinstall.

Also rebooting didn't help at all, and manually setting the network info in dietpi-config didn't help either.

Joulinar, 2021-09-08:
That's exactly the point. Installing netman will break`dietpi-config`

HyperCriSiS, 2021-09-08:
> @HyperCriSiS I had to manually grab netman, and pretty much all it's dependencies and their dependencies. I then moved them into the cache location for apt and then used apt to install everything as [...]

omg, I am not so familiar with Linux so I think I can not fix this. =D
Would it be possible that you share this image, if it is without personal information?

pbanj, 2021-09-08:
> omg, I am not so familiar with Linux so I think I can not fix this. =D
> Would it be possible that you share this image, if it is without personal information?

it has personal stuff in it as i run a bunch more than just home assistant on this pi. but i will see if i still have all the packages saved and ill zip them up and post them

pbanj, 2021-09-08:
ok i dont still have it. do apt install network-manager and give me the list of dependencies it say's missing. and i will grab them all and send it with instructions on how to do it @HyperCriSiS

HyperCriSiS, 2021-09-08:
> it has personal stuff in it as i run a bunch more than just home assistant on this pi. but i will see if i still have all the packages saved and ill zip them up and post them
Awesome, thank you so much!
I wonder how the deltion is done. 
I just had a look at the installation script, but I can't find anything related to deletion of packages.
https://raw.githubusercontent.com/Kanga-Who/home-assistant/master/supervised-installer.sh

pbanj, 2021-09-08:
> 
> 
> > it has personal stuff in it as i run a bunch more than just home assistant on this pi. but i will see if i still have all the packages saved and ill zip them up and post them
> 
> Awesome, thank you so much!
> I wonder how the deletion is done.
> I just had a look at the installation script, but I can't find anything related to deletion of packages.
> https://raw.githubusercontent.com/Kanga-Who/home-assistant/master/supervised-installer.sh

it isnt ha doing the uninstall. it is dietpi if you make changes to the network after installing ha. it will ask if you want to remove stuff you dont need anymore. saying yes is what makes it delete stuff. if you havent done too much on the system i would just grab a backup of your ha install and any other stuff you might need and reinstall dietpi as it is the easiest and fastest way to fix

Joulinar, 2021-09-08:
again guys, if you install HA supervised, you should not use dietpi-config anymore to adjust network details. This has been taken over by Network Manager. Hense nothing will be uninstalled. This is exactly the downside of HA supervised.

pbanj, 2021-09-08:
> 
> 
> again guys, if you install HA supervised, you should not use dietpi-config anymore to adjust network details. This has been taken over by Network Manager. Hense nothing will be uninstalled. This is exactly the downside of HA supervised.

ya you've said this. I'm simply trying to help him fix it now that it is broken, which is pretty obvious. 

As for a "fix" for dietpi maybe put a warning in the network section of dietpi-config telling people not to use it if they are using ha supervised.

HyperCriSiS, 2022-04-23:
The best installation method seems to be a container.
[https://www.home-assistant.io/installation/linux#install-home-assistant-container](url)

Can we maybe integrate this :grinning:?

Joulinar, 2022-04-23:
I guess you linked the HA version without the supervised stuff. Am I right?

HyperCriSiS, 2022-04-23:
> I guess you linked the HA version without the supervised stuff. Am I right?

Yeah I think you're right. Seems not to be supervised. :(
The method that you successfully tested, can't this be integrated into the DietPi installer?

Incredible how much you guys did in the last months. The new Dashboard looks awesome.

Joulinar, 2022-04-24:
The problem is still network manager as it is conflicting with our network setup.

MichaIng, 2022-04-24:
Respectively outstanding tests whether the container can be made running without NetworkManager.

Joulinar, 2022-04-24:
I would like to test but my RPI4B is blocked by a Discourse testing instance. 😉 

Maybe I will setup VM for HA Supervised

HyperCriSiS, 2022-05-07:
😕Unfortunately my Linux skills are too low to find out how to fix this

pbanj, 2022-05-15:
i now have a spare pi i can mess around with. im going to see if i can get this to work just by messing with the docker containers

pbanj, 2022-05-16:
ok so i got everything but home assistant running. 
[image]
no apparmor, no network manager
it yells in the log but it still runs.
[image]

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[03:13:40] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[03:13:40] INFO: Starting local supervisor watchdog...
22-05-16 03:13:46 INFO (MainThread) [__main__] Initializing Supervisor setup
22-05-16 03:13:46 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-05-15 23:13:46 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
22-05-15 23:13:46 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2022.05.2
22-05-15 23:13:46 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
22-05-15 23:13:46 WARNING (MainThread) [supervisor.resolution.evaluations.base] AppArmor is required for Home Assistant. (more-info: https://www.home-assistant.io/more-info/unsupported/apparmor)
22-05-15 23:13:46 WARNING (MainThread) [supervisor.resolution.evaluations.base] D-Bus is required for Home Assistant. (more-info: https://www.home-assistant.io/more-info/unsupported/dbus)
22-05-15 23:13:46 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-15 23:13:47 INFO (MainThread) [__main__] Setting up Supervisor
22-05-15 23:13:47 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
22-05-15 23:13:47 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
22-05-15 23:13:47 ERROR (MainThread) [supervisor.dbus.manager] No D-Bus support on Host. Disabled any kind of host control!
22-05-15 23:13:47 INFO (MainThread) [supervisor.host.info] Updating local host information
22-05-15 23:13:47 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-05-15 23:13:47 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-05-15 23:13:47 INFO (MainThread) [supervisor.jobs] 'NetworkManager.load' blocked from execution, host Network Manager not available
22-05-15 23:13:47 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: set()
22-05-15 23:13:47 WARNING (MainThread) [supervisor.host.apparmor] AppArmor is not enabled on host
22-05-15 23:13:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2022.05.0
22-05-15 23:13:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1
22-05-15 23:13:47 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
22-05-15 23:13:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2022.05.0
22-05-15 23:13:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
22-05-15 23:13:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
22-05-15 23:13:47 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-05-15 23:13:51 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
22-05-15 23:13:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi4-64-homeassistant with version 2022.5.4
22-05-15 23:13:51 INFO (MainThread) [supervisor.os.manager] No Home Assistant Operating System found
22-05-15 23:13:52 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
22-05-15 23:13:52 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
22-05-15 23:13:52 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
22-05-15 23:13:53 INFO (MainThread) [supervisor.store] Loading add-ons from store: 65 all - 65 new - 0 remove
22-05-15 23:13:53 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
22-05-15 23:13:53 INFO (MainThread) [supervisor.backups.manager] Found 0 backup files
22-05-15 23:13:53 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
22-05-15 23:13:53 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
22-05-15 23:13:53 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
22-05-15 23:13:53 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-05-15 23:13:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-05-15 23:13:53 WARNING (MainThread) [supervisor.resolution.evaluations.base] Systemd-Resolved is required for DNS in Home Assistant. (more-info: https://www.home-assistant.io/more-info/unsupported/systemd_resolved)
[... cut ...]
22-05-15 23:13:55 WARNING (MainThread) [supervisor.resolution.evaluations.base] Found unsupported images: {'portainer/portainer-ce'} (more-info: https://www.home-assistant.io/more-info/unsupported/software)
22-05-15 23:13:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-05-15 23:13:57 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-05-15 23:13:57 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete`

i had home assistant working at one point, so i just need to figure out what the issue is

pbanj, 2022-05-16:
found and fixed the issue
[image]
[image]

HyperCriSiS, 2022-05-16:
Amazing!! Will other services run along with it or will docker eat up all resources?

pbanj, 2022-05-19:
> Amazing!! Will other services run along with it or will docker eat up all resources? Are there many changes that need to be done to make it work?

that would more depend on the amount of ram you have and how much each thing uses. im using it on a 2gb pi4 and it has enough ram for other stuff to be run(it's what i originally used this pi for way back when i got it). as for changes, not really. mostly it was just taking settings from my working setup on my 8gb pi4 and using them for this test, reading the old install script to figure stuff out, grabbing a few bin files off the 8gb. theres one issue im trying to work out, dbus is messed up and cant figure out why. 

if i knew more about how to make docker compose files i'd make one(prob gonna look into it). but i will upload the system image once im done so it can be used/help with getting it actually added to diet pi

pbanj, 2022-05-25:
hmmm my install that ive been using for a long time now the supervisor decided it no longer wants to run. the best part is it isnt giving me logs. the test pi is doing the same thing. so thats fun. ill keep things updated.

also a thought occurred to me. with me running everything outside of home assistant, why am i using it supervised. the only addon i ever found useful was the file browser and that is easy to replace by just pointing docker to use an easy to find folder which i already do.

pbanj, 2022-06-08:
```
version: "3"
services:
  hassio_audio:
    cap_add:
      - SYS_NICE
      - SYS_RESOURCE
    container_name: hassio_audio
    dns:
      - 172.30.32.3
    dns_search:
      - local.hass.io
    domainname: local.hass.io
    entrypoint:
      - /init
    environment:
      - TZ=America/New_York
      - SUPERVISOR_MACHINE=raspberrypi4-64
      - PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
      - LANG=C.UTF-8
      - S6_BEHAVIOUR_IF_STAGE2_FAILS=2
      - S6_CMD_WAIT_FOR_SERVICES=1
    hostname: hassio-audio
    image: ghcr.io/home-assistant/aarch64-hassio-audio:2022.05.0
    ipc: private
    labels:
      io.hass.arch: aarch64
      io.hass.base.arch: aarch64
      io.hass.base.image: arm64v8/alpine:3.13
      io.hass.base.name: alpine
      io.hass.base.version: 2022.02.0
      io.hass.type: audio
      io.hass.version: 2022.05.0
      org.opencontainers.image.authors: The Home Assistant Authors
      org.opencontainers.image.created: '''2022-05-04 15:24:30+00:00'''
      org.opencontainers.image.description: Home Assistant Supervisor plugin for Audio
      org.opencontainers.image.documentation: https://www.home-assistant.io/docs/
      org.opencontainers.image.licenses: Apache License 2.0
      org.opencontainers.image.source: https://github.com/home-assistant/plugin-audio
      org.opencontainers.image.title: Home Assistant Audio Plugin
      org.opencontainers.image.url: https://www.home-assistant.io/
      org.opencontainers.image.version: 2022.05.0
    logging:
      driver: journald
      options: {}
    networks:
      - hassio
    restart: always
    security_opt:
      - seccomp=unconfined
    ulimits:
      - Hard: 10
        Name: rtprio
        Soft: 10
    volumes:
      - /dev:/dev:ro
      - /home/dietpi/docker_data/hassio/audio:/data:rw
      - /run/dbus:/run/dbus:ro
      - /run/udev:/run/udev:ro
      - /etc/machine-id:/etc/machine-id:ro
    working_dir: /
  hassio_cli:
    container_name: hassio_cli
    dns:
      - 172.30.32.3
    dns_search:
      - local.hass.io
    domainname: local.hass.io
    entrypoint:
      - /init
[... 324 lines cut ...]
      - label=disable
    volumes:
      - /dev:/dev:ro
      - /run/dbus:/run/dbus:ro
      - /run/udev:/run/udev:ro
      - /home/dietpi/docker_data/hassio/homeassistant:/config:rw
      - /home/dietpi/docker_data/hassio/ssl:/ssl:ro
      - /home/dietpi/docker_data/hassio/share:/share:rw
      - /home/dietpi/docker_data/hassio/media:/media:rw
      - /etc/machine-id:/etc/machine-id:ro
      - /home/dietpi/docker_data/hassio/tmp/homeassistant_pulse:/etc/pulse/client.conf:ro
      - /home/dietpi/docker_data/hassio/audio/external:/run/audio:ro
      - /home/dietpi/docker_data/hassio/audio/asound:/etc/asound.conf:ro
    working_dir: /config
networks:
  hassio:
    external: true
    name: hassio
  host:
    external: true
    name: host

```

found a tool to create a compose file. also my supervisor decided to just start working again so who knows what the hell is up with that. i still have to work out the dbus issue iirc. i accidently wiped the sd card i was using to test this so i have to start over. but this compose file will make things a lot easier to restart

HyperCriSiS, 2022-07-19:
> found a tool to create a compose file. also my supervisor decided to just start working again so who knows what the hell is up with that. i still have to work out the dbus issue iirc. i accidently wiped the sd card i was using to test this so i have to start over. but this compose file will make things a lot easier to restart
> Config file attached. Click to expand!

Hi,
How is your supervisor instance running? :)

HyperCriSiS, 2022-07-21:
> 

You have much more devices to add and also better possibilities to customize the Dashboard which per default is very restricted and kinda bad.

pbanj, 2022-07-22:
it's still up and running. it has stopped like once or twice after i made that post and has been fine since so not sure what it's issue was

pbanj, 2022-07-22:
> > 
> 
> You have much more devices to add and also better possibilities to customize the Dashboard which per default is very restricted and kinda bad.

you can add devices and dashboards using hacs and you can install hacs without the supervisor. at this point i feel the supervisor is more of a hindrance than anything. all the addons are just docker containers, so it isnt really any different than just running those on their own. the only difference is that using it as an "addon" in home assistant means if home assistant breaks/wont start for whatever reason, none of the addons will either. which is pretty shitty tbh. it's a lot better to just use portainer if you want a ui to manage docker containers.

MichaIng, 2022-07-22:
> you can install hacs without the supervisor

Interesting: https://www.wundertech.net/how-to-install-hacs-on-home-assistant/

Probably we can do this even with the native HA installation. Docker is still required for the add-ons, but in theory?

MichaIng, 2022-07-22:
Ah, I understand just now that HACS is not for addons, but for additional integrations, themes and other customisation, isn't it? Thanks for the link, so it can be installed into HA core as well. Looks like a reasonable addition to our HA implementation, but not related to HA supervised.

pbanj, 2022-07-22:
> Ah, I understand just now that HACS is not for addons, but for additional integrations, themes and other customisation, isn't it? Thanks for the link, so it can be installed into HA core as well. Looks like a reasonable addition to our HA implementation, but not related to HA supervised.

correct, it might be enough to make people not want to bother with supervised.

HyperCriSiS, 2022-07-23:
> correct, it might be enough to make people not want to bother with supervised.

But without the supervisor addons are not possible, or is there also a way without supervisor?

MichaIng, 2022-07-23:
At least you'd need to manage the addon's containers manually. So yeah, it is not exactly user friendly to use addons without HA supervised.

pbanj, 2022-07-23:
Could use portainer. Then it's pretty much the same experience. A better one when you realize you can set the data locations to whatever you want so you can easily find them instead of wherever HA puts it

MichaIng, 2022-07-23:
That's true. But I guess users would like to have all HA related stuff in a single web UI and not a second process/webserver/UI for the addons. Also it is simpler to find new addons via HA supervised, compared to browsing the Docker registry or some website manually for them.

But it's good that this is an option as well for those who do not have an issue with above points.

HyperCriSiS, 2022-07-31:
Is there any chance that the supervisor could be part of the next releases? :grinning:

Joulinar, 2022-07-31:
I don't think so.

MichaIng, 2022-07-31:
More likely HACS, but without the supervisor.

MichaIng, 2022-10-15:
Shall we close this issue with the merge of #5802 (HACS integration)? I don't think the whole HA Supervised is feasible for DietPi integration in general.

Joulinar, 2022-10-15:
I agree with you @MichaIng 
That's the easiest we can do.

MichaIng, 2022-10-15:
Merged

closed 2022-10-15 by MichaIng