MichaIng/DietPi #4711: DietPi-Software | qbittorrent login not working ≥ 4.2.0

Joulinar, 2021-09-06:
Using qbittorrent version ≥ 4.2.0, variable to store password in config file has been changed.

Config file location: `/home/qbittorrent/.config/qBittorrent/qBittorrent.conf`

- version < 4.2.0, `WebUI\Password_ha1`
- version ≥ 4.2.0, `WebUI\Password_PBKDF2`

Means our installation procedure will set incorrect value and you are not able to login into WebUi.

https://github.com/MichaIng/DietPi/blob/140fa046bb55b2efe9bc09b8d5da716e8a0956d1/dietpi/dietpi-software#L11594

Workaround would be to manually remove `WebUI\Password_ha1` from config file and restart service.

MichaIng, 2021-09-06:
Good find. We can use the same method used for the global software password to generate the hash.

K92Pi, 2021-09-08:
I have a question regarding to qbittorrent. How do I set it as a daemon, so it runs automatically after restart?

Joulinar, 2021-09-08:
if you use `dietpi-software` to install qbittorrent, it's done automatically.

K92Pi, 2021-09-08:
@Joulinar I've installed with it and the app never starts automatically.

Joulinar, 2021-09-08:
as your issue is not related to the original post, let's continue in #4718

MichaIng, 2021-09-11:
Should be solved with: https://github.com/MichaIng/DietPi/commit/bfc7ebb8debd14fe57d1a6f1a21f4859a7f1b58e

closed 2021-09-12 by MichaIng