MichaIng/DietPi #4712: nanopi m1 plus eth can not request ip

lucas-law, 2021-09-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.10
- Kernel version | `uname -a`
buster
Linux DietPi 5.10.60-sunxi #21.08.1 SMP Wed Aug 25 18:19:32 UTC 2021 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi M1 Plus (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A RAVpower
- SDcard used | (EG: SanDisk ultra)
Samsung 32 EVO +

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dhclient
- Was the software title installed freshly or updated/migrated?
freshly
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`
e77f1110-dd72-4724-8388-464f14fdc84e

#### Steps to reproduce

1. configure eth0 adapter and request ip
2. use tcpdump to capture the dhcp package.

#### Expected behaviour

- Normal request to ip

#### Actual behaviour

- Received nothing dhcp reply pkg from router，request id failed。

#### Extra details

1. wlan0 can request ip.
2. eth0 links ok and can recv other udp msg from network.
3. the wlan0's tcpdump data as fllows:
oot@DietPi:~# dhclient -i wlan0
16:22:04.459956 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from b0:f1:ec:2d:c1:d8 (oui Unknown), length 300, xid 0xe5899552, Flags [none] (0x0000)
	  Client-Ethernet-Address b0:f1:ec:2d:c1:d8 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Hostname Option 12, length 6: "DietPi"
	    Parameter-Request Option 55, length 13: 
	      Subnet-Mask, BR, Time-Zone, Default-Gateway
	      Domain-Name, Domain-Name-Server, Option 119, Hostname
	      Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
	      NTP
	    Client-ID Option 61, length 19: hardware-type 255, ec:2d:c1:d8:00:01:00:01:27:9c:b9:93:02:81:2c:03:fa:12
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 12
16:22:04.495060 IP (tos 0x0, ttl 64, id 0, offset 0, flags [none], proto UDP (17), length 576)
    192.168.178.1.bootps > 192.168.178.77.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 548, xid 0xe5899552, Flags [none] (0x0000)
	  Your-IP 192.168.178.77
	  Server-IP 192.168.178.1
	  Client-Ethernet-Address b0:f1:ec:2d:c1:d8 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Offer
	    Server-ID Option 54, length 4: 192.168.178.1
	    Lease-Time Option 51, length 4: 864000
	    RN Option 58, length 4: 432000
	    RB Option 59, length 4: 756000
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    Default-Gateway Option 3, length 4: 192.168.178.1
	    Domain-Name-Server Option 6, length 4: 192.168.178.1
	    Domain-Name Option 15, length 9: "fritz.box"
	    BR Option 28, length 4: 192.168.178.255
	    NTP Option 42, length 4: 192.168.178.1
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 239
16:22:04.495605 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from b0:f1:ec:2d:c1:d8 (oui Unknown), length 300, xid 0xe5899552, Flags [none] (0x0000)
	  Client-Ethernet-Address b0:f1:ec:2d:c1:d8 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    Server-ID Option 54, length 4: 192.168.178.1
	    Requested-IP Option 50, length 4: 192.168.178.77
	    Hostname Option 12, length 6: "DietPi"
	    Parameter-Request Option 55, length 13: 
	      Subnet-Mask, BR, Time-Zone, Default-Gateway
	      Domain-Name, Domain-Name-Server, Option 119, Hostname
	      Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
	      NTP
	    Client-ID Option 61, length 19: hardware-type 255, ec:2d:c1:d8:00:01:00:01:27:9c:b9:93:02:81:2c:03:fa:12
	    END Option 255, length 0
16:22:04.664402 IP (tos 0x0, ttl 64, id 0, offset 0, flags [none], proto UDP (17), length 576)
    192.168.178.1.bootps > 192.168.178.77.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 548, xid 0xe5899552, Flags [none] (0x0000)
	  Your-IP 192.168.178.77
	  Server-IP 192.168.178.1
	  Client-Ethernet-Address b0:f1:ec:2d:c1:d8 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: ACK
	    Server-ID Option 54, length 4: 192.168.178.1
	    Lease-Time Option 51, length 4: 864000
	    RN Option 58, length 4: 432000
	    RB Option 59, length 4: 756000
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    Default-Gateway Option 3, length 4: 192.168.178.1
	    Domain-Name-Server Option 6, length 4: 192.168.178.1
	    Domain-Name Option 15, length 9: "fritz.box"
	    BR Option 28, length 4: 192.168.178.255
	    NTP Option 42, length 4: 192.168.178.1
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 239

4. the eth0 tcpdump data as fllows:
root@DietPi:~# dhclient -i eth0
16:23:18.079786 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from 02:81:2c:03:fa:12 (oui Unknown), length 300, xid 0xc9aeb208, Flags [none] (0x0000)
	  Client-Ethernet-Address 02:81:2c:03:fa:12 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Hostname Option 12, length 6: "DietPi"
	    Parameter-Request Option 55, length 13: 
	      Subnet-Mask, BR, Time-Zone, Default-Gateway
	      Domain-Name, Domain-Name-Server, Option 119, Hostname
	      Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
	      NTP
	    Client-ID Option 61, length 19: hardware-type 255, 2c:03:fa:12:00:01:00:01:27:9c:b9:93:02:81:2c:03:fa:12
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 12
16:23:27.021965 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from 02:81:2c:03:fa:12 (oui Unknown), length 300, xid 0xc9aeb208, secs 9, Flags [none] (0x0000)
	  Client-Ethernet-Address 02:81:2c:03:fa:12 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Hostname Option 12, length 6: "DietPi"
	    Parameter-Request Option 55, length 13: 
	      Subnet-Mask, BR, Time-Zone, Default-Gateway
	      Domain-Name, Domain-Name-Server, Option 119, Hostname
	      Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
	      NTP
	    Client-ID Option 61, length 19: hardware-type 255, 2c:03:fa:12:00:01:00:01:27:9c:b9:93:02:81:2c:03:fa:12
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 12
root@DietPi:~# 16:23:32.770496 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from 02:81:2c:03:fa:12 (oui Unknown), length 300, xid 0xc7b65164, Flags [none] (0x0000)
	  Client-Ethernet-Address 02:81:2c:03:fa:12 (oui Unknown)
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Hostname Option 12, length 6: "DietPi"
	    Parameter-Request Option 55, length 13: 
	      Subnet-Mask, BR, Time-Zone, Default-Gateway
	      Domain-Name, Domain-Name-Server, Option 119, Hostname
	      Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
	      NTP
	    Client-ID Option 61, length 19: hardware-type 255, 2c:03:fa:12:00:01:00:01:27:9c:b9:93:02:81:2c:03:fa:12
	    END Option 255, length 0
	    PAD Option 0, length 0, occurs 12

5. I installed another OS like debain and it looks good.

MichaIng, 2021-09-07:
If you have a screen+keyboard or serial console attached, please run the following commands and paste the output here:
```sh
ifdown --force wlan0
ifdown --force eth0
ifup eth0
ip l
ip a
ip r
```

Quote tags around console pastes are btw nice for readability (see my edit of your post):
````
```
text from monotype console
```
````

lucas-law, 2021-09-07:
There are some outputs from serial port:
```
root@DietPi:~# ifdown --force wlan0
Removed stale PID file
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b0:f1:ec:2d:c1:d8
Sending on   LPF/wlan0/b0:f1:ec:2d:c1:d8
Sending on   Socket/fallback
DHCPRELEASE of 192.168.178.123 on wlan0 to 192.168.178.1 port 67
root@DietPi:~# ifdown --force eth0
Removed stale PID file
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/02:81:2c:03:fa:12
Sending on   LPF/eth0/02:81:2c:03:fa:12
Sending on   Socket/fallback
root@DietPi:~# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether 02:81:2c:03:fa:12 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether b0:f1:ec:2d:c1:d8 brd ff:ff:ff:ff:ff:ff
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 02:81:2c:03:fa:12 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b0:f1:ec:2d:c1:d8 brd ff:ff:ff:ff:ff:ff
root@DietPi:~# ip r
root@DietPi:~# 

```

Joulinar, 2021-09-07:
I guess command `ifup eth0` is missing, correct?

lucas-law, 2021-09-07:
Yes, ifup eth0 have some erros like:
```
root@DietPi:~# ifup eth0
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/02:81:2c:03:fa:12
Sending on   LPF/eth0/02:81:2c:03:fa:12
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
root@DietPi:~# 
```

Joulinar, 2021-09-07:
did you changed Ethernet cable or used a different port? How does it behave if you switch to STATIC IP on `eth0`

lucas-law, 2021-09-07:
1. switch to static ip and it works not good:
```
root@DietPi:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.178.199  netmask 255.255.255.0  broadcast 192.168.178.255
        ether 02:81:2c:03:fa:12  txqueuelen 1000  (Ethernet)
        RX packets 7431  bytes 616556 (602.1 KiB)
        RX errors 0  dropped 810  overruns 0  frame 0
        TX packets 487  bytes 156390 (152.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 45  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 8  bytes 660 (660.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8  bytes 660 (660.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@DietPi:~# ping 192.168.178.1
PING 192.168.178.1 (192.168.178.1) 56(84) bytes of data.
^C
--- 192.168.178.1 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 134ms

root@DietPi:~# 

```
2. This board just have 1 Ethernet cable, and It works good on other OS. so maybe some errors occured.

Joulinar, 2021-09-07:
can you check for kernel error messages

`dmesg -l err,crit,alert,emerg`

lucas-law, 2021-09-07:
```
root@DietPi:~# dmesg -l err,crit,alert,emerg
[    6.860327] lima 1c40000.gpu: dev_pm_opp_set_regulators: no regulator (mali) found: -19
root@DietPi:~# 
root@DietPi:~# 

```

MichaIng, 2021-09-07:
This is `1`, isn't it?
```sh
cat /sys/class/net/eth0/carrier
```

Can you check whether there are kernel upgrades available, and which kernel packages are installed?
```sh
apt update
apt full-upgrade
dpkg -l | grep 'linux-image-'
```

lucas-law, 2021-09-07:
1. yes , cat /sys/class/net/eth0/carrier and it print 1

2.  dpkg -l print:
```
ii  linux-image-current-sunxi              21.08.1                             armhf        Linux kernel, version 5.10.60-sunxi

```

MichaIng, 2021-09-07:
Does the router/DHCP server show any logs or info about the DietPi device? I don't think that it is a clear issue with hardware or DietPi when everything else, including static IP and DHCP on WiFi work fine and also outgoing DHCP requests do not fail. So it seems to be the DHCP server which does not receive or block those or for another reason denies to assign an IP. Probably it simply does not like the fact that it has a client with this hostname in the list already (due to WiFi DHCP assignment) and then ignores the request with same hostname form Ethernet? So probably removing the WiFi based DHCP reservation/info for that DietPi system helps, or the other way round, changing the systems hostname from `DietPi` to a more specific one.

lucas-law, 2021-09-07:
I tried to connect another router, and set hostname to nanopi instead of DietPi the same problem as the old router.

Joulinar, 2021-09-07:
Do you have any firewall software installed?

lucas-law, 2021-09-07:
No, after installing the new dietpi os, there will be problems configuring the network with eth0.

MichaIng, 2021-09-07:
To check whether its an issue with the kernel or device tree, you could try it with the Armbian image: https://dl.armbian.com/nanopim1plus/Buster_current

Joulinar, 2021-09-07:
If you are assign STATIC IP, are you able to `ping` the device themselves?

lucas-law, 2021-09-07:
ping itself ok and ping router gw failed.
```
root@nanopi:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.111  netmask 255.255.255.0  broadcast 192.168.1.255
        ether 02:81:2c:03:fa:12  txqueuelen 1000  (Ethernet)
        RX packets 242  bytes 18628 (18.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 51  bytes 5802 (5.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 45  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1  bytes 80 (80.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1  bytes 80 (80.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@nanopi:~# ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
^C
--- 192.168.1.1 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 30ms

root@nanopi:~# ping 192.168.1.111
PING 192.168.1.111 (192.168.1.111) 56(84) bytes of data.
64 bytes from 192.168.1.111: icmp_seq=1 ttl=64 time=0.135 ms
64 bytes from 192.168.1.111: icmp_seq=2 ttl=64 time=0.061 ms
64 bytes from 192.168.1.111: icmp_seq=3 ttl=64 time=0.051 ms
64 bytes from 192.168.1.111: icmp_seq=4 ttl=64 time=0.077 ms
^C
--- 192.168.1.111 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 76ms
rtt min/avg/max/mdev = 0.051/0.081/0.135/0.032 ms
root@nanopi:~# 

```

MichaIng, 2021-09-07:
Okay I missed that it doesn't work either with a static IP. Then it looks like a firmware/kernel issue and the next step to debug would be to verify whether or not it is an issue with the kernel itself via above linked Armbian image.

lucas-law, 2021-09-07:
Thanks to debug the problem ASAP.😊

Joulinar, 2021-09-07:
Pls try to run the test with the linked Armbian image above. Because DietPi is not an own OS. It's a set of shell scripts on top of a base image. For your SBC, we use Armbian and it would be helpful to see how a stock image behaves. :wink:

MichaIng, 2021-09-08:
Many thanks for testing. I'll report this in the Armbian forum and see whether someone has an idea how to fix or workaround. But there is sadly no guarantee as this SBC is marked with "No official support (CSC)" at Armbian.

Another thing we could test is the Debian kernel, which has support (a device tree) for NanoPi NEO Plus as well. They don't have a bootloader (u-boot) package for it, but since both is pretty close to upstream and same version, the one from Armbian should just work fine. I'll create a hybrid image, based on Debian Bullseye, let's see how this is going. But give me a few days.

You can try it from backports as well, but note that it may make the device unbootable, when the bootloader is not able to work that kernel and device tree:
```sh
# purge Armbian device tree and root overlay packages
apt purge --autoremove linux-dtb-current-sunxi linux-buster-root-current-nanopim1plus
rm /boot/dtb
# install Debian kernel and link device tree and kernel to effective locations
apt install -t buster-backports linux-image-armmp
ln -sf /boot/vmlinuz-5.10.0-0.bpo.8-armmp /boot/zImage
mkdir -p /boot/dtb/overlay
ln -s /usr/lib/linux-image-5.10.0-0.bpo.8-armmp/sun8i-h3-nanopi-m1-plus.dtb /boot/dtb/sun8i-h3-nanopi-m1-plus.dtb
# Create scripts to update and cleanup the U-Boot initramfs when it gets updated or an old kernel removed
cat << '_EOF_' > /etc/initramfs/post-update.d/99-dietpi-uboot
#!/bin/dash
echo 'update-initramfs: Converting to u-boot format' >&2
mkimage -A arm -O linux -T ramdisk -C gzip -n uInitrd -d $2 /boot/uInitrd-$1 > /dev/null
ln -sf uInitrd-$1 /boot/uInitrd > /dev/null 2>&1 || mv /boot/uInitrd-$1 /boot/uInitrd
exit 0
_EOF_
cat << '_EOF_' > /etc/kernel/preinst.d/dietpi-initramfs_cleanup
#!/bin/dash

# skip if initramfs-tools is not installed
[ -x /usr/sbin/update-initramfs ] || exit 0

# passing the kernel version is required
version="$1"
if [ -z "$version" ]; then
        echo "W: initramfs-tools: ${DPKG_MAINTSCRIPT_PACKAGE:-kernel package} did not pass a version number" >&2
        exit 0
fi

# avoid running multiple times
if [ -n "$DEB_MAINT_PARAMS" ]; then
        eval set -- "$DEB_MAINT_PARAMS"
        if [ -z "$1" ] || [ "$1" != 'upgrade' ]; then
                exit 0
        fi
fi

# loop through existing initramfs images
for v in $(ls -1 /var/lib/initramfs-tools | linux-version sort --reverse); do
        if ! linux-version compare $v eq $version; then
                # try to delete delete old initrd images via update-initramfs
                INITRAMFS_TOOLS_KERNEL_HOOK=y update-initramfs -d -k $v 2>/dev/null
                # delete unused state files
                find /var/lib/initramfs-tools -type f ! -name "$version" -printf 'Removing obsolete file %f\n' -delete
                # delete unused initrd images
                find /boot -name 'initrd.img-*' -o -name 'uInitrd-*' ! -name "*-$version" -printf 'Removing obsolete file %f\n' -delete
        fi
done

exit 0
_EOF_
# Update initramfs
update-initramfs -u
reboot
# after reboot, if it does boot, purge the Armbian kernel
apt purge --autoremove linux-image-current-sunxi
# and last check if everything in boot partition is as it should
ls -l /boot
```

lucas-law, 2021-09-08:
I have test the  image of Armbian_21.08.1_Nanopim1plus_bullseye_current_5.10.60.img.xz ; It not works good again。

MichaIng, 2021-09-08:
Yes that is expected, since it is the same kernel and device tree. But good also to have Debian Buster userspace ruled out 👍.

StephanStS, 2021-09-29:
I reached a running system with these steps:

1. Start the DietPi image (all updates are done during the initialisation procedure of DietPi).
1. `uname -a` should then show kernel version 5.10.60.
1. Activate WLAN and set the WLAN credentials (SSID, password) to be able to contact the system next boot.
1. Reboot.
1. Login to the system. `uname -a` should then show kernel version 5.10.60.
1. Do an install to the legacy kernel via `apt install linux-image-legacy-sunxi linux-dtb-legacy-sunxi linux-u-boot-nanopim1plus-legacy`.
1. Reboot.
1. Then LAN and also WLAN both should work. `uname -a` should then show kernel version 5.4.88.

Remark # 1: Sometimes I not only needed to execute an `apt install ...`, furthermore a following `apt reinstall ...` (with the same packages) helped. I did not find out further details for this, i.e. when this was necessary.

Remark # 2: Sometimes I could switchover to the edge kernel via `apt install linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge` resp. `apt reinstall ...`. This I did after an installation of the legacy kernel. In this case, ´uname -a` showed kernel 5.14 and LAN resp. WLAN worked both.  But this did not work in all cases, the kernel then stayed at 5.4.88.

Remark # 3: Sometimes a reboot needed a power cycle. I do not know whether that this did any changes (i.e. I did not investigate the necessity of a power cycle in all cases).

I did several tests, but did not found a foolproof solution which worked in all situations of life. Have fun, investigating by yourself and post your results here. 😊

MichaIng, 2021-09-29:
Many thanks for testing. I think it may be better to first remove the installed kernel packages and then install the new ones. This probably prevents the need for the reinstalls.

If it works sometimes with the "edge" kernel, while sometimes it still loads the "legacy" kernel 5.4.88 on reboot, this may mean that only the legacy kernel works? So it worked with edge kernel installed only when it was not installed successfully, i.e. that on a reboot still the legacy kernel was loaded? It would explain the inconsistent results.

So to me it looks like we need to update the images to use the legacy Armbian kernel.

StephanStS, 2021-10-05:
After some more (hours of) testing, it seems that the legacy kernel needs once to be installed. No detailed doubt, why. 👶 
The /boot dir then shows:
```
root@DietPi:/boot# ls -l
total 30740
-rw-r--r-- 1 root root 4182558 Sep 14 18:28 System.map-5.10.60-sunxi
-rw-r--r-- 1 root root     253 Oct  2 14:42 armbianEnv.txt
-rw-r--r-- 1 root root  307322 Oct  2 14:42 boot.bmp
-rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
-rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
-rw-r--r-- 1 root root  192403 Sep 14 18:28 config-5.10.60-sunxi
drwxr-xr-x 4 root root    4096 Oct  4 22:11 dietpi
-rw-rw-r-- 1 root root  314088 Sep 29 23:24 dietpi-CHANGELOG.txt
-rw-rw-r-- 1 root root   18092 Sep 29 23:24 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root   14711 Sep 29 23:24 dietpi-README.md
-rw-rw-r-- 1 root root   15150 Oct  4 22:10 dietpi.txt
lrwxrwxrwx 1 root root      17 Oct  5 00:01 dtb -> dtb-5.10.60-sunxi
drwxr-xr-x 3 root root   12288 Oct  5 00:00 dtb-5.10.60-sunxi
-rw-r--r-- 1 root root 8995491 Oct  5 00:01 initrd.img-5.10.60-sunxi
drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
lrwxrwxrwx 1 root root      20 Oct  2 14:31 uInitrd -> uInitrd-5.4.88-sunxi
-rw-r--r-- 1 root root 9387240 Oct  2 14:31 uInitrd-5.4.88-sunxi
-rwxr-xr-x 1 root root 7995288 Sep 14 18:28 vmlinuz-5.10.60-sunxi
lrwxrwxrwx 1 root root      21 Oct  5 00:01 zImage -> vmlinuz-5.10.60-sunxi
root@DietPi:/boot# uname -a
Linux DietPi 5.10.60-sunxi #21.08.2 SMP Tue Sep 14 16:28:44 UTC 2021 armv7l GNU/Linux
root@DietPi:/boot#
```
Interesting is the link to the uInitrd-5.4.88-sunxi file, which is present after a legacy install (and an installation of current afterwards).

Also the edge kernel works if uInitrd links to 5.4.88: 
```
apt install linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge
apt remove linux-image-current-sunxi linux-dtb-current-sunxi linux-u-boot-nanopim1plus-current
```
This then gives
```
root@DietPi:/boot# ls -l
total 30888
-rw-r--r-- 1 root root 4325751 Sep 26 22:35 System.map-5.14.8-sunxi
-rw-r--r-- 1 root root     253 Oct  2 14:42 armbianEnv.txt
-rw-r--r-- 1 root root  307322 Oct  2 14:42 boot.bmp
-rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
-rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
-rw-r--r-- 1 root root  198159 Sep 26 22:35 config-5.14.8-sunxi
drwxr-xr-x 4 root root    4096 Oct  4 22:11 dietpi
-rw-rw-r-- 1 root root  314088 Sep 29 23:24 dietpi-CHANGELOG.txt
-rw-rw-r-- 1 root root   18092 Sep 29 23:24 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root   14711 Sep 29 23:24 dietpi-README.md
-rw-rw-r-- 1 root root   15150 Oct  4 22:10 dietpi.txt
lrwxrwxrwx 1 root root      16 Oct  5 00:09 dtb -> dtb-5.14.8-sunxi
drwxr-xr-x 3 root root   12288 Oct  5 00:08 dtb-5.14.8-sunxi
-rw-r--r-- 1 root root 8748124 Oct  5 00:09 initrd.img-5.14.8-sunxi
drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
lrwxrwxrwx 1 root root      20 Oct  2 14:31 uInitrd -> uInitrd-5.4.88-sunxi
-rw-r--r-- 1 root root 9387240 Oct  2 14:31 uInitrd-5.4.88-sunxi
-rwxr-xr-x 1 root root 8244488 Sep 26 22:35 vmlinuz-5.14.8-sunxi
lrwxrwxrwx 1 root root      20 Oct  5 00:09 zImage -> vmlinuz-5.14.8-sunxi
root@DietPi:/boot# uname -a
Linux DietPi 5.14.8-sunxi #trunk.18 SMP Sun Sep 26 20:35:20 UTC 2021 armv7l GNU/Linux
root@DietPi:/boot#
```

I now generated an image with Buster and Bullseye, both worked with the edge kernel given above.

MichaIng, 2021-10-05:
Ähm wait, in both cases legacy uInitrd is mixed with current/edge kernel. Even if this works for some reason, we shouldn't ship any image like this. Please try to always remove the present kernel, bootloader and device tree before installing a different one. The kernel pre/post install/uninstall hook seems to not handle the case gracefully when two different kernels are installed.

Only if the edge kernel works with `uInitrd -> uInitrd-5.14.8-sunxi` (and the letter generated) we can use it, also as a later kernel upgrade may replace the faulty link. If that does not work, we need to ship it with only legacy kernel, bootloader and device tree.

StephanStS, 2021-10-05:
How to generate the uInitrd-5.14.8-sunxi?

I tried that, but had no success.

MichaIng, 2021-10-05:
You did:
```sh
apt purge linux-image-current-sunxi linux-dtb-current-sunxi linux-u-boot-nanopim1plus-current
apt install linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge
```
and `uInitrd-5.14.*` was not created?

StephanStS, 2021-10-05:
I first installed, then purged. Possibly this was the problem.
Now I did
```sh
apt purge linux-image-current-sunxi linux-dtb-current-sunxi linux-u-boot-nanopim1plus-current
apt purge linux-image-legacy-sunxi linux-dtb-legacy-sunxi linux-u-boot-nanopim1plus-legacy
```
and afterwards the uInitrd incl. symlink was present. Then I moved them away so that this remained:
```
root@DietPi:/boot# ls -l
total 21556
-rw-r--r-- 1 root root 4325751 Sep 26 21:35 System.map-5.14.8-sunxi
-rw-r--r-- 1 root root     253 Oct  2 13:42 armbianEnv.txt
-rw-r--r-- 1 root root  307322 Oct  5 15:26 boot.bmp
-rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
-rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
-rw-r--r-- 1 root root  198159 Sep 26 21:35 config-5.14.8-sunxi
drwxr-xr-x 4 root root    4096 Oct  5 19:25 dietpi
-rw-rw-r-- 1 root root  314088 Sep 29 22:24 dietpi-CHANGELOG.txt
-rw-rw-r-- 1 root root   18092 Sep 29 22:24 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root   14711 Sep 29 22:24 dietpi-README.md
-rw------- 1 root root    2955 Oct  5 15:28 dietpi-wifi.txt
-rw-rw-r-- 1 root root   15149 Oct  5 19:25 dietpi.txt
lrwxrwxrwx 1 root root      16 Oct  4 23:09 dtb -> dtb-5.14.8-sunxi
drwxr-xr-x 3 root root   12288 Oct  5 00:00 dtb-5.14.8-sunxi
-rw-r--r-- 1 root root 8572218 Oct  5 15:16 initrd.img-5.14.8-sunxi
drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
-rwxr-xr-x 1 root root 8244488 Sep 26 21:35 vmlinuz-5.14.8-sunxi
lrwxrwxrwx 1 root root      20 Oct  4 23:53 zImage -> vmlinuz-5.14.8-sunxi
root@DietPi:/boot#
```
Then after
`apt install linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge`
I got no symlink to the uInitrd (same `ls -l` output like above).
I then did a reinstall which did not help:
```
root@DietPi:/boot# apt reinstall linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 3 reinstalled, 0 to remove and 0 not upgraded.
Need to get 43.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://mirror.armbian.de/apt bullseye/main armhf linux-dtb-edge-sunxi armhf 21.11.0-trunk.18 [201 kB]
Get:2 http://armbian.16z.eu/apt bullseye/main armhf linux-image-edge-sunxi armhf 21.11.0-trunk.18 [43.2 MB]
Get:3 https://imola.armbian.com/apt bullseye/main armhf linux-u-boot-nanopim1plus-edge armhf 21.08.1 [242 kB]
Fetched 43.6 MB in 16s (2764 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16846 files and directories currently installed.)
Preparing to unpack .../linux-dtb-edge-sunxi_21.11.0-trunk.18_armhf.deb ...
Unpacking linux-dtb-edge-sunxi (21.11.0-trunk.18) over (21.11.0-trunk.18) ...
Preparing to unpack .../linux-image-edge-sunxi_21.11.0-trunk.18_armhf.deb ...
Unpacking linux-image-edge-sunxi (21.11.0-trunk.18) over (21.11.0-trunk.18) ...
Preparing to unpack .../linux-u-boot-nanopim1plus-edge_21.08.1_armhf.deb ...
Unpacking linux-u-boot-nanopim1plus-edge (21.08.1) over (21.08.1) ...
Setting up linux-image-edge-sunxi (21.11.0-trunk.18) ...
update-initramfs: Generating /boot/initrd.img-5.14.8-sunxi
Setting up linux-u-boot-nanopim1plus-edge (21.08.1) ...
Setting up linux-dtb-edge-sunxi (21.11.0-trunk.18) ...
root@DietPi:/boot#
```
Since there was the output `debconf: delaying package configuration, since apt-utils is not installed` I installed the apt-utils package and did a further reinstall:
```
root@DietPi:/boot# apt reinstall linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 3 reinstalled, 0 to remove and 0 not upgraded.
Need to get 43.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://armbian.16z.eu/apt bullseye/main armhf linux-dtb-edge-sunxi armhf 21.11.0-trunk.18 [201 kB]
[... 16 lines cut ...]
-rw-r--r-- 1 root root 4325751 Sep 26 21:35 System.map-5.14.8-sunxi
-rw-r--r-- 1 root root     253 Oct  2 13:42 armbianEnv.txt
-rw-r--r-- 1 root root  307322 Oct  5 15:26 boot.bmp
-rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
-rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
-rw-r--r-- 1 root root  198159 Sep 26 21:35 config-5.14.8-sunxi
drwxr-xr-x 4 root root    4096 Oct  5 19:25 dietpi
-rw-rw-r-- 1 root root  314088 Sep 29 22:24 dietpi-CHANGELOG.txt
-rw-rw-r-- 1 root root   18092 Sep 29 22:24 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root   14711 Sep 29 22:24 dietpi-README.md
-rw------- 1 root root    2955 Oct  5 15:28 dietpi-wifi.txt
-rw-rw-r-- 1 root root   15149 Oct  5 19:25 dietpi.txt
lrwxrwxrwx 1 root root      16 Oct  5 19:36 dtb -> dtb-5.14.8-sunxi
drwxr-xr-x 3 root root   12288 Oct  5 19:36 dtb-5.14.8-sunxi
-rw-r--r-- 1 root root 8572146 Oct  5 19:36 initrd.img-5.14.8-sunxi
drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
-rwxr-xr-x 1 root root 8244488 Sep 26 21:35 vmlinuz-5.14.8-sunxi
lrwxrwxrwx 1 root root      20 Oct  5 19:36 zImage -> vmlinuz-5.14.8-sunxi
root@DietPi:/boot#
```

No uInitrd incl. symlink is generated.

What should be the next step? I keep the system in this state.
I assume when rebooting, this would crash the system. Then I only come back by moving back the uInitrd incl. symlink...

MichaIng, 2021-10-05:
Strange, the initramfs hook doesn't seem to run, strangely it did with the legacy kernel, as `uInitrd-5.4.88-sunxi` was generated. I need to run a test here. Generally:
```
update-initramfs -u
```
should create `initrd.img-5.14.8-sunxi` first (it does) and then invoke `/etc/initramfs/post-update.d/99-dietpi-uboot` with version and path arguments to create the related `uInitrd-5.14.8-sunxi` + symlink `uInitrd`.

StephanStS, 2021-10-06:
`update-initramfs -u` generates the `initrd.img-5.14.8-sunxi` file but does not generate the uInitrd + symlink (no files and no message): 
```
root@DietPi:/boot# update-initramfs -u
update-initramfs: Generating /boot/initrd.img-5.14.8-sunxi
root@DietPi:/boot#
```
99-dietpi-uboot is present, but does not have the 'x' flag set:
```
root@DietPi:/boot# ls -l /etc/initramfs/post-update.d/99-dietpi-uboot 
-rw-r--r-- 1 root root 251 Oct  5 15:24 /etc/initramfs/post-update.d/99-dietpi-uboot
root@DietPi:/boot# cat /etc/initramfs/post-update.d/99-dietpi-uboot 
#!/bin/dash
echo 'update-initramfs: Converting to u-boot format' >&2
mkimage -A arm -O linux -T ramdisk -C gzip -n uInitrd -d $2 /boot/uInitrd-$1 > /dev/null
ln -sf uInitrd-$1 /boot/uInitrd > /dev/null 2>&1 || mv /boot/uInitrd-$1 /boot/uInitrd
exit 0
root@DietPi:/boot#
```

StephanStS, 2021-10-06:
I now added the execute flag to 99-dietpi-uboot (via `chmod +x`) and did an `apt reinstall linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge`.
Now we have, what we expected:
```
root@DietPi:/boot# ls -l
total 29920
-rw-r--r-- 1 root root 4325751 Sep 26 21:35 System.map-5.14.8-sunxi
-rw-r--r-- 1 root root     253 Oct  2 13:42 armbianEnv.txt
-rw-r--r-- 1 root root  307322 Oct  5 15:26 boot.bmp
-rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
-rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
-rw-r--r-- 1 root root  198159 Sep 26 21:35 config-5.14.8-sunxi
drwxr-xr-x 4 root root    4096 Oct  5 19:25 dietpi
-rw-rw-r-- 1 root root  314088 Sep 29 22:24 dietpi-CHANGELOG.txt
-rw-rw-r-- 1 root root   18092 Sep 29 22:24 dietpi-LICENSE.txt
-rw-rw-r-- 1 root root   14711 Sep 29 22:24 dietpi-README.md
-rw------- 1 root root    2955 Oct  5 15:28 dietpi-wifi.txt
-rw-rw-r-- 1 root root   15149 Oct  5 19:25 dietpi.txt
lrwxrwxrwx 1 root root      16 Oct  6 07:46 dtb -> dtb-5.14.8-sunxi
drwxr-xr-x 3 root root   12288 Oct  6 07:45 dtb-5.14.8-sunxi
-rw-r--r-- 1 root root 8572128 Oct  6 07:46 initrd.img-5.14.8-sunxi
drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
lrwxrwxrwx 1 root root      20 Oct  6 07:46 uInitrd -> uInitrd-5.14.8-sunxi
-rw-r--r-- 1 root root 8572192 Oct  6 07:46 uInitrd-5.14.8-sunxi
-rwxr-xr-x 1 root root 8244488 Sep 26 21:35 vmlinuz-5.14.8-sunxi
lrwxrwxrwx 1 root root      20 Oct  6 07:46 zImage -> vmlinuz-5.14.8-sunxi
root@DietPi:/boot#
```

### Further step
Testing this with a reboot to the edge kernel and prove the original LAN/WLAN issue.
Tataa: The problem is back again, only LAN or WLAN can be activated, but not both in parallel:

[image]

### Final testing
With these steps (`apt remove ...`, `apt install ...`, check for the `/boot` directory contents it gives:

- [ ] **edge kernel (5.14.8) does not work**
- [ ] **current kernel (5.10.60) does not work**
- [x] **legacy kernel (5.4.88) works**

Other kernels are not available:
```
root@DietPi:/boot# apt-cache search linux-image | grep sunxi
linux-image-current-sunxi - Linux kernel, version 5.10.60-sunxi
linux-image-edge-sunxi - Linux kernel, armbian version 5.14.8-sunxi edge
linux-image-legacy-sunxi - Linux kernel, version 5.4.88-sunxi
root@DietPi:/boot# 
```

### Result
**-->> I will generate images for Buster and Bullseye based on the legacy kernel.**

MichaIng, 2021-10-06:
Great fine 👍. Embarrassing. Fixed with: https://github.com/MichaIng/DietPi/commit/c1febe739b136ac7fd04c8eaef4d3097438118df

It is as well an interesting find that only the legacy initramfs does fix the issue already. This may indicate that only a kernel module is missing, which could be loaded manually with the edge kernel as well.

So just if you are in mood, check 'lsmod' with the legacy kernel and compare it with `lsmod` on the current or edge kernel. Probably there are some differences about the Ethernet related modules being loaded. Generally AFAIK the initramfs cannot have an effect on the loaded kernel itself, but probably it can trigger modules being loaded not only for its own environment but for the invoked final system kernel as well.

avworldme, 2022-12-23:
Having this same issue with the latest 8.12 fresh install on an M1Plus.  Except in my case, the Ethernet did not work at all. 
 Reverting Kernel to  legacy kernel (5.4.88) fixes the issue and allowed Eth0 to function separately and along side Wlan0.  

Will having this older Kernel limit the device.  Will it continue to receive updates for other software, namely Home Assistant.  Or will it be locked at current levels?

Joulinar, 2022-12-23:
DietPi version has no relation kernel version. There should be no limitation.

avworldme, 2022-12-23:
Great news.  Thank you.  Will there be a fix for this, as the M1Plus is a long term device that will be produced for a few more years according to the manufacturer.

Joulinar, 2022-12-24:
We don't do any kernel development from our side. I guess this would need to be fixed upstream.

InnovoMagicCube, 2024-03-20:
> I now added the execute flag to 99-dietpi-uboot (via `chmod +x`) and did an `apt reinstall linux-image-edge-sunxi linux-dtb-edge-sunxi linux-u-boot-nanopim1plus-edge`. Now we have, what we expected:
> 
> ```
> root@DietPi:/boot# ls -l
> total 29920
> -rw-r--r-- 1 root root 4325751 Sep 26 21:35 System.map-5.14.8-sunxi
> -rw-r--r-- 1 root root     253 Oct  2 13:42 armbianEnv.txt
> -rw-r--r-- 1 root root  307322 Oct  5 15:26 boot.bmp
> -rw-r--r-- 1 root root    3895 Jan  5  2021 boot.cmd
> -rw-rw-r-- 1 root root    3967 Jan  5  2021 boot.scr
> -rw-r--r-- 1 root root  198159 Sep 26 21:35 config-5.14.8-sunxi
> drwxr-xr-x 4 root root    4096 Oct  5 19:25 dietpi
> -rw-rw-r-- 1 root root  314088 Sep 29 22:24 dietpi-CHANGELOG.txt
> -rw-rw-r-- 1 root root   18092 Sep 29 22:24 dietpi-LICENSE.txt
> -rw-rw-r-- 1 root root   14711 Sep 29 22:24 dietpi-README.md
> -rw------- 1 root root    2955 Oct  5 15:28 dietpi-wifi.txt
> -rw-rw-r-- 1 root root   15149 Oct  5 19:25 dietpi.txt
> lrwxrwxrwx 1 root root      16 Oct  6 07:46 dtb -> dtb-5.14.8-sunxi
> drwxr-xr-x 3 root root   12288 Oct  6 07:45 dtb-5.14.8-sunxi
> -rw-r--r-- 1 root root 8572128 Oct  6 07:46 initrd.img-5.14.8-sunxi
> drwxrwxr-x 2 root root    4096 Jan 21  2021 overlay-user
> lrwxrwxrwx 1 root root      20 Oct  6 07:46 uInitrd -> uInitrd-5.14.8-sunxi
> -rw-r--r-- 1 root root 8572192 Oct  6 07:46 uInitrd-5.14.8-sunxi
> -rwxr-xr-x 1 root root 8244488 Sep 26 21:35 vmlinuz-5.14.8-sunxi
> lrwxrwxrwx 1 root root      20 Oct  6 07:46 zImage -> vmlinuz-5.14.8-sunxi
> root@DietPi:/boot#
> ```
> 
> ### Further step
> Testing this with a reboot to the **edge kernel 5.14.8** and prove the original LAN/WLAN issue. Tataa: The problem is back again, only LAN or WLAN can be activated, but not both in parallel:
> 
> [image]
> 
> The **legacy kernel 5.4.88** shows the working condition:
> 
> [image]
> 
> ### Final testing
> With these steps (`apt remove ...`, `apt install ...`, check for the `/boot` directory contents it gives:
> 
> * [ ]  **edge kernel (5.14.8) does not work**
> * [ ]  **current kernel (5.10.60) does not work**
> * [x]  **legacy kernel (5.4.88) works**
> 
> Other kernels are not available:
> 
> ```
> root@DietPi:/boot# apt-cache search linux-image | grep sunxi
> linux-image-current-sunxi - Linux kernel, version 5.10.60-sunxi
> linux-image-edge-sunxi - Linux kernel, armbian version 5.14.8-sunxi edge
> linux-image-legacy-sunxi - Linux kernel, version 5.4.88-sunxi
> root@DietPi:/boot# 
> ```
> 
> ### Result
> **-->> I will generate images for Buster and Bullseye based on the legacy kernel.**

This is an old issue, but I have a couple of these devices.  I installed a new dietpi image on the device which uses a later kernel and I am having the ethernet / wifi issue again.  Is it possible to lock the M1Plus builds to kernel 5.4.88 as it is the only one that works for this device?  Or if not, how do I roll back to this Kernel and prevent future dietpi updates from updating the Kernel ?