MichaIng/DietPi #4714: Image | RADXA Zero (a Raspberry Pi Zero alternative with horsepower 😉)

docgalaxyblock, 2021-09-07:
#### Formal device information
- Device name | RADXA Zero
- Official product URL | https://wiki.radxa.com/Zero
- GitHub resource URL (not yet avaible) | https://github.com/radxa/
- Image download URL | https://wiki.radxa.com/Zero/downloads

#### Is the SBC officially supported by the Debian installer?
- Seems like not

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Radxa provide one at there [Wiki](https://wiki.radxa.com/Zero/downloads)

#### My Wishes
 - Hope you can develop a good stable image like your rockpi images 😉
 - A [friend](https://github.com/meteyou/mainsail) will get one or two in the next weeks from [here](https://shop.allnetchina.cn/collections/frontpage/products/radxa-zero), ask the shop and maybe you get some (I suggest one 512MB or 1GB and one 2GB+ cause they have different wifi BL chips)
 
 I would like to use these Board in my [small 3D Prineter](https://vorondesign.com/voron0.1)
 I hope the Dev team has some time 🙃
 
 Greetings from Germany 😃

almirus, 2021-10-31:
https://github.com/MichaIng/DietPi/issues/4831

MichaIng, 2021-10-31:
Good find. Support will be added with next release, a testing image is available, but the Radxa kernel is still under development and breaking changes occur here and there and there is not yet a meta kernel package (so that kernel upgrades are done automatically via `apt upgrade` or `apt full-upgrade`).

MichaIng, 2021-11-08:
I'll mark this as closed in favour of the newer request, where a lot of discussion and testing has happened.

closed 2021-11-08 by MichaIng