MichaIng/DietPi #4715: Rock Pi S - APT upgrade fails on first setup

maguilara, 2021-09-08:
#### Required Information
- DietPi version  6.33.3
- Distro version buster 10.10
- Kernel version: Linux DietPi 4.4.143-57-rockchip-g4b8d0571d76e #1 SMP PREEMPT Mon Sep 14 10:40:09 UTC 2020 aarch64 GNU/Linux
- SBC model: ROCK Pi S (aarch64)
- Power supply used: 5V 1A provided by the seller's store (allnet cn)
- SDcard used: SanDisk ultra 32GB

#### Additional Information (if applicable)
Log received

```
 APT upgrade
│  - Command: apt-get -qq upgrade
│  - Exit code: 100
│  - DietPi version: v6.33.3 (MichaIng/master) | HW_MODEL: 73 | HW_ARCH: 3 | DISTRO: 5
│  - Image creator: DietPi Core Team
│  - Pre-image: Radxa
│  - Error log:
│ debconf: delaying package configuration, since apt-utils is not installed
│ (Reading database ... ^M(Reading database ... 5%^M(Reading database ... 10%^M(Reading database ... 1
│ (Reading database ... 20%^M(Reading database ... 25%^M(Reading database ... 30%^M(Reading database .
│ 35%^M(Reading database ... 40%^M(Reading database ... 45%^M(Reading database ... 50%^M(Reading datab
│ ... 55%^M(Reading database ... 60%^M(Reading database ... 65%^M(Reading database ... 70%^M(Reading
│ database ... 75%^M(Reading database ... 80%^M(Reading database ... 85%^M(Reading database ... 90%
│ (Reading database ... 95%^M(Reading database ... 100%^M(Reading database ... 35837 files and
│ directories currently installed.)^M
│ Preparing to unpack .../rockchip-overlay_2.9_all.deb ...^M
│ Unpacking rockchip-overlay (2.9) over (2.7) ...^M
│ dpkg: error processing archive /var/cache/apt/archives/rockchip-overlay_2.9_all.deb (--unpack):^M
│  trying to overwrite '/lib/firmware/rockchip/dptx.bin', which is also in package
│ firmware-misc-nonfree 20190114-2^M
│ Remove two files, hdmi.rules and hdmi-toggle^M
│ Errors were encountered while processing:^M
│  /var/cache/apt/archives/rockchip-overlay_2.9_all.deb^M
│ E: Sub-process /usr/bin/dpkg returned an error code (1)
```
I tried to upload the report via the UI, but it refused for some reason, sorry, couldn't catch the problem.

#### Steps to reproduce
1. Flash OS on to sd
2. Connect SBC to power and put the SD
3. Boot, ssh into the device and try to complete the first setup.
#### Expected behaviour
Run the setup successfully
#### Actual behaviour
Setup fails, and after reboot it asks again to be setup (and fails once again)

#### Extra details
After being greeted by this screen, I exit the setup.
[image]
I receive the following entries in the terminal, which I assume deactivates the first run setup?? But if I reboot the device, it asks once again to setup everything (which of course fails once again). Not sure if this is a setup loop (which I have seen around)

```
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
 ─────────────────────────────────────────────────────
 DietPi v6.33.3 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.0.17 (eth0)
[  OK  ] DietPi-Login | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_AUTOMATED=0
```

Joulinar, 2021-09-08:
Hi,

As long as the first run did not complete fully, it will be repeated on reboot. Looks like conflicting packages, preventing `apt` to run the upgrade.

@MichaIng any ideas on this?

Did you tried to reflash the image? Just to avoid inconsistent.

MichaIng, 2021-09-08:
Many thanks for your report. I think I'll generate a new ROCK Pi S image, based on Armbian, which now have a Linux 5.10 package for it. There is much too often such issues with the ROCK Pi repositories, including the inability to compile kernel modules due to issues with the kernel headers: #3936

They forgot to add conflicts with the pretty common `firmware-misc-nonfree` package. Not sure how to resolve best, as that one provides firmware for quite a number of common devices. However, the `rockchip-overlay` package seems to do the same, so as a workaround purging the generic one should do it:
```sh
apt purge --autoremove firmware-misc-nonfree
```

MichaIng, 2021-09-10:
Reported: #15

Joulinar, 2021-09-10:
wrong issue linked?

MichaIng, 2021-11-07:
Our new Bullseye image btw ships with `rockchip-overlay` v3.0 pre-installed, which does not suffer from the issue. This is from the Radxa testing repository. I'll mark the issue hence as closed here. We'll keep applying this for new images as long as the package is not merged into the Radxa stable repository, but will track the state at the issue linked above.

closed 2021-11-07 by MichaIng