MichaIng/DietPi #4716: unsuccessful backup before dietpi-update

pretorian1986, 2021-09-08:
#### Details:
- Date           | Wed Sep  8 20:31:42 CEST 2021
- Bug report     | c1b26ff8-597d-4307-9662-196e8d2ff461
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | rmdir /mnt/backup_extern/dietpi-backup
- Exit code      | 1
- Software title | DietPi-Backup
#### Steps to reproduce:

1. setup a permanent mount
2. (maybe) use this mount for rsync destination
3. start dietpi backup before diet-update
#### Expected behaviour:

- succesful diet-backup
#### Actual behaviour:

- stops with error message/EXIT CODE: 
- Command        | rmdir /mnt/backup_extern/dietpi-backup
- Exit code      | 1
#### Extra details:

- ...
#### Additional logs:
```
rmdir: failed to remove '/mnt/backup_extern/dietpi-backup': Directory not empty
```

Joulinar, 2021-09-08:
Hi,

some questions. Is that an external disk? If yes, what file system format?
Or is it a network share? If yes, what type of network share?

Once the backup failed, can you scroll up (above the blue error box) on your SSH session and copy the last 10 lines?

MichaIng, 2021-09-09:
Here is btw the related code lines: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-backup#L77-L82
- Only if the directory does not exist yet, to assure that the filesystem type is checked correctly, it is created and again removed afterwards.
- Since it did not exist before, it should be empty, unless something is somehow creating something inside in the millisecond the filesystem type is obtained.
- I could indeed imagine this being the case for a network share hosted on Windows or macOS, which have the habit to automatically create some metadata within directories.

So if this failure happens, it would be interesting to see if and what is actually inside:
```sh
ls -al /mnt/backup_extern/dietpi-backup
```

As the removal failed, the directory does not exist and hence the script won't try to remove it anymore. So repeating the update and backup should work just fine now in any case, as long as the target is on a supported filesystem.

pretorian1986, 2021-09-09:
This is not a network share but a regular mount of external hard drive. It is directly connected to the USB of the dietpi. So at the end the dietpi-backup should be done on this external drive, but for some reason dietpi-backup tries before to delete it.

I guess (but without any deep knowledge or investigation) that it may be linked to two things:
1)  [...] Only if the directory does not exist yet, to assure that the filesystem type is checked correctly, it is created and again removed afterwards. [...]
2) What I observed in the past on some occasions was, that the external drive sometimes stops to switch between active and passive (sleep) mode. The external drive was running all the time. I did not investigate it since the first detection, but I can imagine that it is not only an issue with the power managing mode, but maybe the drive does not work at all as expected in these situation. A reboot (power off) of the system always fixed the issue, so I did not investigate it any further. I'm not sure if this was the case yesterday (problem with power managing), but today the backup worked without any issue. The only thing which was changed since yesterday was that I run the update from dietpi 7.0 to 7.5. Maybe I also rebooted the pi.

So I will keep an eye on this, but at the moment it seems more to be an issue with the drive or the system running the drive, than of dietpi-backup.

Thanks for looking into this issue already some minutes after I posted it.

Joulinar, 2021-09-09:
Yes, if this is a first backup, our script will test if we could r/w to the file system. What file system your are using on that disk? ext4?

As well you can check for kernel error messages `dmesg -l err,crit,alert,emerg`