MichaIng/DietPi #4717: Lighttpd error.

g7kse, 2021-09-08:
Got an issue that I haven't been able to get to the bottom of but I have a feeling it has something to do with letsencrypt.

Using a RPi4 and upgraded to bullseye. 

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
Bullseye

- Kernel version | `uname -a`
Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi4

- Power supply used | (EG: 5V 1A RAVpower)
5v 2.3A

- SDcard used | (EG: SanDisk ultra)
SanDisk EVO

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Lighttpd

- Was the software title installed freshly or updated/migrated?
Updated

- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
ba38d2fb-68f9-42e6-8c73-99cf17d9e646

#### Steps to reproduce

1. ... Follow manual upgrade process from Buster to Bullseye - followed instructions on blog (https://dietpi.com/blog/?p=811)
2. ...Attempt to install livepatch as noticed it was prompted.Sent error but I failed to note it
3.... Attempted to change letsencypt details for new domain. Encountered error and process stopped

#### Expected behaviour

- ...Lighttpd start up without errors

#### Actual behaviour

- ...access to webserver not available. 
_Output from systemctl status lighttpd.service_

lighttpd.service - Lighttpd Daemon
     Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-09-08 19:29:07 BST; 15min ago
    Process: 4025 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)
        CPU: 1.101s

Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Consumed 1.101s CPU time.
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Sep 08 19:29:07 DietPi systemd[1]: Stopped Lighttpd Daemon.
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Consumed 1.101s CPU time.
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Sep 08 19:29:07 DietPi systemd[1]: Failed to start Lighttpd Daemon

_Output from journalctl -xe_
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
Sep 08 19:29:07 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit lighttpd.service has entered the 'failed' state with result 'exit-code'.
Sep 08 19:29:07 DietPi systemd[1]: Failed to start Lighttpd Daemon.
░░ Subject: A start job for unit lighttpd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit lighttpd.service has finished with a failure.

#### Extra details

- ...

Joulinar, 2021-09-08:
Hi,

pls can you post following

`/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf`

g7kse, 2021-09-08:
Ah this is interesting....
```

2021-09-08 20:49:09: plugin.c.195) dlopen() failed for: /usr/lib/lighttpd/mod_openssl.so /usr/lib/lighttpd/mod_openssl.so: cannot open shared object file: No such file or directory
2021-09-08 20:49:09: server.c.1238) loading plugins finally failed
```

Joulinar, 2021-09-08:
yes this seems to be needed on Bullseye. Can you try following and restart server afterwards

`apt install lighttpd-mod-openssl`

g7kse, 2021-09-08:
Result!
```

lighttpd.service - Lighttpd Daemon
Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
Active: active (running) since Wed 2021-09-08 21:15:58 BST; 6s ago
Process: 10693 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
```

I'll mark this as closed in case anyone else gets the same error and keep an eye on it. DietPi is back to being rock solid as usual 

Ta

closed 2021-09-08 by g7kse