MichaIng/DietPi #4718: qbittorrent not starting on reboot

Joulinar, 2021-09-08:
I have a question regarding to qbittorrent. How do I set it as a daemon, so it runs automatically after restart?

_Originally posted by @K92Pi in https://github.com/MichaIng/DietPi/issues/4711#issuecomment-915530650_

Joulinar, 2021-09-08:
@K92Pi 
pls can you post following

`systemctl status qbittorrent`

K92Pi, 2021-09-10:
```
root@MediaServer:~# systemctl start qbittorrent
Job for qbittorrent.service failed because a fatal signal was delivered to the control process.
See "systemctl status qbittorrent.service" and "journalctl -xe" for details.
root@MediaServer:~# systemctl status qbittorrent
● qbittorrent.service - qBittorrent
     Loaded: loaded (/etc/systemd/system/qbittorrent.service; disabled; vendor preset: enabled)
     Active: failed (Result: signal) since Fri 2021-09-10 12:58:05 EDT; 11s ago
    Process: 1985 ExecStart=/usr/bin/qbittorrent -d (code=killed, signal=ABRT)
        CPU: 54ms

Sep 10 12:58:05 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 5.
Sep 10 12:58:05 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 12:58:05 MediaServer systemd[1]: qbittorrent.service: Start request repeated too quickly.
Sep 10 12:58:05 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 12:58:05 MediaServer systemd[1]: Failed to start qBittorrent.
```

Joulinar, 2021-09-10:
could you post following as well pls

`journalctl -u qbittorrent`

K92Pi, 2021-09-10:
@Joulinar 
Of course!

```
root@MediaServer:~# journalctl -u qbittorrent
-- Journal begins at Fri 2021-08-06 23:58:19 EDT, ends at Fri 2021-09-10 13:34:07 EDT. --
Sep 10 13:31:50 MediaServer systemd[1]: Starting qBittorrent...
Sep 10 13:31:51 MediaServer qbittorrent[1058]: qt.qpa.xcb: could not connect to display
Sep 10 13:31:51 MediaServer qbittorrent[1058]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Sep 10 13:31:51 MediaServer qbittorrent[1058]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Sep 10 13:31:51 MediaServer qbittorrent[1058]: Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Sep 10 13:31:51 MediaServer systemd[1]: qbittorrent.service: Control process exited, code=killed, status=6/ABRT
Sep 10 13:31:51 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:51 MediaServer systemd[1]: Failed to start qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 1.
Sep 10 13:31:52 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: Starting qBittorrent...
Sep 10 13:31:52 MediaServer qbittorrent[1242]: qt.qpa.xcb: could not connect to display
Sep 10 13:31:52 MediaServer qbittorrent[1242]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Sep 10 13:31:52 MediaServer qbittorrent[1242]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Sep 10 13:31:52 MediaServer qbittorrent[1242]: Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Control process exited, code=killed, status=6/ABRT
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:52 MediaServer systemd[1]: Failed to start qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 2.
Sep 10 13:31:52 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: Starting qBittorrent...
Sep 10 13:31:52 MediaServer qbittorrent[1357]: qt.qpa.xcb: could not connect to display
Sep 10 13:31:52 MediaServer qbittorrent[1357]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Sep 10 13:31:52 MediaServer qbittorrent[1357]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Sep 10 13:31:52 MediaServer qbittorrent[1357]: Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Control process exited, code=killed, status=6/ABRT
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:52 MediaServer systemd[1]: Failed to start qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 3.
Sep 10 13:31:52 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 13:31:52 MediaServer systemd[1]: Starting qBittorrent...
Sep 10 13:31:52 MediaServer qbittorrent[1412]: qt.qpa.xcb: could not connect to display
Sep 10 13:31:52 MediaServer qbittorrent[1412]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Sep 10 13:31:52 MediaServer qbittorrent[1412]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Sep 10 13:31:52 MediaServer qbittorrent[1412]: Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Control process exited, code=killed, status=6/ABRT
Sep 10 13:31:52 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:52 MediaServer systemd[1]: Failed to start qBittorrent.
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 4.
Sep 10 13:31:53 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 13:31:53 MediaServer systemd[1]: Starting qBittorrent...
Sep 10 13:31:53 MediaServer qbittorrent[1413]: qt.qpa.xcb: could not connect to display
Sep 10 13:31:53 MediaServer qbittorrent[1413]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Sep 10 13:31:53 MediaServer qbittorrent[1413]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Sep 10 13:31:53 MediaServer qbittorrent[1413]: Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Control process exited, code=killed, status=6/ABRT
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:53 MediaServer systemd[1]: Failed to start qBittorrent.
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 5.
Sep 10 13:31:53 MediaServer systemd[1]: Stopped qBittorrent.
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Start request repeated too quickly.
Sep 10 13:31:53 MediaServer systemd[1]: qbittorrent.service: Failed with result 'signal'.
Sep 10 13:31:53 MediaServer systemd[1]: Failed to start qBittorrent.
```

Joulinar, 2021-09-10:
how did you installed qBit? Did you used `dietpi-software`?

K92Pi, 2021-09-10:
@Joulinar Yes, I did use `dietpi-software`.

Joulinar, 2021-09-10:
could you share following pls

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

K92Pi, 2021-09-10:
@Joulinar 
No problem!
```
root@MediaServer:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
```
```
root@MediaServer:~# echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 1
```
```
root@MediaServer:~# uname -a
Linux MediaServer 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux
```
```
root@MediaServer:~# echo $G_HW_MODEL_NAME
RPi 4 Model B (armv7l)
```

MichaIng, 2021-09-10:
It tries to connect to a display. Also the binary that is executed by the service is not the intended one. Can you show the following:
```sh
systemctl cat qbittorrent
dpkg -l | grep qbittorrent
```

K92Pi, 2021-09-10:
@MichaIng 
Tries to connect to a display? Is there anything to do with VNC? 
```
root@MediaServer:~# systemctl cat qbittorrent
# /etc/systemd/system/qbittorrent.service
[Unit]
Description=qBittorrent
After=network.target

[Service]
Type=forking
User=qbittorrent
Group=qbittorrent
UMask=002
ExecStart=/usr/bin/qbittorrent -d
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

```
root@MediaServer:~# dpkg -l | grep qbittorrent
ii  qbittorrent                                                 4.2.5-0.1                                                   armhf        bittorrent client based on libtorrent-rasterbar with a Qt5 GUI
```

MichaIng, 2021-09-10:
That one was not installed via `dietpi-software` but is the result of `apt install qbittorrent`, which is not an ideal package for a server system. `dietpi-software` installs the `qbittorrent-nox` package which does not come with the GUI stuff but can be controlled via web interface instead.

K92Pi, 2021-09-11:
@MichaIng 

Yes, qbittorrent-nox doesen't have GUI, that's why I've installed the other qbittorrent as well. I've tested it, to install qbittorrent-nox with dietpi-software, but the web interface wasn't available. Eventually I used apt install, to install the qbittorrent with a GUI. What would you recommend to install apps that aren't listed in dietpi-software?

Joulinar, 2021-09-11:
let's turn around the question. Why you did not report the issue with `qbittorrent-nox` and the missing web interface. We could have checked this before installing something else.

MichaIng, 2021-09-11:
> What would you recommend to install apps that aren't listed in dietpi-software?

You can definitely do that, I was just confused by "Yes, I did use dietpi-software.", which made us think it's our implementation which throws this error. This GUI version of qBittorrent doesn't seem to support running as a daemon without any terminal attached, or some flag or package is missing. Did you create or edit that service file manually? As there is none shipped with the package from what I see (same with `qbittorrent-nox` through, so we create it in `dietpi-software`). Probably it works when you add `StandardOutput=tty` to the `[Service]` section of the unit, so it has a terminal and hence a screen to connect to. To better debug, I'd first of all try to call `qbittorrent` or `qbittorrent -d` manually from console, to check whether it principally works.

But as long as you do not require a GUI on the attached monitor, I'd indeed suggest to keep using and debugging `qbittorrent-nox` instead. A simple reinstall should do it:
```sh
dietpi-software reinstall 46
```
Then we can see why the web interface is not reachable.

K92Pi, 2021-09-12:
@Joulinar 
I have installed qbittorrent-nox first through the dietpi-software. It installed successfully, but the web GUI didn't work, so after all I've installed the qbittorrent package from the terminal. I see it now, even though the qbittorrent has a nicer and more user friendly surface, it has a lot of problems that no one wants to deal with. It should work out of the box, after installation. I've spent hours to figure out why the torrent client doesn't want to start any downloading, then of course I want it to start automatically, now the downloads are in "stalled" status, which can mean a lot of things. I'm going to install Transmission client as well and see if that works better.

MichaIng, 2021-09-12:
Can you give some more detail about what you mean with "but the web GUI didn't work"?
- The service should start automatically, which you can check via: `systemctl status qbittorrent`
- The web interface should be reachable at port 1340.
- But on Bullseye, there is a known issue that the login does not work, which has been fixed for next release and can be worked around: #4711

Joulinar, 2021-09-12:
@MichaIng 
I guess the password issue is valid for all OS version as it changed with one of last qbit release in general

K92Pi, 2021-09-12:
Transmission isn't working as well. Permission denied. The drive-manager overwrites the manually entered stuff from /etc/fstab so I've changed it back, but it looks like to me there's a huge bug. I've reinstalled the Raspberry Pi a couple of times, because I had this permission error ongoing. I've manually changed the entry in /etc/fstab as @MichaIng has suggested: 
For my exfat drive: `UUID=6051-10CB /mnt/data exfat noatime,lazytime,rw,nofail,noauto,x-systemd.automount,gid=dietpi,fmask=0002,dmask=0002`

and 

for the other, ext4 drive`UUID=e60f968f-3162-443c-93e3-23115c91f15e /mnt/data2 ext4 defaults,auto,users,rw,nofail,noatime 0 0`

I'm very frustrated now, dietpi should make life easier, I need to make my Pi to work.

MichaIng, 2021-09-12:
> I've reinstalled the Raspberry Pi a couple of times, because I had this permission error ongoing.

The permissions of the exFAT drive look good. With upcoming release, the drive manager will apply those as well. In case of the ext4 drive, you need to change the group instead of the user and use 775/664 modes:
```
chown -R root:dietpi /mnt/data2/Downloads
find /mnt/data2/Downloads -type d -exec chmod 775 {} +
find /mnt/data2/Downloads -type f -exec chmod 664 {} +
```
Transmission and qBittorrent should create everything with `qbittorrent:dietpi` respectively `debian-transmission:dietpi` and 775/664 modes, so they R/W permissions for all members of the "dietpi" group is kept.