MichaIng/DietPi #4720: PineBook Pro error occurred during pre-patching no Release file

mydogboris, 2021-09-09:
#### Details:
- Date           | Fri Sep 10 00:01:27 BST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | Pinebook Pro (aarch64) (ID=46)
- Kernel version | Linux DietPi 5.10.35-rockchip64 #21.05.1 SMP PREEMPT Fri May 7 13:53:11 UTC 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get --allow-releaseinfo-change-suite -q update
- Exit code      | 100
- Software title | DietPi-Pre-patches
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://deb.debian.org/debian-security buster/updates InRelease
Hit:4 http://deb.debian.org/debian buster-backports InRelease
Ign:5 https://apt.armbian.com buster InRelease
Err:6 https://apt.armbian.com buster Release
  Redirection from https to 'http://armbian.tnahosting.net/apt/dists/buster/Release' is forbidden [IP: 173.45.128.12 443]
Reading package lists...
E: The repository 'https://apt.armbian.com buster Release' does not have a Release file.
```
---------------------------------------------------------------------

Joulinar, 2021-09-10:
Hi,

This is basically know issue outside DietPi as the challenge is with Armbian apt repository. The repository tries to redirect from https to http, which is not allowed by the client. Usually this should have been fixed by Armbian already but looks like not. 

As a workaround you could change Armbian source list and switch from https to http. Just open following list

`nano /etc/apt/sources.list.d/armbian.list`

And replace https with http

mydogboris, 2021-09-10:
I did replace with http.  If you notice in the Additional logs listed above they are all http with Hit:1, Hit:2 etc.  It still doesn't work.

Joulinar, 2021-09-10:
You need replace it for the Armbian source list as well. In your output it's https

`Err:6 https://apt.armbian.com`

MichaIng, 2021-09-10:
I pinged the admin again 😄. It is fixed for a long time, but seems for some reason an old version of the router is deployed again and again.

> You need replace it for the Armbian source list as well.

Do it only for the Armbian list. While APT has an integrated authenticity and integrity mechanism via keys and checksums, I would always recommend to keep using encrypted HTTPS connections where possible for any public access.

mydogboris, 2021-09-10:
Thank you Joulinar and Michalng.  I was changing the incorrect sources.  I appreciate your help and patience.

Joulinar, 2021-09-10:
Ok I'm going to close the issue for now. Feel free to reopen if needed.

closed 2021-09-10 by Joulinar

MichaIng, 2021-09-10:
> I was changing the incorrect sources.

The sources are not incorrect, but the routing server did an incorrect redirect from encrypted HTTPS to plain HTTP, which is reasonably denied by APT.

MichaIng, 2021-09-12:
Okay now the issue really "really" got solved. The problem was that the router sits behind a reverse proxy which is always contacted via plain HTTP. So the auto-detection didn't work, respectively the plain HTTP connection from proxy to router was detected, not the HTTPS connection from client to proxy. Now the respective proxy header is used. So I recommend to change the `armbian.list` back to `https://` entries to have those connections authenticated and encrypted.

mydogboris, 2021-09-13:
I just tried the change and still get the same failure of no release file and forbidden redirection.  Switched back to http and it worked again.

MichaIng, 2021-09-13:
Verified. Yesterday it worked reliably and it does now at `dl.armbian.com` but at `apt.armbian.com` it is now consequently redirected to plain HTTP while the scheme was detected correctly (according to response headers).

Joulinar, 2021-09-22:
@MichaIng 
issue is back again. https://dietpi.com/phpbb/viewtopic.php?p=38013#p38013