MichaIng/DietPi #4721: Bullseye VMware issues (Nextcloud, Blynk, etc.)

Phil1988, 2021-09-10:
### Creating a bug report/issue

#### Required Information
- DietPi version:
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version
11.0

- Kernel version
Linux DietPi 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux

- SBC model 
Virtual Machine (x86_64)

#### Additional Information (if applicable)
Freshly taken VMware image, copied over the dietpi_userdata folder from an old (Stretch->Buster->Bullseye upgraded) system.

#### Steps to reproduce
1. Select this software from dietpi-software to install: Nextcloud, Blynk, Certbot, Pihole
Webserver was chosen as apache2 before
While installing, it also asks (and I allowed) to install Unbound and other software like JRE etc.
2. During installation I got this error:
[image]
3. I connected via a different PC (to not interrupt the installation) and did `mkdir /var/log/blynk/` and chose "Retry" on the main machine. Installation continued.
4. After finishing, apache didnt start. Checking with `systemctl status apache2` showed, that the directory "/log/var/apache2" is missing. So I created it via terminal and restarted apache2.
5. Apache2 does work, Pi-Hole is accessable (not tested the functionality entirely yet), but Nextcloud is not working with this message via browser:
[image]

#### Expected behaviour
The log files should be created on the software installation
Nextcloud should work :) 

Normally I just wanted to post whats working at [The Debian Bullseye image updates sticky](https://github.com/MichaIng/DietPi/issues/4555), but I will do after I can confirm everything for sure :)

Joulinar, 2021-09-10:
Hi,

I did a test installation on a VirtualBox VM running Bullseye and used the same software components as you. But I wasn't facing any issues. There are no error messages during the entire installation and all services are up and running afterwards.

Regarding your log folder issue: did you changed the log method from RAMlog to FULL or something like this in parallel?

Nextcloud: DietPi will always install latest available NextCloud version. Current version is 22.1.1.2 and it's supporting PHP 7.4 as shown on NextCloud docs https://docs.nextcloud.com/server/latest/admin_manual/installation/system_requirements.html

Do you know the NextCloud version of the system you copied over? Maybe it was an older version not compatible with PHP version 7.4. You could try to check for updates

```
ncc maintenance:mode --on
sudo -u www-data php7.4 /var/www/nextcloud/updater/updater.phar
ncc maintenance:mode --off
```

Phil1988, 2021-09-10:
That is interessting, because I had this exact problem twice (using VM Workstation 14).

Regarting the RAMlog to FULL: No I did not change anything there.
The current setting are looking like this:
[image]

I am pretty sure I was on 21.0.3 (with PHP 7.3) and stayed there because after upgrading to 22.0.0 some apps didnt work any more.
Upgrade to 22.0.0 was only possible using a bit of a weird "trick" to trigger the update to this using the "beta channel" via command-line. (see here: https://github.com/MichaIng/DietPi/issues/4511)

However updating NC looks like this:
[image]

Do I understand this correctly, that NC is on Version 14?

Joulinar, 2021-09-10:
yes your current version is 14. I just can assume you copied a pretty old dietpi_userdata folder

Usually `/var/log/` folders should be created automatically by the software installation process. 
@MichaIng any ideas why they are missing?

Phil1988, 2021-09-10:
Hmmm that is strage, because there is just one dietpi_userdata folder.
This includes an nextcloud_data that has a backups from version 19 in it... so at least it must be a version 19/20 one...
[image]

MichaIng, 2021-09-10:
> yes your current version is 14. I just can assume you copied a pretty old dietpi_userdata folder

The Nextcloud version is not stored in `/mnt/dietpi_userdata` (aside of backups which you don't need to use unless a Nextcloud internal update broke) but only in `/var/www/nextcloud`. It seems like you copied or replaced quite some other files/dirs from/in `/var` to the new system?

> that if there is a folder inside the dietpi_userdata, that is named like the software, it will then not make that log/var/ folder ?!

That is completely unrelated. The Apache2 package install itself creates the `/var/log/apache2` directory, so as long as that package is installed, the directory is.
In case of Blynk, there is indeed the issue that the logs directory is only created when `/mnt/dietpi_userdata/blynk/server.properties` does not yet exist, else it is assumed that it exists already. I'll fix that.

Phil1988, 2021-09-10:
I just did a complete dietpi-sync from the old /mnt/dietpi_userdata/ to the new one.

I did not touch any other folder (this includes also /var/www/nextcloud or the /var folder in general).

Should I uninstall nextcloud, copy the /var/www/nextcloud/ folder to the new system and reinstall it again?

Regarding apache2:
After Rebooting the system I saw the [FAILED] apache2 message (something like this).
The status gave me a missing /var/log/apache2 error.

I also went back the the snapshot after the rsync of the dietpi_userdata folder plus installing
Nextcloud
Blynk
Certbot
Pi-hole
(Unbound)
all at once

After that I changed the keyboard locales to german and rebootet the system.

Here is a video of a snapshot that I made right after just the above mentioned (sadly not earlier)
https://youtu.be/CgBVD83yPA0

As you can see, the missing /log/var/apache2/ was the problem that this service could not start.

Phil1988, 2021-09-10:
I did try to remove this folder, uninstall nextcloud and install it back again, but got stuck with the following error:

[image]
Any ideas?

MichaIng, 2021-09-10:
> invalid for anther 24min 47s

Your system time is out of sync. Did you disable the time sync or switched to custom mode? This is actually okay in VMware as long as you have the host <> guest time sync enabled in the VM settings. Else keep time sync on at least daily basis enabled, which implies a sync at boot.

Phil1988, 2021-09-10:
Hmm... yeah I remember that I also had to change this like 2 years ago on the Stretch VM (see here https://github.com/MichaIng/DietPi/issues/2777#issuecomment-491616889)..

My settings have been like this:
[image]

I changed it to this:
[image]

I did this using putty with an additional terminal.

Then swichted back to retry the command and it worked until here:
[image]

For me it seems that the `/var/www/nextcloud/` folder also has to be copied from the old to the new system, right?

Joulinar, 2021-09-10:
well you are using an old database file or there is a different password on config file now as you copied it from old system,. Now it doesn't fit to the one set during re-installation. 

Set new passowrd for NectCloud you would need to fetch password from config file
```
root@DietPi:~# cat /var/www/nextcloud/config/config.php | grep dbpassword 
    'dbpassword' => 'topsecret', 
root@DietPi:~#
```

```
mysql -u root -e "SET PASSWORD FOR 'oc_admin'@localhost = PASSWORD(\"topsecret\");"
dietpi-services restart
```

Phil1988, 2021-09-10:
Thanks Joulinar,

I did the `mysql -u root -e "SET PASSWORD FOR 'oc_admin'@localhost = PASSWORD(\"topsecret\");"`

and replaced the "topsecret"-String of cource accordingly, but I got this error:
`ERROR 1133 (28000) at line 1: Can't find any matching row in the user table`

Joulinar, 2021-09-10:
Show user
`mysql -e "SELECT User, Host FROM mysql.user;"`

Show database
```
mysql -e "SHOW DATABASES;"
mysql -e "SELECT db, host, user FROM mysql.db;"
```

Joulinar, 2021-09-10:
your system is confusing more and more. Now you are missing NextCloud database user 😨 
Let's try to create
`mysql -e "grant all privileges on nextcloud.* TO 'oc_admin'@localhost' identified by 'topsecret';"`

check user again
`mysql -e "SELECT User, Host FROM mysql.user;"`

Phil1988, 2021-09-10:
Yeah... the thing is, that my HDD partially died and I could ne perform a dietpi-backup...
So I took a new bullseye VM image and just copied the /mnt/dietpi_userdata/ folder to the new system.

I thought thats all needed to restore it (if I understood this https://github.com/MichaIng/DietPi/issues/4670#issuecomment-903701911 correctly)

```
root@DietPi:~# mysql -e "grant all privileges on nextcloud.* TO 'oc_admin'@localhost' identified by 'TOPSECRET';"
ERROR 1064 (42000) at line 1: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' identified by 'zMN3B+JQCo4rFGrPa1C5/UQZNo0tvE'' at line 1

```

```
root@DietPi:~# mysql -e "SELECT User, Host FROM mysql.user;"
+-------------+-----------+
| User        | Host      |
+-------------+-----------+
| mariadb.sys | localhost |
| root        | localhost |
+-------------+-----------+
root@DietPi:~#
```

MichaIng, 2021-09-10:
Do you still have the change to run the [migration script](https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-cloud-migration) on the old system, to have a fresh backup of all three, database, install directory and data directory to copy to the new system? Otherwise copy the whole `/var/www` content together with `/mnt/dietpi_userdata`. The first makes perfectly sense anyway, and if a full MariaDB dump has been done (which always makes sense), copying those two directories makes the migration script obsolete. MariaDB was able to migrate the raw table files whenever I tested it and with fitting Nextcloud install dir and database, those parts are skipped and only the webserver, PHP and Redis install and setup done around it.

I checked and we reasonably deny an install if only the database is present but no install directory, as the risk is too high that we do wrong assumptions and break the instance.

I wonder how the install actually went that far. Does '/mnt/dietpi_userdata/dietpi-nextcloud-database-backup.sql' exists as well from the same old NC14 instance?

Phil1988, 2021-09-10:
I can try to make the old VM readable... 
adding the old vmdk to the new one as a "hard drive" did additionally change something, so that the old VM cant be started any more.

I can try to bring it alive tho... but if I have success, I currently dont know how to run the migration script.
Just do `dietpi-cloud-migration`?

MichaIng, 2021-09-10:
I edited the above post. Don't use the migration script then, instead:
- To assure no leftovers from those install attempts are present, remove (or better rename to have a backup) `/var/www` and `/mnt/dietpi_userdata` from the new system and copy those two from the old one.
- Remove `/mnt/dietpi_userdata/dietpi-nextcloud-database-backup.sql` and `/mnt/dietpi_userdata/dietpi-nextcloud-installation-backup` so that nothing from the old NC14 instance is tried to be recovered.
- Then rerun `dietpi-software install 114`.

I think I'll remove the migration script and its handling. It makes the install code complicated and it makes more sense to copy the raw files and have a full database dump only as an additional backup, in case MariaDB really fails to migrate the raw table files (which I haven't seen so far).

Joulinar, 2021-09-10:
did you uninstall PiHole as well? If yes you probably confirmed to remove packages on PiHole uninstall script?

MichaIng, 2021-09-10:
> Copy over the dietpi_userdata takes almost 3 days

Okay, as the install didn't went through, I think it is sufficient when you keep `/mnt/dietpi_userdata` on the new system and don't copy this over again, but only `/var/www` then along with the other two steps above.

Phil1988, 2021-09-10:
just to make it clear:
Are you sure I should delete the entire /var/www/ folder?
I would only delete (but with a backup) the /var/www/nextcloud/ folder, as in /var/www is also pihole, html, admin

Joulinar, 2021-09-10:
never, never confirm to remove any packages if ask by PiHole. This will break your system. We tried to convince PiHole goys to change the behaviour but still the same dangerous question on their uninstall script :)

Phil1988, 2021-09-10:
Hahah... yeah... I also felt a big "duhhh", when I was asked to confirm the purge of all these depencies :D  
lesson learned

MichaIng, 2021-09-10:
> I would only delete (but with a backup) the /var/www/nextcloud/ folder, as in /var/www is also pihole, html, admin

Ah yes, as you installed Pi-hole already, removing `/var/www/nextcloud` only is fine.

> I also felt a big "duhhh", when I was asked to confirm the purge of all these depencies :D

Old issue, reminds me to go on with that topic at Pi-hole to have clearer warnings, defaults to "no", skip the important system core packages (iproute2, libcap2-bin, psmisc, uiuiui) or skip the whole thing and only print a message about which packages may have been installed by Pi-hole. On first attempt it ended with some unrelated fight with one of the Pi-hole devs and hence I stopped. But we are good now and hence we are able to productively further steps on this topic now.

Phil1988, 2021-09-10:
As `dietpi-software install 114` did not work (because already installed) I did
`dietpi-software reinstall 114`

It stopes here:
[image]

MichaIng, 2021-09-10:
> because already installed

Oh the install even went through?

Possible reason is that the database directory user is invalid, as the user ID did change:
```sh
ls -l /mnt/dietpi_userdata/mysql
```
To fix it:
```sh
chown -R mysql:mysql /mnt/dietpi_userdata/mysql
systemctl restart mariadb
```

Phil1988, 2021-09-10:
Im not sure... I would say it is stuck in between because od the MariaDB puts out errors and the apache2 was already installed, so it added some additional configs.

i made a video while re-installing:
https://www.youtube.com/watch?v=zPKTtFLxtS8

Phil1988, 2021-09-10:
I can confirm, that this is working:
Blynk
Certbot
Pi-hole
vsftpd

Nextcloud does also wokk, bit it gave me again some fearfull moments :D:D
Overall it works, but it was on Version 20.0.5 which is odd, because it should have been on 21.0.3.
I remember, that from 20.0.5 it was only possible to upgrade further using the "beta" version and/or push it from the command line to do the update...
Maybe something internally did go wrong on that path?

I did update to 20.0.12, but first errors happened:
[image]
[image]

I then had to "try to" put the maintainance mode off
[image]

[image]

and then id did go through
[image]
[image]

MichaIng, 2021-09-10:
Finally, more difficult than expected but at least now it works. I took quite a few hints for our blog from it, regarding system migration to a freshly flashed one, like copying `/var/www` as well and checking + in case fixing `/mnt/dietpi_userdata/<program>` owners.

I furthermore applied a change to the script: https://github.com/MichaIng/DietPi/commit/9415f73ae632733106a61c1b35aab353e0250046
The package postinst script actually tries to apply the correct owner to an existing directory, but it tries to find the files via `find /var/lib/mysql`, which only finds the symlink but not the content. So it only does `chown mysql /var/lib/mysql`, which effectively changes the owner for `/mnt/dietpi_userdata/mysql` but not the content. Luckily even that the service restart hence fails as part of the postinst script, it does not break the package install itself, so we can check and fix permissions afterwards without having any break.

Phil1988, 2021-09-10:
Yeah it indeed was not "basic".... (at least for me) and thanks to you smart guys everything works now.

I hope that my problems will help others with your great "behind the scenes" work of posting it at the blog (which is btw pretty well written).

Nextcloud still doesnt like to go to v22 tho :D 
[image]

closed 2021-09-10 by Phil1988

MichaIng, 2021-09-10:
Btw, upgrading Nextcloud may work a little more reliably via command line:
```sh
sudo -u www-data php /var/www/nextcloud/updater/updater.phar
```
or if the first part was done via web interface already:
```
ncc upgrade
```

Still strange that you don't get the current version offered. Have you tried switching to beta temporarily, and if/after update back to stable? There is no NC23 beta yet, so it can only offer you the latest NC22.

Phil1988, 2021-09-11:
I will do what you mentioned to try to push the "beta" NC22 upgrade and report back.

Meanwhile I started the old dietpi instance (where I copied all the files from) and can confirm that it was [indeed](https://github.com/MichaIng/DietPi/issues/4721#issuecomment-917081045) a Nectcloud v21.0.3:
[image]

I have completely NO idea why the recovered version was a Nextcloud v20.0.5

I also checked the files on the current cloud.
The latest file change has been done 4 month ago:
[image]

While the "original" has of course a lot newer files.

Luckily this is nothing to worry about, because I now just have to make sure, that the nextcloud sync is done correctly from my devices to the cloud and I (hopefully) did not change that much regarding the server services in that time.

Just a bit confusing and maybe helpful for you to know, that this does not result in the latest status of the cloud...

MichaIng, 2021-09-11:
> The latest file change has been done 4 month ago:

Hmm, was the database and system ordinary stopped/shutdown before you copied over the database files? Although non-temporary tables should be synced to disk regularly anyway. Could you have a look whether the missing files are actually present in:
```sh
ls -l /mnt/dietpi_userdata/nextcloud_data/<username>/files/
```

Phil1988, 2021-09-12:
So I switched to beta and did the `sudo -u www-data php /var/www/nextcloud/updater/updater.phar`.
I am now on Nextcloud 22.1.1

The missing files are not in 
`/mnt/dietpi_userdata/nextcloud_data/<username>/files/`

But now I think I have a REAL problem.
On my Laptop the Nextcoud client was still running and I worked yesterday on it.

It looks like the Nextcloud client did delete files on my windows 10 laptop in backgroud.
[image]

Today I copied the nextcloud-sync folder and wanted to start the client... then I saw, that it was already running and deleted the files localy on the laptop.
Of course these files are not on the cloud and never have been.
I had a look at the windows trash bin, but they are not there.
Now Im really worried, because these files are pretty important and the weekly backup would be tonight :(

The only idea I currently have is to use a regular recovery software (which I experienced to be semi successfull in the past).

Does the client really silently delete files unrecoverable or is there any way to get these files back?

MichaIng, 2021-09-12:
> Of course these files are not on the cloud and never have been.

So these were the newly created files from the last four months which were missing after the migration? Whether files on the client which are missing on the server are synced to the server or removed from the client depends on the timestamps and the sync protocol, I guess, but I'm also not 100% sure about the details. The Nextcloud community should be able to answer this: https://help.nextcloud.com/

But indeed if Nextcloud client removes files, they are permanently removed, not moved to the trash bin. So try with recovery software quickly, if these files are not still stored on another client.

What exactly has gone wrong is hard to say with this strange incomplete migration from an older Nextcloud version, however this was possible, including the older data directory (as you say the files are indeed not there) and the matching older database. I'm puzzled also how such can be when you shutdown that server and copy the those files over to a new server. That four months or filesystem writes are cleanly lost, without corruption and consistent across files, database and NC install directory, is next to impossible. I guess it has to do with the previous trouble of that VM drive. Probably it stored to a revision while you copied from a snapshot of that drive or so (okay snapshots are a VirtualBox thing).

Phil1988, 2021-09-13:
> > Of course these files are not on the cloud and never have been.
> 
> So these were the newly created files from the last four months which were missing after the migration? 

yes exactly.

> 
> But indeed if Nextcloud client removes files, they are permanently removed, not moved to the trash bin. 

very good to know

> What exactly has gone wrong is hard to say with this strange incomplete migration from an older Nextcloud version, however this was possible, including the older data directory (as you say the files [...]

Yes you are right it is indeed strange.

I did try Recuva over night, but surprisingly I can not even see the deleted files.
Much older files are visible (but mostly not well recoverable -> with a red or orange symbol).
Maybe NVMe SSDs are harder to recover than HDDs.

The good thing is, that my PC did a complete sync on 15.08. (and the PC NC client is off).

So the most important data is still there and only ~3 weeks of files are gone.
The latest ones are still open on my laptop and I saved them at a new directory.
So overall I was really lucky!

Very good to know, that die NC Client deletes the files in a somehow unrecoverable way (at least not a trash bin way.. and in my case really not easy to recover.... but I guess strange things are tending to come to me these days  :D:D ).

MichaIng, 2021-09-13:
Actually it would be good to learn some details about how Nextcloud handles file removals. It is rare, but you're not the only case where this happened unexpectedly. And of course it is an extremely problematic behaviour, so even with very rare cases, this is something to investigate and probably fix at Nextcloud side. In case there is no 100% clear removal log server side, it is always better to either sync from the client back to the server or do nothing but a conflict warning when unclear, before removing files locally. But I lack the details about how removals are tracked server side and then synced/tracked at the client. Again, something to ask at the Nextcloud forum, if not a similar question came up already (it did, but probably hard to find).

I think when storing such large amount of data from hundreds of GiB to TiB, it is better to have them stored on a real drive instead of a resulting huge single virtual drive file (even when those can be split). Visualizers support both, a shared directory between host and guest, and passing through USB devices/drives and probably even SATA/SCSI ones to the guest. The first is not really handy in case of Nextcloud, as the server expects UNIX permissions (a problem when the host is Windows) and also direct raw file accesses are not tracked by Nextcloud, so that the database would lack behind. So the second looks most promising to me: So you have a physical drive that you do not mount on the host but pass through to the guest to have a native Linux filesystem and the Nextcloud data on it. So if anything happens with the VM, the physical drive can be attached to any other Linux system, and it does not all depends on the integrity of the single huge virtual drive file.