MichaIng/DietPi #4728: Open Beta v7.5 | Please help testing and hardening the upcoming release

MichaIng, 2021-09-11:
RC version | v7.6.0
---------- | -------
v7.6.0 PR | #4727
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2021-08-18**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.5
____
### Beta v7.6.0
_(2021-09-11)_

#### Changes
- General | Initial detection and support for Debian 12 Bookworm (the new "testing" version) has been added to DietPi. Everyone is invited to upgrade to Bookworm to stay on bleeding edge. Just note that due to continued breaking changes coming with package upgrades, some features and software installs are broken or will break. We are happy to receive your bug report then, to have necessary changes implemented in DietPi as fast as breaking changes in Bookworm happen, until it becomes the new stable Debian release, expected in summer 2023.
- DietPi-FS_partition_resize | On first boot, DietPi automatically expands the root partition and filesystem to cover the full disk size. In some circumstances, e.g. on older kernel versions, it may happen that the two used commands "partprobe" and "partx -u" fail to inform the kernel about the changed partition table and hence the filesystem is not expanded. In such case, the system is now rebooted automatically once to assure that the new partition table is loaded and the filesystem is then expanded. This change affects only new images with contain DietPi v7.6 already, since this expansion is done before DietPi updates itself. Many thanks to @Dtrieb for reporting a case where the filesystem expansion failed: https://github.com/MichaIng/DietPi/issues/4582
- DietPi-Drive_Manager | Native Linux exFAT support is now detected and respected by skipping the obsolete FUSE driver install in case. Additionally the new "exfatprogs" are installed on Bullseye, written and implemented with Debian Bullseye along with the native Linux exFAT implementation.
- DietPi-Drive_Manager | exFAT mounts now have 775 mode and are owned by the "dietpi" group, if it was not manually removed. By default, when the FUSE driver is used, mounts have 777 mode so that all users have full access, which isn't optimal for security reasons. With native Linux exFAT support mounts have 755 mode by default, so that DietPi-Software's download and media software implementations don't have write access. "775 root:dietpi" is a good compromise if the "dietpi" group exists, matching the permissions for content directories in /mnt/dietpi_userdata. Many thanks to @K92Pi for reporting a related issue: https://github.com/MichaIng/DietPi/issues/4680
- DietPi-Software | TigerVNC: Instead of x11vnc, TigerVNC's own scraping server is now used for shared desktop mode, which is a bit lighter and shares a lot of libraries with the TigerVNC standaline server package.
- DietPi-Software | RealVNC: Having desktop autologin enabled does not force the shared desktop VNC mode anymore. Furthermore our "vncserver.service" does not call RealVNC's "vncserver-x11-serviced.service" for the shared desktop mode, but instead the actual vncserver-x11 executable is called directly. This has some benefits, e.g. it allows to use our service for a virtual desktop while using RealVNC's service to allow connections to the shared local desktop independently. Many thanks to @K92Pi for bringing this idea to us: https://github.com/MichaIng/DietPi/issues/4672
- DietPi-Software | RealVNC: The VNC server is now started with "VncAuth" authentication by default, which allows any VNC viewer to connect, not just RealVNC's ones. Use "vncpasswd" to change the password, which is now independent of the UNIX user passwords. It defaults to the global software password on a fresh install.
- DietPi-Software | Amiberry: Due to changes in DietPi v7.5, enabling Amiberry fast boot won't change the TTY for boot/kernel on Raspberry Pi anymore, as they do not directly interfere with Amiberry screen output anymore. But due to this change, during screen mode changes, the raw console output is visible for a short time, which breaks the Amiga feeling. To address the issue, instead of changing the kernel command line (cmdline), Amiberry will now switch to a different expectedly empty TTY on startup and switch back to the main TTY when being stopped, including the case of failure. As this is achieved with the Amiberry service itself, the Amiberry standard boot option benefits from it as well, as well as manual "systemctl start amiberry" calls, all the same way, and it is not limited to the Raspberry Pi, as the previous TTY change was. Many thanks to @zompiexx for reporting this issue: https://github.com/MichaIng/DietPi/issues/4692
- DietPi-Software | MPD: The service file and default mpd.conf are not overwritten anymore (from Buster on), but instead the required changes are added to the default, shipped with the Debian package. This resolves some error messages on MPD startup on Bullseye and leaves most custom changes untouched on a reinstall. Furthermore, by default MPD now logs to journal, accessible via "journalctl -u mpd". Many thanks to @maartenlangeveld for reporting the MPD startup errors: https://github.com/MichaIng/DietPi/issues/4690
- DietPi-Software | Kodi: Our Raspberry Pi accelerated Kodi package is now installed on ARMv7 RPi systems as well.

#### Fixes
- DietPi-Drive_Manager | Corrected false information when doing an exFAT filesystem check & repair: Until Stretch, only checking for exFAT errors is supported but not repairing them. From Buster on, both is fully supported.
- DietPi-LetsEncrypt | When enabling HTTPS redirect or HSTS and ownCloud or Nextcloud are installed, the "overwrite.cli.url" setting in the config.php is updated accordingly to contain the primary HTTPS domain. This is required for cron jobs and the occ/ncc commands to access ownCloud/Nextcloud through the webserver, as the Let's Encrypt certificate is only valid for the external domain name and not "localhost". Many thanks to @droogi for reporting a possibly related issue: https://github.com/MichaIng/DietPi/issues/4353
- DietPi-Software | TigerVNC: Resolved an issue where remote connections didn't work by default on Bullseye systems, as a different configuration file is used.
- DietPi-Software | LXDE: Resolved an issue on Raspberry Pi, where calling lxappearance (Customize Look and Feel) failed due to incompatible RPi desktop packages. Many thanks to @pinipon for reporting this issue: https://github.com/MichaIng/DietPi/issues/4687
- DietPi-Software | Blynk: Resolved an issue where the log directory may be missing, which breaks the service start, when the userdata were migrated from one system to a new one. Many thanks to @Phil1988 for reporting this issue: https://github.com/MichaIng/DietPi/issues/4721
- DietPi-Software | qBittorrent: Resolved an issue on Bullseye systems where login to the web interface was not possible with the global software password since the required hash algorithm has changed. Many thanks to @aftensleuk for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=22564#p22564

MichaIng, 2021-09-15:
#### Beta v7.6.1 has been merged: #4735

KamikazeePL, 2021-09-16:
There is a problem with FileBrowser user dietpi doesn't have **ADMIN** and **SHARE** permissions.
I fixed it. 
`dietpi-services stop filebrowser
 /opt/filebrowser/filebrowser users update dietpi --perm.admin --perm.share -d /mnt/dietpi_userdata/filebrowser/filebrowser.db`

Joulinar, 2021-09-16:
Thx for sharing, we will have a look.

Joulinar, 2021-09-16:
hmm at least `Share` permissions should be active by default isn't it?

```
root@DietPi4:/mnt/dietpi_userdata/filebrowser# /opt/filebrowser/filebrowser users ls
ID  Username  Scope  Locale  V. Mode  S.Click  Admin  Execute  Create  Rename  Modify  Delete  Share  Download  Pwd Lock
1   dietpi    .      en      list     false    false  true     true    true    true    true    true   true      false
root@DietPi4:/mnt/dietpi_userdata/filebrowser#
```

MichaIng, 2021-09-17:
Definitely! Many thanks for pointing this out, especially since I get my dark mode now 😄. Related docs: https://filebrowser.org/cli/filebrowser-users-add#options

MichaIng, 2021-09-18:
#### DietPi v7.6 has just been released. Many thanks to all testers! ❤️

closed 2021-09-18 by MichaIng