MichaIng/DietPi #4729: MPD won't start anymore (libvchiq_arm.so missing)

maartenlangeveld, 2021-09-11:
RPi Zero W (armv6), DietPi v7.5.2 Bullseye

Tonight I rebooted my RPi Zero W music player after a couple of weeks and MPD did not start. I have run "apt upgrade" a couple of times in between last reboot and reboot a couple of weeks ago.

Uninstall and (re-)install using "dietpi-software" menu did not solve the problem.

```
2021-09-11 23:36:08 root@M-MUSIC-PLAYER:~# service mpd status
● mpd.service - Music Player Daemon (DietPi)
     Loaded: loaded (/lib/systemd/system/mpd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2021-09-11 23:36:05 CEST; 11s ago
       Docs: man:mpd(1)
             man:mpd.conf(5)
    Process: 4284 ExecStart=/usr/bin/mpd --no-daemon /etc/mpd.conf (code=exited, status=127)
   Main PID: 4284 (code=exited, status=127)
        CPU: 397ms

Sep 11 23:36:03 M-MUSIC-PLAYER systemd[1]: Starting Music Player Daemon (DietPi)...
Sep 11 23:36:04 M-MUSIC-PLAYER mpd[4284]: /usr/bin/mpd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
Sep 11 23:36:04 M-MUSIC-PLAYER systemd[1]: mpd.service: Main process exited, code=exited, status=127/n/a
Sep 11 23:36:05 M-MUSIC-PLAYER systemd[1]: mpd.service: Failed with result 'exit-code'.
Sep 11 23:36:05 M-MUSIC-PLAYER systemd[1]: Failed to start Music Player Daemon (DietPi).
```

Joulinar, 2021-09-11:
Hi,

I guess it's a side effect of apt package update for `libraspberrypi0`. This changed the location for quite some shared libraries.

maartenlangeveld, 2021-09-11:
Have tried to install mpd manually (apt install mpd) and that did not solve the problem either. Is there a workaround? Cannot listen music or radio anymore...

Joulinar, 2021-09-11:
you could try following

```
cd /usr/lib/arm-linux-gnueabihf
ln -s libmmal_core.so.0 libmmal_core.so
ln -s libmmal_util.so.0 libmmal_util.so
ln -s libmmal_vc_client.so.0 libmmal_vc_client.so
ln -s libbcm_host.so.0 libbcm_host.so
ln -s libvcsm.so.0 libvcsm.so
ln -s libvchiq_arm.so.0 libvchiq_arm.so
ln -s libvcos.so.0 libvcos.so
systemctl restart mpd
```

@MichaIng 
another software title suffering from the change on `libraspberrypi0`

MichaIng, 2021-09-11:
I so now think the path was never the issue but only the name. There must be one of the RPi repo libraries which pulls those in but which has not yet been updated to import the new VC library names.

Looks like instead of shipping this workaround via Kodi package, we should apply it on DietPi update. Also we may try to create the symlinks at /usr/local/lib so that another libraspberrypi0 update does not fail, in case it wants to ship the symlinks as well or revert the name change.

maartenlangeveld, 2021-09-11:
@Joulinar @MichaIng 

You guys are terrific ! Joulinar, thanks for the workaround, it works. Really appreciate,

Joulinar, 2021-09-13:
@MichaIng 
there is another user reporting issues with missing shared libraries https://dietpi.com/phpbb/viewtopic.php?p=37625#p37625
We definitely need to do something on next update to work around until it has been fixed by RPi devs

Joulinar, 2021-09-13:
And next one https://dietpi.com/phpbb/viewtopic.php?f=11&t=9418

Are we the only one having such issues. Usually it should impact all Raspberry OS Bullseye users. Isn't it? 🤔

MichaIng, 2021-09-13:
Since there is no official Raspberry Pi OS Bullseye image yet, it is rare, I guess.

Joulinar, 2021-09-14:
something we would need to report? We have multiple issues on our support platform (forum + GitHub) every day now 😒

MichaIng, 2021-09-14:
Here the workaround via update: https://github.com/MichaIng/DietPi/commit/dcc548b16f147b9a78b9b05bebbe745abc367088

Issue reported: https://github.com/RPi-Distro/firmware/issues/8

MichaIng, 2021-09-14:
We're one step further. Not the Kodi build (or MPD or ReadyMedia/MiniDLNA) are the issue, but FFmpeg which pulls in the RPi VC libraries. So as fast as new FFmpeg packages are pushed to the repo, the workaround is not required anymore 😀. Now I understand why I didn't need to have `libraspberrypi-dev` installed for the first Kodi build, because it is not used by Kodi directly, but by the FFmpeg libraries, which are shipped by the RPi repo.

Joulinar, 2021-09-14:
let's see if RPi devs see the need to recreate the `FFmpeg` package as well 😉

MichaIng, 2021-09-14:
I guess that is/was planned anyway, but probably not thought about the direct dependency when libraspberrypi was updated. However, hopefully it is not more than hitting a "build" button once again, so that it is done before v7.6 release and we can hence drop the workaround.

MichaIng, 2021-09-15:
Awesome, the new FFmpeg packages are there already with:
```
ffmpeg (7:4.3.2-0+rpt2+deb11u2) bullseye; urgency=medium

  * Rebuild against new raspberrypi-userland 

 -- Serge Schneider <serge@raspberrypi.com>  Tue, 14 Sep 2021 18:32:11 +0100
```
@maartenlangeveld 
Could you do a quick test whether this makes the workaround obsolete in your case:
```
apt update
apt upgrade
cd /usr/lib/arm-linux-gnueabihf
rm libmmal_core.so
rm libmmal_util.so
rm libmmal_vc_client.so
rm libbcm_host.so
rm libvcsm.so
rm libvchiq_arm.so
rm libvcos.so
systemctl restart mpd
systemctl status mpd
```

maartenlangeveld, 2021-09-15:
Done a quick test, workaround seems obsolete after upgrade of ffmpeg.

MichaIng, 2021-09-15:
Same with Kodi 👍.

legends2k, 2021-09-19:
@MichaIng I only've `jellyfin` installed; don't have `ffmpeg`, `kodi` or `minidlna` installed.  I used to run `/usr/lib/jellyfin-ffmpeg/ffmpeg` which is now broken with

```
# ffmpeg 
ffmpeg: error while loading shared libraries: libmmal_core.so: cannot open shared object file: No such file or directory
```

Doing the aforementioned workarounds of linking to missing shared object names does work.  What would be a long term fix for me?  I've tried doing `dietpi-update`, `apt update` and `apt upgrade` after following [How to upgrade to “Bullseye”](https://dietpi.com/blog/?p=811#2.2-manual-upgrade).

MichaIng, 2021-09-19:
Makes sense. Jellyfin needs to rebuild their FFmpeg package for Raspbian Bullseye as well. Should be reported here: https://github.com/jellyfin/jellyfin-ffmpeg/issues

MichaIng, 2021-11-29:
I'll mark this as closed. Most cases have been addressed, we implemented a workaround which creates the symlinks which are also shipped by the `libraspberrypi-dev` package.

closed 2021-11-29 by MichaIng