MichaIng/DietPi #4731: Error updating to 7.5

geoffae, 2021-09-12:
#### Details:
- Date           | Sun Sep 12 20:20:44 BST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Get:3 https://packages.sury.org/php stretch InRelease [6839 B]
Err:3 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Fetched 6839 B in 2s (3043 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

geoffae, 2021-09-12:
as per #4697 output from bash -xc "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"

bash -xc "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
+ . /boot/dietpi/func/dietpi-globals
++ unset -v G_PROGRAM_NAME
++ [[ '' != [01] ]]
++ [[ -t 0 ]]
++ G_INTERACTIVE=1
++ [[ -f /boot/dietpi/.install_stage ]]
++ G_DIETPI_INSTALL_STAGE=2
++ [[ -f /boot/dietpi/.hw_model ]]
++ . /boot/dietpi/.hw_model
+++ G_HW_MODEL=2
+++ G_HW_MODEL_NAME='RPi 2 Model B (armv7l)'
+++ G_HW_ARCH=2
+++ G_HW_ARCH_NAME=armv7l
+++ G_HW_CPUID=0
+++ G_HW_CPU_CORES=4
+++ G_DISTRO=4
+++ G_DISTRO_NAME=stretch
+++ G_ROOTFS_DEV=/dev/mmcblk0p2
+++ G_HW_UUID=f9b8062d-2ed1-46fa-96ec-3267c3dbb91c
+++ G_RASPBIAN=1
+++ G_HW_ONBOARD_WIFI=0
+++ G_HW_REVISION=a01041
+++ G_HW_PCB_REVISION=1
+++ G_HW_MEMORY_SIZE=1024
+++ G_HW_MANUFACTURER='Sony UK'
++ [[ -f /boot/dietpi/.version ]]
++ . /boot/dietpi/.version
+++ G_DIETPI_VERSION_CORE=7
+++ G_DIETPI_VERSION_SUB=4
+++ G_DIETPI_VERSION_RC=2
+++ G_GITBRANCH=master
+++ G_GITOWNER=MichaIng
+++ G_LIVE_PATCH_STATUS[0]='not applied'
+++ G_LIVE_PATCH_STATUS[1]='not applied'
+++ G_LIVE_PATCH_STATUS[2]='not applicable'
++ [[ -n 7 ]]
++ [[ -n 4 ]]
++ [[ -n 2 ]]
++ [[ -n master ]]
++ [[ -n MichaIng ]]
++ [[ -f /var/www/owncloud/occ ]]
++ [[ -f /var/www/nextcloud/occ ]]
++ G_WHIP_DESTROY
++ unset -v G_WHIP_DEFAULT_ITEM G_WHIP_SIZE_X_MAX G_WHIP_BUTTON_OK_TEXT G_WHIP_BUTTON_CANCEL_TEXT G_WHIP_MENU_ARRAY G_WHIP_CHECKLIST_ARRAY
++ [[ '' == 1 ]]
++ :
+ readonly G_PROGRAM_NAME=DietPi-Pre-patches
+ G_PROGRAM_NAME=DietPi-Pre-patches
+ G_CHECK_ROOT_USER
+ ((  0  ))
+ return 0
+ G_CHECK_ROOTFS_RW
+ [[ '' == 1 ]]
+ grep -q '[[:blank:]]/[[:blank:]].*[[:blank:]]ro,' /proc/mounts
+ export G_CHECK_ROOTFS_RW_VERIFIED=1
+ G_CHECK_ROOTFS_RW_VERIFIED=1
+ G_INIT
+ export LC_ALL=C.UTF-8 LANG=C.UTF-8 PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ LC_ALL=C.UTF-8
+ LANG=C.UTF-8
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ [[ -n DietPi-Pre-patches ]]
+ [[ '' =~ ^[0-9]+$ ]]
+ export HIERARCHY=0
+ HIERARCHY=0
+ [[ '' != 1 ]]
+ local timer=0 limit=5
+ [[ -d /tmp/DietPi-Pre-patches ]]
+ trap G_EXIT EXIT
+ mkdir -p /tmp/DietPi-Pre-patches
+ cd /tmp/DietPi-Pre-patches
+ return
+ declare -F G_EXEC
+ [[ -f /etc/apt/trusted.gpg.d/dietpi-jellyfin.gpg ]]
+ apt-key list 95BD4743
+ grep -q 'expires: 2021-'
+ [[ -f /etc/apt/trusted.gpg.d/dietpi-influxdb.gpg ]]
+ [[ -d /mnt/dietpi_userdata/bitwarden_rs ]]
+ [[ 4 -lt 5 ]]
+ [[ -f /etc/apt/sources.list.d/armbian.list ]]
+ ((  7 < 7 || ( 7 == 7 && 4 < 4 )  ))
+ getent passwd debian-transmission
+ exit 0
+ G_EXIT
+ cd /tmp
+ [[ '' != 1 ]]
+ [[ -d /tmp/DietPi-Pre-patches ]]
+ rm -R /tmp/DietPi-Pre-patches
+ declare -F G_EXIT_CUSTOM

Joulinar, 2021-09-12:
This is a known issue. Pls can you do following before we fix it

```
ls -l /etc/apt/trusted.gpg.d/dietpi-php.gpg
apt-key list '95BD4743'
apt-key list
ls -l /var/lib/apt/lists
```

geoffae, 2021-09-12:
> Fix would be

Thank you, that resolved it. 

> Maybe you could consider to update to old Debian Buster or stable Debian Bullseye to ensure continues support.

On my to do list, downloaded latest image.

closed 2021-09-12 by geoffae

Joulinar, 2021-09-12:
You could do an in-place upgrade as well :)