MichaIng/DietPi #4733: dietpi-drive_manager Transfer RootFS not working in Odroid C2

yandritos, 2021-09-13:
**- DietPi version |** 
 G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

**- Distro version |** 
10.10

**- Kernel version |** 
Linux DietPi 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux

**- SBC model |**
Odroid C2 (aarch64)

**- Power supply used |** 
5V 2A 

**- SDcard used |** 
SanDisk ultra 16Gb

#### Additional Information (if applicable)
**- Was the software title installed freshly or updated/migrated?**
-Fresh install

 **Can this issue be replicated on a fresh installation of DietPi?**
Yes

**- Bug report ID | `echo $G_HW_UUID`**
[  OK  ] DietPi-Bugreport | Sending bug report
[ INFO ] DietPi-Bugreport | Reference code: f1f426f5-51d7-4d2b-bd70-ed8902dc2967

#### Steps to reproduce
1. After a fresh install of dietPI in Odroid C2
2. Execute the dietpi-drive_manager
3. Select an external HDD connected via USB
4. select the option !Transfer RootFS" to this unit and follow the steps (format ext 4, type no matter is "drive" or "partition" same result.

#### Expected behaviour
Complete the transfer process

#### Actual behaviour
Raises this next error:
_DietPi-Drive_Manager ├───────────────────────────┐
│ - Command: sed -i s##root=UUID=24d16755-9c44-4541-b2ea-0fa717ad2bec#g │
│ /boot/boot.ini │
│ - Exit code: 1 │
│ - DietPi version: v7.5.2 (MichaIng/master) | HW_MODEL: 12 | HW_ARCH: 3 | │
│ DISTRO: 5 │
│ - Image creator: DietPi Core Team │
│ - Pre-image: Meveric │
│ - Error log: │
│ sed: -e expression #1, char 0: no previous regular expression_

#### Extra details
This is the full trace of the execution:

_dietpi-drive_manager
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/sda1 > /mnt/usb_1
[ OK ] DietPi-Drive_Manager | mv .fstab /etc/fstab
[ OK ] DietPi-Drive_Manager | systemctl daemon-reload
[ OK ] DietPi-Drive_Manager | sync
dumpe2fs 1.44.5 (15-Dec-2018)
[ OK ] DietPi-Drive_Manager | umount /mnt/usb_1
[ OK ] DietPi-Drive_Manager | systemctl stop mnt-usb_1.automount
[ OK ] DietPi-Drive_Manager | rmdir /mnt/usb_1
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager | - Detected unmounted drive: /dev/sda1
[ OK ] DietPi-Drive_Manager | mv .fstab /etc/fstab
[ OK ] DietPi-Drive_Manager | systemctl daemon-reload
[ OK ] DietPi-Drive_Manager | sync
[ INFO ] DietPi-Drive_Manager | Writing zeros to partition /dev/sda1
[ OK ] DietPi-Drive_Manager | dd if=/dev/zero of=/dev/sda1 bs=4K count=1337
[ OK ] DietPi-Drive_Manager | mkfs.ext4 -F -m 0 /dev/sda1
resize2fs 1.44.5 (15-Dec-2018)
The filesystem is already 122096385 (4k) blocks long. Nothing to do!

[ OK ] DietPi-Drive_Manager | sync
[ OK ] DietPi-Drive_Manager | Created ext4 filesystem: /dev/sda1
[ OK ] DietPi-Drive_Manager | mkdir -p /mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec
[ OK ] DietPi-Drive_Manager | mount /dev/sda1 /mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager | - Detected mounted physical drive: /dev/sda1 > /mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec
[ OK ] DietPi-Drive_Manager | mv .fstab /etc/fstab
[ OK ] DietPi-Drive_Manager | systemctl daemon-reload
[ OK ] DietPi-Drive_Manager | sync
[ SUB1 ] DietPi-Services > stop
[ OK ] DietPi-Services | stop : cron
[ OK ] DietPi-Services | stop : plexmediaserver
[ INFO ] DietPi-Drive_Manager | Checking for required APT packages: rsync
[ OK ] DietPi-Drive_Manager | All required APT packages are already installed.
[ SUB1 ] DietPi-Set_swapfile > Applying 0 /var/swap
[ INFO ] DietPi-Set_swapfile | Disabling and deleting all existing swap files
[ OK ] DietPi-Set_swapfile | swapoff -a
[ OK ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_SIZE=0
[ OK ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
[ INFO ] DietPi-Set_swapfile | Setting /tmp tmpfs size: 863 MiB
[ OK ] DietPi-Set_swapfile | mount -o remount /tmp
[ OK ] DietPi-Drive_Manager | cp -a /etc/fstab /etc/fstab_new
[ OK ] DietPi-Drive_Manager | sed -i \@[[:blank:]]/mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec[[:blank:]]@d /etc/fstab_new
[ OK ] DietPi-Drive_Manager | sed -i \@UUID=[^[:blank:]]*[[:blank:]]\+/[[:blank:]]@cUUID=24d16755-9c44-4541-b2ea-0fa717ad2bec / ext4 noatime,lazytime,rw 0 1 /etc/fstab_new
[ OK ] DietPi-Drive_Manager | mkdir -p /tmp/tmp_rootfs
[ OK ] DietPi-Drive_Manager | mount /dev/mmcblk0p2 /tmp/tmp_rootfs
[ OK ] DietPi-Drive_Manager | rsync -aHv --delete /tmp/tmp_rootfs/ /mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec/
[ OK ] DietPi-Drive_Manager | mv /etc/fstab_new /mnt/24d16755-9c44-4541-b2ea-0fa717ad2bec/etc/fstab
[FAILED] DietPi-Drive_Manager | sed -i s##root=UUID=24d16755-9c44-4541-b2ea-0fa717ad2bec#g /boot/boot.ini_

MichaIng, 2021-09-13:
Many thanks for your report.

Indeed there is a bug in the script, which is able to detect the root entry on Odroid XU4 correctly (https://github.com/MichaIng/DietPi/blob/master/boot_xu4.ini#L11) but not on C2 due to the leading `"` double quote character in front "root=": https://github.com/MichaIng/DietPi/blob/master/boot_c2.ini#L126

Fixed with: https://github.com/MichaIng/DietPi/commit/ed19661a5bf09252867e19173adedd775f4c0d54
Changelog: https://github.com/MichaIng/DietPi/commit/82fb47a9ead3c732f3ac4a692dbb7ee4e5cdc412

yandritos, 2021-09-13:
Dear Michalng,

Thanks a lot for your very fast reply and the fix. I just finish to impement it and all worked!

Thanks again and have a nice day!

closed 2021-09-13 by yandritos