MichaIng/DietPi #4734: cannot upgrade dietpi to 7.5.2.

strupp666, 2021-09-13:
### Creating a bug report/issue
#### Details:
- Date           | Mon Sep 13 17:46:48 CEST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | `Linux DietPi 4.14.231+ #1 SMP PREEMPT Fri Apr 16 13:31:42 UTC 2021 armv7l GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up initramfs-tools (0.130) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
ln: failed to create hard link '/boot/initrd.img-4.14.231+.dpkg-bak' => '/boot/initrd.img-4.14.231+': Operation not permitted
cp: error writing '/boot/initrd.img-4.14.231+.dpkg-bak': No space left on device
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

strupp666, 2021-09-13:
TY

Joulinar, 2021-09-13:
Your file system is full. Can you post following

`df -h`

MichaIng, 2021-09-13:
The boot partition seems to be full, like we had in a few other cases:
```sh
ls -l /boot
```
Probably we should clean it up via DietPi update. I'll also check back the scripts whether with the new kernel packages this is indeed done automatically as expected.

strupp666, 2021-09-13:
yes that is what I presumed

Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            931M       0  931M    0% /dev
tmpfs           200M     28M  172M   14% /run
/dev/mmcblk1p2   15G    1,5G   13G   11% /
tmpfs           998M    1,9M  996M    1% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           998M       0  998M    0% /sys/fs/cgroup
/dev/sda1       1,8T    818G  923G   47% /mnt/9a6520b8-5fad-497c-8783-5b51d91fa500
tmpfs            50M    4,2M   46M    9% /var/log
tmpfs          1023M     11M 1013M    2% /tmp
/dev/mmcblk1p1   71M     71M  312K  100% /boot

Joulinar, 2021-09-13:
Pls can you post as well
`ls -l /boot`

strupp666, 2021-09-13:
Bitte schön

insgesamt 70768
-rwxr-xr-x 1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x 1 root root  151274 Okt 13  2020 config-4.14.180+
-rwxr-xr-x 1 root root  151306 Apr 16 15:34 config-4.14.231+
-rwxr-xr-x 1 root root  151102 Aug 23  2018 config-4.14.66+
drwxr-xr-x 4 root root    4096 Aug 29 23:32 dietpi
-rwxr-xr-x 1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x 1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x 1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x 1 root root 5729039 Jan 26  2021 initrd.img-4.14.180+
-rwxr-xr-x 1 root root 5740784 Jun 28 22:37 initrd.img-4.14.231+
-rwxr-xr-x 1 root root  233472 Sep 13 18:55 initrd.img-4.14.231+.dpkg-bak
-rwxr-xr-x 1 root root 5718986 Sep  2  2020 initrd.img-4.14.66+
-rwxr-xr-x 1 root root 2704571 Okt 13  2020 System.map-4.14.180+
-rwxr-xr-x 1 root root 2711917 Apr 16 15:34 System.map-4.14.231+
-rwxr-xr-x 1 root root 2695982 Aug 23  2018 System.map-4.14.66+
drwxr-xr-x 2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x 1 root root 5740848 Jun 28 22:37 uInitrd
-rwxr-xr-x 1 root root 5729103 Jun 28 22:37 uInitrd-4.14.180+
-rwxr-xr-x 1 root root 5740848 Jun 28 22:37 uInitrd-4.14.231+
-rwxr-xr-x 1 root root 5719050 Jan 26  2021 uInitrd-4.14.66+
-rwxr-xr-x 1 root root 5702280 Okt 13  2020 vmlinuz-4.14.180+
-rwxr-xr-x 1 root root 5731992 Apr 16 15:34 vmlinuz-4.14.231+
-rwxr-xr-x 1 root root 5677560 Aug 23  2018 vmlinuz-4.14.66+
-rwxr-xr-x 1 root root 5731992 Jun 28 22:37 zImage

Danke schön

Joulinar, 2021-09-13:
You could try following to clean some old kernel packages

```
dpkg -P linux-image-4.14.180+
dpkg -P linux-image-4.14.66+
```

strupp666, 2021-09-13:
After cleanup consol suggest apt --fix-broken install.
I have done this.
This is the log again

#### Details:
- Date           | Mon Sep 13 20:25:59 CEST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | `Linux DietPi 4.14.231+ #1 SMP PREEMPT Fri Apr 16 13:31:42 UTC 2021 armv7l GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up initramfs-tools (0.130) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
ln: failed to create hard link '/boot/initrd.img-4.14.241+.dpkg-bak' => '/boot/initrd.img-4.14.241+': Operation not permitted
update-initramfs: Generating /boot/initrd.img-4.14.241+
Image Name:   uInitrd
Created:      Mon Sep 13 20:25:57 2021
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    5740784 Bytes = 5606.23 kB = 5.47 MB
Load Address: 00000000
Entry Point:  00000000

gzip: stdout: No space left on device
E: mkinitramfs failure cpio 141 gzip 1
update-initramfs: failed for /boot/initrd.img-4.14.241+ with 1.
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2021-09-13:
Still your file system seems to be full. Can you share again

```
df -h
ls -la /boot
```

strupp666, 2021-09-13:
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            931M       0  931M    0% /dev
tmpfs           200M     28M  172M   14% /run
/dev/mmcblk1p2   15G    1,4G   13G   11% /
tmpfs           998M    1,9M  996M    1% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           998M       0  998M    0% /sys/fs/cgroup
/dev/sda1       1,8T    818G  923G   47% /mnt/9a6520b8-5fad-497c-8783-5b51d91fa500
tmpfs            50M    5,7M   45M   12% /var/log
tmpfs          1023M     11M 1013M    2% /tmp
/dev/mmcblk1p1   71M     63M  8,0M   89% /boot

strupp666, 2021-09-13:
insgesamt 62947
drwxr-xr-x  4 root root   16384 Sep 13 20:25 .
drwxr-xr-x 21 root root    1024 Aug 29 23:35 ..
-rwxr-xr-x  1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x  1 root root  151306 Apr 16 15:34 config-4.14.231+
-rwxr-xr-x  1 root root  151341 Jul 28 18:57 config-4.14.241+
drwxr-xr-x  4 root root    4096 Aug 29 23:32 dietpi
-rwxr-xr-x  1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x  1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x  1 root root      56 Feb 27  2020 .dropbox.device
-rwxr-xr-x  1 root root   56387 Jul 28 19:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x  1 root root   63543 Jul 28 19:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x  1 root root   62488 Jul 28 19:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x  1 root root   63518 Jul 28 19:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x  1 root root   63662 Jul 28 19:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x  1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x  1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x  1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x  1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x  1 root root 5740784 Jun 28 22:37 initrd.img-4.14.231+
-rwxr-xr-x  1 root root  233472 Sep 13 18:55 initrd.img-4.14.231+.dpkg-bak
-rwxr-xr-x  1 root root 5755702 Sep 13 20:25 initrd.img-4.14.241+
-rwxr-xr-x  1 root root 2711917 Apr 16 15:34 System.map-4.14.231+
-rwxr-xr-x  1 root root 2728720 Jul 28 18:57 System.map-4.14.241+
drwxr-xr-x  2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x  1 root root 5740848 Sep 13 20:25 uInitrd
-rwxr-xr-x  1 root root 5729103 Jun 28 22:37 uInitrd-4.14.180+
-rwxr-xr-x  1 root root 5740848 Sep 13 20:25 uInitrd-4.14.231+
-rwxr-xr-x  1 root root 5755766 Sep 13 20:25 uInitrd-4.14.241+
-rwxr-xr-x  1 root root 5719050 Jan 26  2021 uInitrd-4.14.66+
-rwxr-xr-x  1 root root 5731992 Apr 16 15:34 vmlinuz-4.14.231+
-rwxr-xr-x  1 root root 5764120 Jul 28 18:57 vmlinuz-4.14.241+
-rwxr-xr-x  1 root root 5764120 Sep 13 20:25 zImage

Joulinar, 2021-09-13:
Pls can you delete the following 2 files as well `uInitrd-4.14.180+` and `uInitrd-4.14.66+`

strupp666, 2021-09-13:
Sorry but I cannot remove them.
sudo apt purge does not work.
I´m sorry for my dummy questions.

Joulinar, 2021-09-13:
```
rm /boot/uInitrd-4.14.180+
rm /boot/uInitrd-4.14.66+
```

strupp666, 2021-09-13:
ahh okay rm works

insgesamt 51746
-rwxr-xr-x 1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x 1 root root  151306 Apr 16 15:34 config-4.14.231+
-rwxr-xr-x 1 root root  151341 Jul 28 18:57 config-4.14.241+
drwxr-xr-x 4 root root    4096 Aug 29 23:32 dietpi
-rwxr-xr-x 1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x 1 root root   56387 Jul 28 19:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x 1 root root   63543 Jul 28 19:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x 1 root root   62488 Jul 28 19:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x 1 root root   63518 Jul 28 19:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x 1 root root   63662 Jul 28 19:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x 1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x 1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x 1 root root 5740784 Jun 28 22:37 initrd.img-4.14.231+
-rwxr-xr-x 1 root root  233472 Sep 13 18:55 initrd.img-4.14.231+.dpkg-bak
-rwxr-xr-x 1 root root 5755702 Sep 13 20:25 initrd.img-4.14.241+
-rwxr-xr-x 1 root root 2711917 Apr 16 15:34 System.map-4.14.231+
-rwxr-xr-x 1 root root 2728720 Jul 28 18:57 System.map-4.14.241+
drwxr-xr-x 2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x 1 root root 5740848 Sep 13 20:25 uInitrd
-rwxr-xr-x 1 root root 5740848 Sep 13 20:25 uInitrd-4.14.231+
-rwxr-xr-x 1 root root 5755766 Sep 13 20:25 uInitrd-4.14.241+
-rwxr-xr-x 1 root root 5731992 Apr 16 15:34 vmlinuz-4.14.231+
-rwxr-xr-x 1 root root 5764120 Jul 28 18:57 vmlinuz-4.14.241+
-rwxr-xr-x 1 root root 5764120 Sep 13 20:25 zImage

strupp666, 2021-09-13:
/dev/mmcblk1p1   71M     52M   19M   74% /boot

ty

Joulinar, 2021-09-13:
Try `apt update && apt upgrade` now

strupp666, 2021-09-13:
Everything is fine now.

DietPi v7.5.2 : 21:04 - Mo 13.09.2021

Thank you very much

Should I close this issue?

Joulinar, 2021-09-13:
Wait, pls do a full reboot and post following again

`ls -la /boot`

I guess further clean up is needed.

strupp666, 2021-09-13:
insgesamt 51765
drwxr-xr-x  4 root root   16384 Jan  1  1970 .
drwxr-xr-x 21 root root    1024 Aug 29 23:35 ..
-rwxr-xr-x  1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x  1 root root  151306 Apr 16 15:34 config-4.14.231+
-rwxr-xr-x  1 root root  151341 Jul 28 18:57 config-4.14.241+
drwxr-xr-x  4 root root    4096 Sep 13 21:09 dietpi
-rwxr-xr-x  1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x  1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x  1 root root      56 Feb 27  2020 .dropbox.device
-rwxr-xr-x  1 root root   56387 Jul 28 19:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x  1 root root   63543 Jul 28 19:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x  1 root root   62488 Jul 28 19:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x  1 root root   63518 Jul 28 19:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x  1 root root   63662 Jul 28 19:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x  1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x  1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x  1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x  1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x  1 root root 5740784 Jun 28 22:37 initrd.img-4.14.231+
-rwxr-xr-x  1 root root  233472 Sep 13 18:55 initrd.img-4.14.231+.dpkg-bak
-rwxr-xr-x  1 root root 5755685 Sep 13 21:02 initrd.img-4.14.241+
-rwxr-xr-x  1 root root 2711917 Apr 16 15:34 System.map-4.14.231+
-rwxr-xr-x  1 root root 2728720 Jul 28 18:57 System.map-4.14.241+
drwxr-xr-x  2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x  1 root root 5740848 Sep 13 21:02 uInitrd
-rwxr-xr-x  1 root root 5740848 Sep 13 21:02 uInitrd-4.14.231+
-rwxr-xr-x  1 root root 5755766 Sep 13 20:25 uInitrd-4.14.241+
-rwxr-xr-x  1 root root 5731992 Apr 16 15:34 vmlinuz-4.14.231+
-rwxr-xr-x  1 root root 5764120 Jul 28 18:57 vmlinuz-4.14.241+
-rwxr-xr-x  1 root root 5764120 Sep 13 20:25 zImage

strupp666, 2021-09-13:
/dev/mmcblk1p1   71M     52M   19M   74% /boot

Joulinar, 2021-09-13:
Let's try to remove one more kernel version 231

`dpkg -P linux-image-4.14.231+`

strupp666, 2021-09-13:
done

/dev/mmcblk1p1   71M     39M   33M   55% /boot

Joulinar, 2021-09-13:
Last time check `ls -la`

strupp666, 2021-09-13:
insgesamt 37742
-rwxr-xr-x 1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x 1 root root  151341 Jul 28 18:57 config-4.14.241+
drwxr-xr-x 4 root root    4096 Sep 13 21:09 dietpi
-rwxr-xr-x 1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x 1 root root   56387 Jul 28 19:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x 1 root root   63543 Jul 28 19:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x 1 root root   62488 Jul 28 19:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x 1 root root   63518 Jul 28 19:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x 1 root root   63662 Jul 28 19:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x 1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x 1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x 1 root root  233472 Sep 13 18:55 initrd.img-4.14.231+.dpkg-bak
-rwxr-xr-x 1 root root 5755685 Sep 13 21:02 initrd.img-4.14.241+
-rwxr-xr-x 1 root root 2728720 Jul 28 18:57 System.map-4.14.241+
drwxr-xr-x 2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x 1 root root 5740848 Sep 13 21:02 uInitrd
-rwxr-xr-x 1 root root 5740848 Sep 13 21:02 uInitrd-4.14.231+
-rwxr-xr-x 1 root root 5755766 Sep 13 20:25 uInitrd-4.14.241+
-rwxr-xr-x 1 root root 5764120 Jul 28 18:57 vmlinuz-4.14.241+
-rwxr-xr-x 1 root root 5764120 Sep 13 20:25 zImage

Joulinar, 2021-09-13:
You could delete as well `initrd.img-4.14.231+.dpkg-bak` and `uInitrd-4.14.231+`

strupp666, 2021-09-13:
Okay
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            931M       0  931M    0% /dev
tmpfs           200M     14M  186M    7% /run
/dev/mmcblk1p2   15G    1,4G   13G   10% /
tmpfs           998M    1,8M  996M    1% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           998M       0  998M    0% /sys/fs/cgroup
/dev/sda1       1,8T    818G  923G   47% /mnt/9a6520b8-5fad-497c-8783-5b51d91fa500
tmpfs          1023M       0 1023M    0% /tmp
tmpfs            50M    348K   50M    1% /var/log
/dev/mmcblk1p1   71M     33M   39M   47% /boot

strupp666, 2021-09-13:
insgesamt 31906
-rwxr-xr-x 1 root root    5447 Mai  5  2020 boot.ini
-rwxr-xr-x 1 root root  151341 Jul 28 18:57 config-4.14.241+
drwxr-xr-x 4 root root    4096 Sep 13 21:09 dietpi
-rwxr-xr-x 1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root   11505 Jul 10 20:31 dietpi.txt
-rwxr-xr-x 1 root root   56387 Jul 28 19:02 exynos5422-odroidhc1.dtb
-rwxr-xr-x 1 root root   63543 Jul 28 19:02 exynos5422-odroidxu3.dtb
-rwxr-xr-x 1 root root   62488 Jul 28 19:02 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x 1 root root   63518 Jul 28 19:02 exynos5422-odroidxu4.dtb
-rwxr-xr-x 1 root root   63662 Jul 28 19:02 exynos5422-odroidxu4-kvm.dtb
-rwxr-xr-x 1 root root    6144 Jan  1  1980 FSCK0000.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0001.REC
-rwxr-xr-x 1 root root  622592 Jan  1  1980 FSCK0002.REC
-rwxr-xr-x 1 root root   12288 Jan  1  1980 FSCK0003.REC
-rwxr-xr-x 1 root root 5755685 Sep 13 21:02 initrd.img-4.14.241+
-rwxr-xr-x 1 root root 2728720 Jul 28 18:57 System.map-4.14.241+
drwxr-xr-x 2 root root    2048 Feb 27  2020 System Volume Information
-rwxr-xr-x 1 root root 5740848 Sep 13 21:02 uInitrd
-rwxr-xr-x 1 root root 5755766 Sep 13 20:25 uInitrd-4.14.241+
-rwxr-xr-x 1 root root 5764120 Jul 28 18:57 vmlinuz-4.14.241+
-rwxr-xr-x 1 root root 5764120 Sep 13 20:25 zImage

Joulinar, 2021-09-13:
@strupp666 
Looks good now. You are running oldold Debian Stretch and you might consider to upgrade to Debian Buster 😉

@MichaIng 
Do you like to include some automatic cleanup on next release?

strupp666, 2021-09-13:
TY and good night

closed 2021-09-13 by strupp666

MichaIng, 2021-09-13:
> Do you like to include some automatic cleanup on next release?

I'll do that.