MichaIng/DietPi #4738: Dietpi upgrade to bullseye issue

Reen85, 2021-09-15:
Hi,
I am currently trying to upgrade DietPi to Bullseye, but I am stuck in the middle of the process 2.2.3 UPGRADE THE SYSTEM, seeing the process aborting.

299 upgraded, 70 newly installed, 0 to remove and 7 not upgraded.
Need to get 156 MB of archives.
After this operation, 222 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Abort.

I have no idea what to do next and I do not want to ruin the installation... Please help.

Regards

Joulinar, 2021-09-15:
Pls reboot the system and retry.

Reen85, 2021-09-15:
no, unfortunately the same result:
299 upgraded, 70 newly installed, 0 to remove and 7 not upgraded.
Need to get 156 MB of archives.
After this operation, 222 MB of additional disk space will be used.
Do you want to continue? [Y/n] y   
Abort.

Joulinar, 2021-09-15:
What about `apt upgrade -y`

MichaIng, 2021-09-15:
I have similar issues with this confirmation when I type a keyboard character (like `space` or `return`) before the prompt actually appears. Also you may try to type no character at all but only `return`, as `Y` is the default answer. Also hitting `CTRL+U` clears all possible input from the line (including the invisible ones), after which hitting `return` or `y` may succeed.

Or follow Joulinar's suggestion which bypasses the manual confirmation 🙂.

Reen85, 2021-09-15:
yes, joulinar's suggestion seems to work... process continues. Crossing fingers ;-)
Thanks a lot! I'll keep you posted if this solves the issue.

Reen85, 2021-09-15:
for some reason the command
dietpi-software reinstall 38 40 47 48 56 83 84 85 88 89 93 114 143
lead to the attempt reinstall php7.3 which fails in "E: Unable to locate package php7.3-fpm".
This seems to be strange, as I thought it was remove a few steps before and switching to php7.4?

Joulinar, 2021-09-15:
do you know on which of the reinstall steps it fails? Maybe you can share some parts of the log?

Reen85, 2021-09-15:
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing PHP: Hypertext Preprocessor for dynamic web content

[ INFO ] DietPi-Software | APT install for: php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql php7.3-sqlite3 php7.3-redis, please wait...
E: Unable to locate package php7.3-fpm
E: Couldn't find any package by glob 'php7.3-fpm'
E: Couldn't find any package by regex 'php7.3-fpm'
E: Unable to locate package php7.3-apcu
E: Couldn't find any package by glob 'php7.3-apcu'
E: Couldn't find any package by regex 'php7.3-apcu'
E: Unable to locate package php7.3-curl
E: Couldn't find any package by glob 'php7.3-curl'
E: Couldn't find any package by regex 'php7.3-curl'
E: Unable to locate package php7.3-gd
E: Couldn't find any package by glob 'php7.3-gd'
E: Couldn't find any package by regex 'php7.3-gd'
E: Unable to locate package php7.3-mbstring
E: Couldn't find any package by glob 'php7.3-mbstring'
E: Couldn't find any package by regex 'php7.3-mbstring'
E: Unable to locate package php7.3-xml
E: Couldn't find any package by glob 'php7.3-xml'
E: Couldn't find any package by regex 'php7.3-xml'
E: Unable to locate package php7.3-zip
E: Couldn't find any package by glob 'php7.3-zip'
E: Couldn't find any package by regex 'php7.3-zip'
E: Unable to locate package php7.3-mysql
E: Couldn't find any package by glob 'php7.3-mysql'
E: Couldn't find any package by regex 'php7.3-mysql'
E: Unable to locate package php7.3-sqlite3
E: Couldn't find any package by glob 'php7.3-sqlite3'
E: Couldn't find any package by regex 'php7.3-sqlite3'
E: Unable to locate package php7.3-redis
E: Couldn't find any package by glob 'php7.3-redis'
E: Couldn't find any package by regex 'php7.3-redis'
[FAILED] DietPi-Software | APT install for: php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql php7.3-sqlite3 php7.3-redis
 - Command: apt-get -qq --allow-change-held-packages install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql php7.3-sqlite3 php7.3-redis

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Sep 15 20:55:37 BST 2021
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql php7.3-sqlite3 php7.3-redis`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to locate package php7.3-fpm
E: Couldn't find any package by glob 'php7.3-fpm'
E: Couldn't find any package by regex 'php7.3-fpm'
E: Unable to locate package php7.3-apcu
E: Couldn't find any package by glob 'php7.3-apcu'
E: Couldn't find any package by regex 'php7.3-apcu'
E: Unable to locate package php7.3-curl
E: Couldn't find any package by glob 'php7.3-curl'
E: Couldn't find any package by regex 'php7.3-curl'
E: Unable to locate package php7.3-gd
E: Couldn't find any package by glob 'php7.3-gd'
E: Couldn't find any package by regex 'php7.3-gd'
E: Unable to locate package php7.3-mbstring
E: Couldn't find any package by glob 'php7.3-mbstring'
E: Couldn't find any package by regex 'php7.3-mbstring'
E: Unable to locate package php7.3-xml
E: Couldn't find any package by glob 'php7.3-xml'
E: Couldn't find any package by regex 'php7.3-xml'
E: Unable to locate package php7.3-zip
E: Couldn't find any package by glob 'php7.3-zip'
E: Couldn't find any package by regex 'php7.3-zip'
E: Unable to locate package php7.3-mysql
E: Couldn't find any package by glob 'php7.3-mysql'
E: Couldn't find any package by regex 'php7.3-mysql'
E: Unable to locate package php7.3-sqlite3
E: Couldn't find any package by glob 'php7.3-sqlite3'
E: Couldn't find any package by regex 'php7.3-sqlite3'
E: Unable to locate package php7.3-redis
E: Couldn't find any package by glob 'php7.3-redis'
E: Couldn't find any package by regex 'php7.3-redis'
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2021-09-15:
did you already executed as it looks like your system is being detected as Buster and not as Bullseye

```
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
```

can you check as well `cat /etc/os-release`

Reen85, 2021-09-16:
Hi guys, thanks for support!
The "cat" command showed I have already been on bullseye, don't know what went wrong. 
The goods news is, that the "/boot/dietpi/func/dietpi-obtain_hw_model" command seemed to work well, software reinstallation went without any issues, awesome!

Thanks for your support, everything seems to work now!

Joulinar, 2021-09-16:
ok in this case I'm going to close the issue. Fell free to reopen if needed.

closed 2021-09-16 by Joulinar