MichaIng/DietPi #4739: Nginx config error: /var/log/nginx/error.log (no such file or directory) after rebooting pi-hole in a vbox VM.

hashborgir, 2021-09-15:
### Creating a bug report/issue

Hello. I am not sure why this happens. Here is the problem. Provided is my solution.

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7, G_DIETPI_VERSION_SUB=5, G_DIETPI_VERSION_RC=2
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux
- SBC model | VirtualMachine (x86_64)
- Power supply used | N/A
- SDcard used | N/A

#### Additional Information (if applicable)
- Software title | N/A
- Was the software title installed freshly or updated/migrated? Freshly.
- Can this issue be replicated on a fresh installation of DietPi? I don't know.

#### Steps to reproduce

1. Install diet pi in virtualbox vm, per the download.
2. Install pihole with nginx
3. When all is done and system is up, everything is hunky dory and works fine.
4. After some time, needed to reboot host machine. I closed the dietpi vm, rebooted, restarted dietpi vm, pi hole is functional, but nginx httpd fails to start the service, and so no http GUI.

#### Expected behaviour

- nginx should start.

#### Actual behaviour

- nginx fails to start because  `/var/log/nginx/error.log (no such file or directory)`. 
- if you type `nginx -t` it passes syntax check but fails config file check since the specified path to the error_log does not actually exist on the FS, so nginx fails as the config is broken.

#### Extra details

- /var/log/nginx is not a valid directory that exists. 
- mkdir /var/log/nginx
- touch /var/log/nginx/error.log (not sure if required, since creating the directory should allow the httpd to create the log file)
- `$service nginx start; service nginx status` and it should be running now.

Joulinar, 2021-09-15:
Hi,

question: did you safely shutdown the VM or just cancelled the running instance? Because it's needed to always `poweroff` the system to ensure to write down files from `tmpfs`.

Usually PiHole should be reachable `http://your_IP/admin`

hashborgir, 2021-09-15:
> Hi,
> 
> question: did you safely shutdown the VM or just cancelled the running instance? Because it's needed to always `poweroff` the system to ensure to write down files from `tmpfs`.
> 
> Usually PiHole should be reachable `http://your_IP/admin`

I did `shutdown` command.

Joulinar, 2021-09-15:
does `/var/log/nginx` still exist after a reboot?

hashborgir, 2021-09-15:
> does `/var/log/nginx` still exist after a reboot?

It does.

Thank you for said information. I believe perhaps not shutting down the vm cleanly might have caused an issue in the file system.

Joulinar, 2021-09-15:
`/var/log` and `/run` are temporary file systems (tmpfs) and it's important to perform a clean `poweroff`. Especially after a software installation to ensure files are written down from tmpfs to disk.

Are you able to access PiHole now?

MichaIng, 2021-09-16:
In case of `/run` nothing from there is stored back to disk, for the logs indeed a clean shutdown is required at least once after a new software install to have the possibly new log directory stored on disk for DietPi-RAMlog to recover at boot.

closed 2021-09-18 by MichaIng