MichaIng/DietPi #4740: qBittorrent (PH mirror)

jcchikikomori, 2021-09-16:
#### Details:
- Date           | Fri Sep 17 00:30:43 +08 2021
- Bug report     | 9e5b80f6-8121-4d98-bcec-5464b43c7b01
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux obama 5.10.60+ #1449 Wed Aug 25 14:59:20 BST 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install qbittorrent-nox`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://mirror.rise.ph/raspbian/raspbian/pool/main/q/qbittorrent/qbittorrent-nox_4.2.5-0.1_armhf.deb  File has unexpected size (5092016 != 5082768). Mirror sync in progress? [IP: 43.226.6.79 80]
   Hashes of expected file:
    - SHA256:bd4ba51e558492076d397d5a537918d296b21a45c528767b684f1a5530d568d2
    - SHA1:837a4608111bca8d3d2e91dc6ebc402845035f44 [weak]
    - MD5Sum:143694ef0b1ee9667e9d33625a50dfb7 [weak]
    - Filesize:5082768 [weak]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2021-09-16:
Looks like a temporary issue at apt package resource side. Something out of our control. Could you try again later?

jcchikikomori, 2021-09-17:
> Looks like a temporary issue at apt package resource side. Something out of our control. Could you try again later?

`rise.ph` mirror is always terrible, since i started using Linux. can i disable this mirror on my own, though?

Joulinar, 2021-09-17:
Well this redirect is done directly at Raspbian package source web side. In theory you could change package source list away from global pool to a specific mirror.

GvY85, 2021-09-18:
If you want a more recent build of qbitttorrent(-nox) I suggest using this reo: https://github.com/qbittorrent/qBittorrent/issues/12616#issuecomment-875368452

jcchikikomori, 2021-09-19:
I found a workaround for this.
https://github.com/c0re100/qBittorrent-Enhanced-Edition
[image]

MichaIng, 2021-09-19:
Well, good to know bit not relates to the faulty Raspbian APT mirror, is it? 🙂

jcchikikomori, 2021-09-19:
> Well, good to know but not related to the faulty Raspbian APT mirror, is it? 🙂

Yes. But the mirror issue still persists as of this moment

MichaIng, 2021-09-21:
I checked the actual files and now the hashes in the package list files match the actual package file, so it should work now:
- http://mirror.rise.ph/raspbian/raspbian/dists/bullseye/main/binary-armhf/
- http://mirror.rise.ph/raspbian/raspbian/pool/main/q/qbittorrent/

closed 2021-10-04 by jcchikikomori