MichaIng/DietPi #4742: DietPi Update Issue

buffalo0207, 2021-09-17:
#### Details:
- Date           | Fri Sep 17 01:12:37 BST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.52-v8+ #1440 SMP PREEMPT Tue Jul 27 09:56:40 BST 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `apt-get --allow-releaseinfo-change-suite -q update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Err:1 https://deb.debian.org/debian buster InRelease
  Temporary failure resolving 'deb.debian.org'
Err:2 https://archive.raspberrypi.org/debian buster InRelease
  Temporary failure resolving 'archive.raspberrypi.org'
Err:3 https://deb.debian.org/debian buster-updates InRelease
  Temporary failure resolving 'deb.debian.org'
Err:4 https://deb.debian.org/debian-security buster/updates InRelease
  Temporary failure resolving 'deb.debian.org'
Err:5 https://deb.debian.org/debian buster-backports InRelease
  Temporary failure resolving 'deb.debian.org'
Reading package lists...
W: Failed to fetch https://deb.debian.org/debian/dists/buster/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian/dists/buster-updates/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian-security/dists/buster/updates/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian/dists/buster-backports/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2021-09-17:
Hi,

This looks like a network issue on your side as you are not able to resolve DNS name for global debian.org domain. Pls check your name resolution.

danmoo117, 2021-09-21:
I had this same issue with one of the pi's I manage, but it was more intermittent as some of the hits were successful and others weren't. And coincidently I lost external ssh access too. 

The resolution was to ssh in on the local network using an on-site pc as the "bridge" and then follow the buster to bullseye upgrade steps. Once this was done the dietpi-update process ran without issue and my external ssh access was restored.

MichaIng, 2021-09-21:
Just to make it clear: You do **not** need to upgrade the system from Buster to Bullseye in any case. Moreover, when APT fails to resolve hostnames (`archive.raspberrypi.org` failed as well), then the distro upgrade would fail for the same reason.

To check the resolver, please run the following commands and paste the output here:
```sh
getent ahosts deb.debian.org
cat /etc/resolv.conf
```

MichaIng, 2021-09-21:
This was more addressed to OP 😉.

MichaIng, 2021-10-20:
I'll mark this as closed, which was likely a temporary issue with DNS only.