MichaIng/DietPi #4743: I am facing this error

ygowthamr, 2021-09-17:
#### Details:
- Date           | Fri Sep 17 05:11:26 BST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -c 1 -W 10 9.9.9.9`
- Exit code      | 1
- Software title | DietPi-Login
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2021-09-17:
Hi

This looks like a network issue on your side as you are not able to ping global Quad9 DNS server. Maybe it is blocked somewhere or you have connection issues in general. Are you able to ping something else on the internet like Google DNS server 8.8.8.8 ?

ygowthamr, 2021-09-17:
solved after some time

ygowthamr, 2021-09-29:
This fixes #4743

Joulinar, 2021-09-29:
What is the purpose of reopening?

ygowthamr, 2021-09-29:
``

closed 2021-09-29 by ygowthamr